jneilliii/OctoPrint-TPLinkSmartplug #3: HS105 error

dezmaeth, 2017-09-07:
got an error when trying to turn on or off 

TP-Link Smartplug Error
Status unknown. Double check ip address in TPLinkSmartplug Settings.

im using the HS105 

http://www.tp-link.com/us/products/details/cat-5516_HS105.html

[image]

jneilliii, 2017-09-07:
I've only tested with the TP-Link HS100 as that's what I have.  A big portion of my plugin was ripped off from [softScheck's work](https://github.com/softScheck/tplink-smartplug) and it states works for TP-Link HS100 and HS110 WiFi Smart Plugs.  I'm not saying it's impossible for it to work with your HS105, I jut don't have one to test with.  Did you go into settings and set the ip address of your HS105?  If not it has no way of knowing how to communicate with it.

jneilliii, 2017-09-07:
Also, I've only tested with OctoPrint running on a Pi.  If running from windows there may be some additional items that have to be done to get it to work.

mamejay, 2017-09-09:
Hi Jneilliii.  I get the same error while using a HS100.  Its an Australian model so hoping this is not the issue.  Anyway of tracing what is happening?

jneilliii, 2017-09-09:
Hey, if you install the plugin from the dev branch I added some more debugging. When it runs the toggle code if it gets a weird response from the plug that it didn't understand it will write additional info to the octoprint.log. Should help us debug a little more. Also, check your log now for any errors.

quidan, 2017-09-09:
I'm having what appears to be the same issue with a HS105. I've installed the dev branch and here is the snippet from my log:

> 2017-09-09 18:49:02,574 - octoprint.plugins.tplinksmartplug - INFO - Checking status.
> 2017-09-09 18:49:02,685 - octoprint.plugins.tplinksmartplug - INFO - Sending command info
> 2017-09-09 18:49:02,686 - octoprint - ERROR - Exception on /api/plugin/tplinksmartplug [POST]
> Traceback (most recent call last):
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
>     response = self.full_dispatch_request()
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
>     rv = self.handle_user_exception(e)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
>     reraise(exc_type, exc_value, tb)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
>     rv = self.dispatch_request()
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
>     return self.view_functions[rule.endpoint](**req.view_args)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/util/flask.py", line 1106, in decorated_view
>     return flask.ext.login.login_required(func)(*args, **kwargs)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg/flask_login.py", line 637, in decorated_view
>     return func(*args, **kwargs)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/api/__init__.py", line 99, in pluginCommand
>     response = api_plugin.on_api_command(command, data)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 90, in on_api_command
>     self.check_status()
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 68, in check_status
>     response = self.sendCommand("info")
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 134, in sendCommand
>     return json.loads(self.decrypt(data[4:]))
>   File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
>     return _default_decoder.decode(s)
>   File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
>     obj, end = self.raw_decode(s, idx=_w(s, 0).end())
>   File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
>     raise ValueError("No JSON object could be decoded")
> ValueError: No JSON object could be decoded
> 2017-09-09 18:49:02,689 - tornado.access - ERROR - 500 POST /api/plugin/tplinksmartplug (127.0.0.1) 146.65ms

jneilliii, 2017-09-09:
Thanks for that.  I've just updated the dev branch again with some additional debugging code to see what the response is coming from your tplink.  If you could please reinstall and get the last lines of octoprint.log after reload that would be great.  Also, what OS is your octoprint instance running on?  I've only tested on an OctoPi distro and it's worked on both of mine.

jneilliii, 2017-09-09:
Looks like there may also be some issues based on the version of python you're running.  The base code I appropriated for the calls to the plug are not compatible with Python 3, from what I'm seeing on it's github issues.

mamejay, 2017-09-10:
Pardon my ignorance but how can i install the plugin from the DEV branch?  I found the branch from the drop down but can someone help with the command?  I am familiar with SSH to the Pi

mamejay, 2017-09-10:
Here is an extract of what i receive in the log file.  
2017-09-10 21:34:04,181 - octoprint.plugins.tplinksmartplug - INFO - Checking status.
2017-09-10 21:34:05,298 - octoprint.plugins.tplinksmartplug - INFO - Sending command info to ip 192.168.1.112
2017-09-10 21:34:05,299 - octoprint.plugins.tplinksmartplug - INFO - 
2017-09-10 21:34:05,302 - octoprint - ERROR - Exception on /api/plugin/tplinksmartplug [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/util/flask.py", line 1106, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg/flask_login.py", line 637, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/api/__init__.py", line 99, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 90, in on_api_command
    self.check_status()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 68, in check_status
    response = self.sendCommand("info")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 135, in sendCommand
    return json.loads(self.decrypt(data[4:]))
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
2017-09-10 21:34:05,316 - tornado.access - ERROR - 500 POST /api/plugin/tplinksmartplug (127.0.0.1) 1197.81ms
2017-09-10 21:34:07,047 - octoprint.plugins.tplinksmartplug - INFO - Checking status.
2017-09-10 21:34:07,184 - octoprint.plugins.tplinksmartplug - INFO - Sending command info to ip 192.168.1.112
2017-09-10 21:34:07,186 - octoprint.plugins.tplinksmartplug - INFO - 
2017-09-10 21:34:07,187 - octoprint - ERROR - Exception on /api/plugin/tplinksmartplug [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/util/flask.py", line 1106, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg/flask_login.py", line 637, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/api/__init__.py", line 99, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 90, in on_api_command
    self.check_status()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 68, in check_status
    response = self.sendCommand("info")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 135, in sendCommand
    return json.loads(self.decrypt(data[4:]))
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
2017-09-10 21:34:07,193 - tornado.access - ERROR - 500 POST /api/plugin/tplinksmartplug (127.0.0.1) 207.04ms
2017-09-10 21:34:20,617 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.104
[octoprint.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1290474/octoprint.txt)

dezmaeth, 2017-09-10:
im running octopi on a raspberry pi, i did set the ip address of the HS105, gonna check the dev branch so i can get the log out

jneilliii, 2017-09-10:
Go into plugin manager of octoprint settings and install the debug_logging branch. This can be accomplished by opening the plugin manager in octoprint settings, click get more and use the following in the url field. Once installed go into tplinksmarplug settings and check the box for debugging. All debug messages will be saved into a separate log file that can be downloaded from the logs section in octoprint settings dialog. Hopefully we can get to the bottom of it. 

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/debug_logging.zip

mamejay, 2017-09-11:
Thanks for that.  I have now installed the debug version.  There is no icon to switch on and off in the top tool bar.  Here is an extract of the log.

> [2017-09-11 09:46:18,560] DEBUG: Checking status.
[2017-09-11 09:46:18,564] DEBUG: IP 192.168.1.112 is valid.
[2017-09-11 09:46:18,753] DEBUG: Sending command info to 192.168.1.112
[2017-09-11 09:46:18,754] DEBUG: 
[2017-09-11 09:47:50,191] DEBUG: Checking status.
[2017-09-11 09:47:50,194] DEBUG: IP 192.168.1.112 is valid.
[2017-09-11 09:47:50,267] DEBUG: Sending command info to 192.168.1.112
[2017-09-11 09:47:50,267] DEBUG: 
[2017-09-11 09:47:53,081] DEBUG: Checking status.
[2017-09-11 09:47:53,083] DEBUG: IP 192.168.1.112 is valid.
[2017-09-11 09:47:53,170] DEBUG: Sending command info to 192.168.1.112
[2017-09-11 09:47:53,170] DEBUG: 
[2017-09-11 09:47:53,194] DEBUG: Checking status.
[2017-09-11 09:47:53,197] DEBUG: IP 192.168.1.112 is valid.
[2017-09-11 09:47:53,315] DEBUG: Sending command info to 192.168.1.112
[2017-09-11 09:47:53,315] DEBUG: 
[2017-09-11 09:49:29,508] DEBUG: Checking status.
[2017-09-11 09:49:29,511] DEBUG: IP 192.168.1.112 is valid.
[2017-09-11 09:49:29,554] DEBUG: Sending command info to 192.168.1.112
[2017-09-11 09:49:29,555] DEBUG: 
[2017-09-11 09:49:32,245] DEBUG: Checking status.
[2017-09-11 09:49:32,248] DEBUG: IP 192.168.1.112 is valid.
[2017-09-11 09:49:32,312] DEBUG: Sending command info to 192.168.1.112
[2017-09-11 09:49:32,313] DEBUG: 
[2017-09-11 09:49:32,336] DEBUG: Checking status.
[2017-09-11 09:49:32,339] DEBUG: IP 192.168.1.112 is valid.
[2017-09-11 09:49:32,457] DEBUG: Sending command info to 192.168.1.112
[2017-09-11 09:49:32,458] DEBUG: 

> 

Let me know what you think

jneilliii, 2017-09-11:
Ok, based on the previous octoprint.log and this separate log it appears that the smartplug your trying to control is not responding to the api calls.  I'm not sure if this is because of the firmware on your smartplug because of it being Australian model or that is doesn't support the connection at all.  The only think I can think of moving forward would be to try and run the original python script I based this work on to see if it has the same results.  You would need to ssh to your pi and run the commands below.
```
wget https://raw.githubusercontent.com/softScheck/tplink-smartplug/master/tplink-smartplug.py
sudo chmod +x tplink-smartplug.py
./tplink-smartplug.py -t 192.168.1.112 -c info
```

jneilliii, 2017-09-11:
Oh, and this latest debug_logging branch also won't display the button if it's not able to communicate with the smartplug.  That's actually by design now.

mamejay, 2017-09-11:
This is what I get using the direct command.

> pi@octopi:~ $ ./tplink-smartplug.py -t 192.168.1.112 -c info
Sent:      {"system":{"get_sysinfo":{}}}
Received:
pi@octopi:~ $

> 
Just an FYI.  Firmware is 1.2.5 and Hardware version is 2.0 of the HS100(AU)

mamejay, 2017-09-11:
Not sure if this helps but ran the script VIA my PC and used Wireshark to capture the traffic.  This was an off command.

> Frame 6522: 100 bytes on wire (800 bits), 100 bytes captured (800 bits) on interface 0
    Interface id: 0 (\Device\NPF_{04593795-C79B-4433-9BD8-662AE900044A})
        Interface name: \Device\NPF_{04593795-C79B-4433-9BD8-662AE900044A}
    Encapsulation type: Ethernet (1)
    Arrival Time: Sep 11, 2017 21:37:05.060490000 AUS Eastern Standard Time
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1505129825.060490000 seconds
    [Time delta from previous captured frame: 0.000113000 seconds]
    [Time delta from previous displayed frame: 30.423521000 seconds]
    [Time since reference or first frame: 258.308766000 seconds]
    Frame Number: 6522
    Frame Length: 100 bytes (800 bits)
    Capture Length: 100 bytes (800 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:ip:tcp:tplink-smarthome:json]
    [Coloring Rule Name: TCP]
    [Coloring Rule String: tcp]
Ethernet II, Src: LiteonTe_9f:28:72 (28:e3:47:9f:28:72), Dst: Tp-LinkT_66:b7:40 (50:c7:bf:66:b7:40)
    Destination: Tp-LinkT_66:b7:40 (50:c7:bf:66:b7:40)
        Address: Tp-LinkT_66:b7:40 (50:c7:bf:66:b7:40)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Source: LiteonTe_9f:28:72 (28:e3:47:9f:28:72)
        Address: LiteonTe_9f:28:72 (28:e3:47:9f:28:72)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: IPv4 (0x0800)
Internet Protocol Version 4, Src: 192.168.1.104, Dst: 192.168.1.112
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
        0000 00.. = Differentiated Services Codepoint: Default (0)
        .... ..00 = Explicit Congestion Notification: Not ECN-Capable Transport (0)
    Total Length: 86
    Identification: 0x750a (29962)
    Flags: 0x02 (Don't Fragment)
        0... .... = Reserved bit: Not set
        .1.. .... = Don't fragment: Set
        ..0. .... = More fragments: Not set
    Fragment offset: 0
    Time to live: 128
    Protocol: TCP (6)
    Header checksum: 0x016f [validation disabled]
    [Header checksum status: Unverified]
    Source: 192.168.1.104
    Destination: 192.168.1.112
    [Source GeoIP: Unknown]
    [Destination GeoIP: Unknown]
Transmission Control Protocol, Src Port: 53998, Dst Port: 9999, Seq: 1, Ack: 1, Len: 46
    Source Port: 53998
    Destination Port: 9999
    [Stream index: 39]
    [TCP Segment Len: 46]
    Sequence number: 1    (relative sequence number)
    [Next sequence number: 47    (relative sequence number)]
    Acknowledgment number: 1    (relative ack number)
    0101 .... = Header Length: 20 bytes (5)
    Flags: 0x018 (PSH, ACK)
        000. .... .... = Reserved: Not set
        ...0 .... .... = Nonce: Not set
        .... 0... .... = Congestion Window Reduced (CWR): Not set
        .... .0.. .... = ECN-Echo: Not set
        .... ..0. .... = Urgent: Not set
        .... ...1 .... = Acknowledgment: Set
        .... .... 1... = Push: Set
        .... .... .0.. = Reset: Not set
        .... .... ..0. = Syn: Not set
        .... .... ...0 = Fin: Not set
        [TCP Flags: ·······AP···]
    Window size value: 17520
    [Calculated window size: 17520]
    [Window size scaling factor: -2 (no window scaling used)]
    Checksum: 0x6543 [unverified]
    [Checksum Status: Unverified]
    Urgent pointer: 0
    [SEQ/ACK analysis]
        [iRTT: 0.002318000 seconds]
        [Bytes in flight: 46]
        [Bytes sent since last PSH flag: 46]
    TCP payload (46 bytes)
TP-Link Smart Home Protocol (decrypted)
    {"system":{"set_relay_state":{"state":0}}}
JavaScript Object Notation
    Object
        Member Key: system
            Object
                Member Key: set_relay_state
                    Object
                        Member Key: state
                            Number value: 0
                            Key: state
                    Key: set_relay_state
            Key: system

>

stuartsmi, 2017-09-11:
Mamejay, its interesting that your firmware is different to mine 1.0.10 HS100(AU). I checked for a firmware update and it said there wasn't one. Yours seems newer than mine. I am having the same problem as you though.

jneilliii, 2017-09-11:
So I'm at a loss for you guys.  My hardware is version 1.0 and sw version is 1.1.1 Build 160725.  From the results of your wireshark output, it appears that there is information being passed back and forth. Could you run the same wireshark using the "info" command to see if it's returning anything? I'm wondering if TP-Link may have changed their internal api commands maybe.

mamejay, 2017-09-11:
I am starting to think that is what TP-Link may have done.  Here is a wireshark extract from the info command.

> Frame 327: 87 bytes on wire (696 bits), 87 bytes captured (696 bits) on interface 0
    Interface id: 0 (\Device\NPF_{04593795-C79B-4433-9BD8-662AE900044A})
        Interface name: \Device\NPF_{04593795-C79B-4433-9BD8-662AE900044A}
    Encapsulation type: Ethernet (1)
    Arrival Time: Sep 12, 2017 07:34:14.491607000 AUS Eastern Standard Time
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1505165654.491607000 seconds
    [Time delta from previous captured frame: 0.000096000 seconds]
    [Time delta from previous displayed frame: 0.000000000 seconds]
    [Time since reference or first frame: 11.144488000 seconds]
    Frame Number: 327
    Frame Length: 87 bytes (696 bits)
    Capture Length: 87 bytes (696 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:ip:tcp:tplink-smarthome:json]
    [Coloring Rule Name: TCP]
    [Coloring Rule String: tcp]
Ethernet II, Src: LiteonTe_9f:28:72 (28:e3:47:9f:28:72), Dst: Tp-LinkT_66:b7:40 (50:c7:bf:66:b7:40)
    Destination: Tp-LinkT_66:b7:40 (50:c7:bf:66:b7:40)
        Address: Tp-LinkT_66:b7:40 (50:c7:bf:66:b7:40)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Source: LiteonTe_9f:28:72 (28:e3:47:9f:28:72)
        Address: LiteonTe_9f:28:72 (28:e3:47:9f:28:72)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: IPv4 (0x0800)
Internet Protocol Version 4, Src: 192.168.1.104, Dst: 192.168.1.112
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
        0000 00.. = Differentiated Services Codepoint: Default (0)
        .... ..00 = Explicit Congestion Notification: Not ECN-Capable Transport (0)
    Total Length: 73
    Identification: 0x12fb (4859)
    Flags: 0x02 (Don't Fragment)
        0... .... = Reserved bit: Not set
        .1.. .... = Don't fragment: Set
        ..0. .... = More fragments: Not set
    Fragment offset: 0
    Time to live: 128
    Protocol: TCP (6)
    Header checksum: 0x638b [validation disabled]
    [Header checksum status: Unverified]
    Source: 192.168.1.104
    Destination: 192.168.1.112
    [Source GeoIP: Unknown]
    [Destination GeoIP: Unknown]
Transmission Control Protocol, Src Port: 54983, Dst Port: 9999, Seq: 1, Ack: 1, Len: 33
    Source Port: 54983
    Destination Port: 9999
    [Stream index: 21]
    [TCP Segment Len: 33]
    Sequence number: 1    (relative sequence number)
    [Next sequence number: 34    (relative sequence number)]
    Acknowledgment number: 1    (relative ack number)
    0101 .... = Header Length: 20 bytes (5)
    Flags: 0x018 (PSH, ACK)
        000. .... .... = Reserved: Not set
        ...0 .... .... = Nonce: Not set
        .... 0... .... = Congestion Window Reduced (CWR): Not set
        .... .0.. .... = ECN-Echo: Not set
        .... ..0. .... = Urgent: Not set
        .... ...1 .... = Acknowledgment: Set
        .... .... 1... = Push: Set
        .... .... .0.. = Reset: Not set
        .... .... ..0. = Syn: Not set
        .... .... ...0 = Fin: Not set
        [TCP Flags: ·······AP···]
    Window size value: 17520
    [Calculated window size: 17520]
    [Window size scaling factor: -2 (no window scaling used)]
    Checksum: 0x122e [unverified]
    [Checksum Status: Unverified]
    Urgent pointer: 0
    [SEQ/ACK analysis]
        [iRTT: 0.002552000 seconds]
        [Bytes in flight: 33]
        [Bytes sent since last PSH flag: 33]
    TCP payload (33 bytes)
TP-Link Smart Home Protocol (decrypted)
    {"system":{"get_sysinfo":{}}}
JavaScript Object Notation
    Object
        Member Key: system
            Object
                Member Key: get_sysinfo
                    Object
                    Key: get_sysinfo
            Key: system

> 
[info.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1294140/info.txt)

jneilliii, 2017-09-12:
Yeah, it definitely is responding but without any data coming back from the info command like the on command is.  I can easily enable a bypass for the ip check for you guys until we can figure out what the command needs to be. I'll try to get that up in the next few days.

jneilliii, 2017-09-12:
Actually, I guess I didn't ask.  Does the on/off commands from the python script work at all?

mamejay, 2017-09-12:
Not getting any off/on response from the device using the python script.  Only working VIA the Kasa APP on a mobile device.  Strange that different models don't work.  Would have thought all models would use the same commands to perform tasks

mamejay, 2017-09-12:
Wonder if there is a way to force a firmware update onto the device and use a known working firmware

stuartsmi, 2017-09-12:
I am assuming that the firmware and hardware version differences are because of the differing channels available in different regions for wifi. My HS100(AU) is hardware version 2.0 and firmware 1.0.10. Kasa tells me there are no available firmware updates. Did you update yours or did it come with that firmware?

jneilliii, 2017-09-12:
I'd be scared of swapping firmware if it's relative to electricity as it might mess things up.  I did find a link to a zip file from the guys that reversed engineered with a bunch of firmware in it, but I personally wouldn't do it. http://static.tp-link.com/iotUpgradeTool_V1.0.zip.

stuartsmi, 2017-09-12:
I wouldn't be too concerned with forcing firmware if the hardware version matched. Its just a relay in the device that is switching so voltage is unlikely to be an issue. It would be interesting to see if there are differing hardware versions out there. As it is, they HS100(AU) is only AU$39. Its not a great loss if it gets bricked.

jneilliii, 2017-09-12:
The one thing you could try prior to potentially bricking is wireshark the communication between the plug and the kasa app if you can man in the middle the communication, or port mirror on your switch.  I don't have that capability here.  That would allow you to see if the wireshark dissector is able to decrypt the information and then you'd know for sure if the encryption algorithms are different.

stuartsmi, 2017-09-12:
Jneilliii, what hardware and firmware version are your HS100?

jneilliii, 2017-09-12:
The KASA app is reporting it as hardware version 1.0 and firmware version 1.1.1.  Model is HS100(US).

jneilliii, 2017-09-12:
The detailed build information by using the script reveals it is 1.1.1 Build 160725 and the KASA app is reporting no firmware updates available, which seems wrong when the zip file I linked has newer versions assuming the bin names match the build number.

jneilliii, 2017-09-12:
I just ran the update tool and was able to find the device.  I'm too chicken to upgrade it and break my set-up. I wonder if you could wireshark that upgrade tool.

jneilliii, 2017-09-12:
Ok, so I went ahead and did the firmware upgrade using the tool and it worked and my plugin is still operational. It is now showing version 1.1.2 Build 161014 Rel.164426 in the info response.

stuartsmi, 2017-09-12:
Its interesting that it couldn't find the update just like mine cant, yet mamejays one is the same hardware version and model as mine, but has more modern firmware. Makes one think the firmware update in kasa is broken.

stuartsmi, 2017-09-12:
I just tried the upgrade tool and it failed with a download timeout.

mamejay, 2017-09-12:
Its strange.  Mine said there was not update but ran on tge second try and updated.  I am an amateur with wireshark but will see if i can get some more info.

stuartsmi, 2017-09-12:
All I am getting is:
2017/09/12 13:24:56>> Start upgrade device HS100(AU) 50:C7:BF:66:B6:E9
2017/09/12 13:24:56>> Target type is SmartPlug
2017/09/12 13:24:56>> Send upgrade request
2017/09/12 13:24:56>> Upgrade request status is 0
2017/09/12 13:24:56>> Start downloading...
2017/09/12 13:25:11>> Download status get timeout

Not sure if its a timeout getting the download from TPlink or performing the download to the device.

mamejay, 2017-09-12:
I am starting to think that as its hardware version 2 that could be the issue.  Going by the firmware listing of the iotUpgradeTool.

> hs100v1_eu_1.1.2_Build_161014_Rel.164550.bin
hs100v1_fr_1.1.2_Build_161014_Rel.170646.bin
hs100v1_uk_1.1.2_Build_161014_Rel.170836.bin
hs100v1_us_1.1.2_Build_161014_Rel.164426.bin
hs110v1_eu_1.1.2_Build_161014_Rel.164733.bin
hs110v1_fr_1.1.2_Build_161014_Rel.170531.bin
hs110v1_uk_1.1.2_Build_161014_Rel.170948.bin
hs110v1_us_1.1.2_Build_161014_Rel.164223.bin
hs200v1_us_1.1.0_Build_160521_Rel.085826.bin
smartBulb_FCC_1.4.3_Build_170504_Rel.144921.bin
smartPlug_CE_HS100_110(AU)_1.2.5_Build_170615_Rel.142226.bin

> smartPlug_FCC_HS105(US)_1.2.1_Build_170406_Rel.135348.bin

stuartsmi, 2017-09-12:
mamejay, yours and mine are the same model and hardware version. the kasa app wont upgrade mine to the version yours is running. says there is none available. I wonder if they pulled your release.

stuartsmi, 2017-09-12:
That's interesting. After telling me that the update failed, I power cycled the smartplug to attempt the update again to have the tool report that it is up to date. It now reports 1.2.5. Unfortunately I am unable to do much with the Octopi as I am in the middle of a long print.

jneilliii, 2017-09-12:
Are the internet restrictions in Australia more stringent than the FCC in the US?  It's possible that they patched some of these "vulnerabilities" in that newer hardware/firmware versions.

jneilliii, 2017-09-12:
You guys might get better traction on a possible resolution over on the other [github repository](https://github.com/softScheck/tplink-smartplug) where the guys that actually did the reverse engineering are.  A lot of that stuff is over my head.

jneilliii, 2017-09-12:
Sorry, didn't mean to close.

mamejay, 2017-09-13:
Hi jneilliii.  I suspect that this is all related to v2 of the device.  Just checking the firmware size you can see that v2 is 513k and all others are 3mb in size.  There is a us version also with the same firmware size so i think this is not related to Australian laws but to a rewrite of the code.  I will raise an issue with the original people that reversed engineered the code.

stuartsmi, 2017-09-13:
It would surprise me if they have changed the API with v2 hardware. It just doesn't make much sense.

jneilliii, 2017-09-13:
I agree that wouldn't make sense, but there is something definitely different in how that newer hardware is responding and I did also notice the difference in firmware size on those newer hardware models.

jkliman, 2017-09-22:
So I have HS105, I am in the US. It is running H/W v1.0  F/W v1.2.9. Could this possibly be an administrator issue, I have tried turning remote access off to see if the KASA software fowarding the IP address could have some sort of effect on the local control of the device, with no effect. My debug files shows the same exact results as mamejay's did 12 days ago.

uwwint, 2017-09-22:
Hi guys. I had the same problem and wiresharked my way through HW 2.0/FW 1.0.10(AU) version. I believe they do things different down under. Nevertheless, with the following change it works:

 ```
               commands = {'info' : ('{"schedule":{"get_next_action":{}},"system":{"get_sysinfo":{}}}','\x3f'),
                        'on'       : ('{"system":{"set_relay_state":{"state":1}}}','\x2a'),
                        'off'      : ('{"system":{"set_relay_state":{"state":0}}}','\x2a'),
                        'emeter'   : ('{"emeter":{"get_daystat":{"year":2017,"month":9}},"schedule":{"get_daystat":{"year":2017,"month":9}}}','\x65'),
                        'wlanscan'    : ('{"netif":{"get_scaninfo":{"refresh":1,"timeout":17}}}','\x35')
                }
```

       ```
 def encrypt(self, string):
                key = 171
                k = string[1]
                string = string[0]
                result = "\0\0\0"+k
                for i in string:
                        a = key ^ ord(i)
                        key = a
                        result += chr(a)
                return result
```
don't know if this works with the non Australian ones though.

jkliman, 2017-09-22:
I am in the middle of a print, I will update this lines of code and report back in

uwwint, 2017-09-22:
Update... I am a moron. Only update encrypt with the following:

` def encrypt(self, string):
                key = 171
                result = "\0\0\0"+chr(len(string))
                for i in string:
                        a = key ^ ord(i)
                        key = a
                        result += chr(a)
                return result`

jneilliii, 2017-09-22:
I've updated the dev branch with this encryption code change. If you install into OctoPrint from that branch it should work for you guys if it's working in the default python scripts.

jneilliii, 2017-09-22:
Testing dev branch with version 1.0 hardware to see if it is cross-compatible with new encryption function.

jneilliii, 2017-09-22:
Update encrypt code appears to be working with Hardware version 1.0 as well.  I'll modify a couple of more files and release a version update shortly.

jkliman, 2017-09-22:
updated dev pull is working on the HS105 H/W: 1.0 F/W 1.2.9 US version flawlessly.........Thanks for the quick responses!

jneilliii, 2017-09-22:
Awesome, I just released version 0.5.0 to the master branch with these changes as it appears it's working for my hardware as well.  Should now be more universal.  Thanks again go out to @uwwint for the hard work.

jneilliii, 2017-09-22:
Fixed and release in version 0.5.0.

closed 2017-09-22 by jneilliii

mamejay, 2017-09-22:
uwwint you are a superstar!!!  This is now working.  Thanks so much to all you clever people.  jneilliii thanks so much for your patience through all of these emails.  Hopefully this will be the last change needed to cover all models now.

jneilliii, 2017-09-22:
No worries @mamejay, I'm glad we were able to figure it out.  The wireshark/decryption is beyond my knowledge, but hacking code together is fairly easy for me.

stuartsmi, 2017-09-23:
Just upgraded to 0.5.0 and it all works. Thanks very much for all your work.

stuartsmi, 2017-09-23:
Have you considered allowing multiple smartplugs to be controlled. I'd like to be able to control the leds I use to light my printer as well as turn the printer on and off. I dont have a need to have the leds on all the time that the printer is running. Just when I am checking on a print.

jneilliii, 2017-09-23:
see #14. I'm working on it.

[5 more replies]