jneilliii/OctoPrint-TPLinkSmartplug #4: Power icon doesn't show

TFMvdBroek, 2017-09-08:
Power icon doesn't show up for me. Entered the IP address, restarted Pi, still nothing.

jneilliii, 2017-09-09:
What OS is your octoprint running on, and which browser are you using?

jneilliii, 2017-09-11:
Also, if you could upload your octoprint.log file that may give us some more information into the possible issue.

TFMvdBroek, 2017-09-13:
I am on 1.3.4 and the log shows a rights error:

```
File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 82, in on_api_command
    return make_response("Insufficient rights", 403)
NameError: global name 'make_response' is not defined
2017-09-13 16:54:32,124 - tornado.access - ERROR - 500 POST /api/plugin/tplinksmartplug (127.0.0.1) 97.28ms
2017-09-13 16:54:32,168 - tornado.access - WARNING - 403 POST /api/settings (127.0.0.1) 65.13ms
2017-09-13 16:54:32,497 - tornado.access - WARNING - 403 GET /api/users (127.0.0.1) 17.85ms
2017-09-13 16:54:32,520 - tornado.access - WARNING - 403 GET /api/system/commands (127.0.0.1) 18.80ms
2017-09-13 16:54:32,542 - tornado.access - WARNING - 403 GET /plugin/announcements/channels (127.0.0.1) 17.32ms
2017-09-13 16:54:32,563 - tornado.access - WARNING - 403 GET /api/plugin/pluginmanager (127.0.0.1) 17.23ms
2017-09-13 16:54:38,609 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2017-09-13 16:54:38,896 - octoprint.filemanager.analysis - INFO - Starting analysis of local:biohazard_coaster(4).gcode
2017-09-13 16:54:45,009 - tornado.access - WARNING - 403 POST /api/settings (127.0.0.1) 74.52ms
2017-09-13 16:54:47,496 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-09-13 16:54:48,006 - tornado.access - WARNING - 403 POST /api/settings (127.0.0.1) 67.42ms
2017-09-13 16:54:48,071 - tornado.access - WARNING - 403 POST /api/settings (127.0.0.1) 43.40ms
2017-09-13 16:54:56,636 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-09-13 16:54:58,205 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-09-13 16:54:59,018 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin V1.0.2; Sprinter/grbl mashup for gen6"
2017-09-13 16:55:02,312 - tornado.access - WARNING - 403 POST /api/settings (127.0.0.1) 155.37ms
2017-09-13 16:55:07,671 - tornado.access - WARNING - 403 GET /api/languages (127.0.0.1) 35.77ms
2017-09-13 16:55:07,756 - tornado.access - WARNING - 403 GET /api/logs (127.0.0.1) 41.36ms
2017-09-13 16:55:11,789 - tornado.access - WARNING - 403 POST /api/settings (127.0.0.1) 179.58ms
2017-09-13 16:55:14,109 - tornado.access - WARNING - 403 GET /api/languages (127.0.0.1) 41.15ms
2017-09-13 16:55:14,328 - tornado.access - WARNING - 403 GET /api/logs (127.0.0.1) 26.97ms
```

jneilliii, 2017-09-13:
Ok, I've added the missing reference that was causing that error.  See if that helps you.  The fact that I haven't run into this one before makes me curious if there is a user error of some sort due to not being logged in maybe.  I don't use OctoPrint's account control so haven't run into it yet.  Give the latest 0.3.1 version a try and let me know if you're still seeing those errors now.

FireBookDuo, 2017-09-22:
Having a similar issue with the icon not showing up

```
2017-09-22 19:58:20,460 - octoprint - ERROR - Exception on /api/plugin/tplinksmartplug [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/util/flask.py", line 1106, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg/flask_login.py", line 637, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/api/__init__.py", line 99, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 136, in on_api_command
    self.check_status()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 113, in check_status
    response = self.sendCommand("info")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 196, in sendCommand
    return json.loads(self.decrypt(data[4:]))
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
2017-09-22 19:58:20,468 - tornado.access - ERROR - 500 POST /api/plugin/tplinksmartplug (127.0.0.1) 178.81ms
2017-09-22 20:00:41,418 - octoprint - ERROR - Exception on /api/plugin/tplinksmartplug [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/util/flask.py", line 1106, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg/flask_login.py", line 637, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/server/api/__init__.py", line 99, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 136, in on_api_command
    self.check_status()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 113, in check_status
    response = self.sendCommand("info")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 196, in sendCommand
    return json.loads(self.decrypt(data[4:]))
  File "/usr/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
```

uwwint, 2017-09-22:
Hi @FireBookDuo. Have you tried version 0.5 just released? @jneilliii implemented a change in the encryption function.

eightheads, 2017-09-25:
I'm running into this occasionally as well. However, if I logout then back in it will show again.

dezmaeth, 2017-09-29:
it's happening to me too , i don't see the icon but it's in the html,  id="tplinksmartplug_indicator", it's just hidden and doesn't really do anything

jneilliii, 2017-09-30:
It's hidden until you enter a valid ip address or hostname for your plug in settings. Once it's able to verify communication with the plug it should show then. If you have already done that then enable debugging in the plugin settings and upload both octoprint.log and the tplinksmartplug-debug.log.

slimshizn, 2017-10-30:
[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1427087/octoprint.log)

[plugin_tplinksmartplug_debug.log.2017-10-27.zip](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1427093/plugin_tplinksmartplug_debug.log.2017-10-27.zip)

Can't connect, weird because a I was able to before in a previous version. This is the MINI plug. Can't use that bulky one they sell due to using a UPS.

jneilliii, 2017-10-30:
If you upgraded from a previous version it will wipe out the settings in order to accommodate changes. Have you gone into settings and updated the ip since ugprade?

slimshizn, 2017-10-30:
I never upgraded from a previous though, this was the first I’ve bought. Yes I’ve checked the IP.

From: jneilliii
Sent: Monday, October 30, 2017 5:09 PM
To: jneilliii/OctoPrint-TPLinkSmartplug
Cc: slimshizn; Comment
Subject: Re: [jneilliii/OctoPrint-TPLinkSmartplug] Power icon doesn't show(#4)

If you upgraded from a previous version it will wipe out the settings in order to accommodate changes. Have you gone into settings and updated the ip since ugprade?
—

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

jneilliii, 2017-10-30:
Ok, just looked at your debug log and I see where it is talking to the plug correctly.  Can you do me a favor and try a Ctrl+F5 on the page to reload the page completely?

slimshizn, 2017-10-30:
Okay trying that I’ll let you know if it works.

From: jneilliii
Sent: Monday, October 30, 2017 5:16 PM
To: jneilliii/OctoPrint-TPLinkSmartplug
Cc: slimshizn; Comment
Subject: Re: [jneilliii/OctoPrint-TPLinkSmartplug] Power icon doesn't show(#4)

Ok, just looked at your debug log and I see where it is talking to the plug correctly. Can you do me a favor and try a Ctrl+F5 on the page to reload the page completely?
—

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

slimshizn, 2017-10-30:
It’s showing in grey still.

From: jneilliii
Sent: Monday, October 30, 2017 5:16 PM
To: jneilliii/OctoPrint-TPLinkSmartplug
Cc: slimshizn; Comment
Subject: Re: [jneilliii/OctoPrint-TPLinkSmartplug] Power icon doesn't show(#4)

Ok, just looked at your debug log and I see where it is talking to the plug correctly. Can you do me a favor and try a Ctrl+F5 on the page to reload the page completely?
—

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

slimshizn, 2017-10-30:
Sorry about the email crap, yeah I tried the Ctrl F5 and it's still grey.

slimshizn, 2017-10-30:
[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1428852/octoprint.log)

New log.

jneilliii, 2017-10-30:
The log is showing a 401/unauthorized error. Try logging out and back in like mentioned earlier and see if that works. Also, when the plug is gray, clicking it should try to get the current status of the plug.

slimshizn, 2017-10-31:
Tried logging back out and in, reloading and clicking. Didn't help, when I click the grey icon it shows nothing.

slimshizn, 2017-11-01:
[plugin_tplinksmartplug_debug (2).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1432755/plugin_tplinksmartplug_debug.2.log)
[octoprint (1).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1432756/octoprint.1.log)

New logs after a delete and restart.

jneilliii, 2017-11-08:
another change in encrypt function now on dev branch for testing.  Install using url below in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/dev.zip

slimshizn, 2017-11-08:
Copy will do

closed 2018-01-29 by jneilliii