jneilliii/OctoPrint-TPLinkSmartplug #5: Would be great to have

Mattis30, 2017-09-08:
Hi, 
would be great to have something like a shutdown command for the plug afteer a print is finished. I mean, that the pi stays after the print is finished but the smartplug shuts down. 
Maybe there is something like that and i'm just to stupid to find out...

jneilliii, 2017-09-09:
I'll consider adding that as an option in the future.

Flibblebot, 2017-09-09:
I think probably the easiest way to do this would be to use M80/M81 gcodes to turn the plug on and off. You should be able to do this using gcode hooks - I think the PSUControl plugin does this, so it should be possible with the Smartplug plugin too.

jneilliii, 2017-09-09:
That definitely would work. I've done other gcode hooks for my M117 plugins and I actually stripped that part out of the psucontrol plugin when making this one. The nice thing about that method is it would allow individual control per print, and not have to change settings prior to printing.

jneilliii, 2017-09-10:
So after testing this option I came to realize you could only benefit from the M80 command if you're using the smartplug to control some other device other than the printer.  Because once the printer powers off using M81 OctoPrint disconnects and can't send any additional gcode.  I went ahead and added it for that purpose (ie controlling external led strip power maybe). You can get it from the gcodeprocessing branch. Install via url https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/gcodeprocessing.zip in plugin manager.  I will merge with master in the future once I work out some of these other "not working for me" issues.

Mattis30, 2017-09-12:
WEll, excuse me for not knowing better but i would think it should be relatively easy to implement such a functionality. There are plugins as well which shut down octoprint after finishing. I guess it could be done to combine it with the print state change from Printing to Operational or something similar

Huntermountain, 2017-09-12:
I'm just a user, but I would love to be able to shut down the mains to my entire printer setup by shutting off the Tp-Link plug after the print has finished. I just received my plug today, and I actually thought this was part of the functionality of this plug in.

jneilliii, 2017-09-12:
@Mattis30: I'm not sure where the confusion is in your response, but I have implemented the functionality in that separate branch.  The steps to get this functionality are install via plugin manager using the url https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/gcodeprocessing.zip.  Once installed, go into TPLinkSmartplug Settings dialog and check the box for process M80 and M81 commands from gcode and save.  Then either modify the OctoPrint settings for the after print job completes in the gcode scripts section to include M81 at the end of it, or modify your end gcode commands in your slicer to have that M81 command.

@Huntermountain: See instructions above for this functionality.

jneilliii, 2017-09-13:
I've merged these changes into the master branch and released version 0.3.0.

closed 2017-09-13 by jneilliii