jneilliii/OctoPrint-TPLinkSmartplug #9: External Commands when turned on/off

ttocsneb, 2017-09-13:
I would love to see a feature where you can enter a command to be run when the smart plug is turned on, or another when it is turned off.

jneilliii, 2017-09-14:
Can you please elaborate on this enhancement request?  Just trying to get a better idea of a use case for that.

ttocsneb, 2017-09-15:
Yes, similar to the server section of the octoprint settings, have the option to enter two shell commands. One to execute when the plug is turned on, and the other for when it is turned off.  

This way If you were to turn off the printer through the button, or an M81 command, a script would be run.  For me it would turn off the camera, and a fan, and a separate command that would turn the camera and fan on when the plug is turned on.

I hope that helps

jneilliii, 2017-09-16:
I've got the initial framework for this working I think.  I just used Octoprint's reboot server command as a test and it definitely rebooted my pi.  If you don't mind giving it a try and reporting back that would be great.  To install go into plugin manager and install from URL https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/sys_cmd.zip.  Once installed restart OctoPrint and SHIFT+F5 reload to make sure the interface updates.  Then in TPLinkSmartplug Settings enable the options and enter the commands you want to run.

jneilliii, 2017-09-16:
It issues the system command entered after a "smartplug on" command and before a "smartplug off" command, which makes sense to me if you're trying to do things to stuff that might be plugged into the smartplug.

jneilliii, 2017-09-16:
Released changes to version 0.4.0.

closed 2017-09-16 by jneilliii