jneilliii/OctoPrint-TPLinkSmartplug #12: Not working with the HS105

jkliman, 2017-09-22:
I have an HS105, ip address is correct. I think the plug-in identifies there is a unit there but will not display the power icon on the dashboard below is the debug log.
[2017-09-22 03:55:43,675] DEBUG: Checking status.
[2017-09-22 03:55:43,676] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 03:55:43,810] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 03:55:43,811] DEBUG: 
[2017-09-22 03:57:38,722] DEBUG: Checking status.
[2017-09-22 03:57:38,723] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 03:57:38,789] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 03:57:38,789] DEBUG: 
[2017-09-22 03:58:52,961] DEBUG: Checking status.
[2017-09-22 03:58:52,963] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 03:58:54,290] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 03:58:55,450] DEBUG: 
[2017-09-22 04:02:52,853] DEBUG: Checking status.
[2017-09-22 04:02:52,855] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 04:02:54,119] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 04:02:54,120] DEBUG: 
[2017-09-22 04:06:56,868] DEBUG: Checking status.
[2017-09-22 04:06:56,871] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 04:06:57,097] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 04:06:57,098] DEBUG: 
[2017-09-22 04:08:31,597] DEBUG: Checking status.
[2017-09-22 04:08:31,599] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 04:08:31,741] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 04:08:31,741] DEBUG: 
[2017-09-22 04:10:56,909] DEBUG: Checking status.
[2017-09-22 04:10:56,911] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 04:10:57,078] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 04:10:57,078] DEBUG: 
[2017-09-22 04:45:15,603] DEBUG: Checking status.
[2017-09-22 04:45:15,606] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 04:45:16,877] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 04:45:16,878] DEBUG: 
[2017-09-22 04:56:00,957] DEBUG: Checking status.
[2017-09-22 04:56:01,320] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 04:56:01,453] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 04:56:01,454] DEBUG: 
[2017-09-22 05:01:33,241] DEBUG: Checking status.
[2017-09-22 05:01:33,245] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 05:01:33,428] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 05:01:33,428] DEBUG: 
[2017-09-22 05:05:17,701] DEBUG: Checking status.
[2017-09-22 05:05:17,702] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 05:05:20,069] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 05:05:20,070] DEBUG: 
[2017-09-22 05:06:17,411] DEBUG: Checking status.
[2017-09-22 05:06:17,412] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 05:06:28,314] DEBUG: Could not connect to 192.168.1.188.
[2017-09-22 05:06:28,315] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}}
[2017-09-22 05:38:17,620] DEBUG: Checking status.
[2017-09-22 05:38:17,622] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 05:38:18,393] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 05:38:18,393] DEBUG: 
[2017-09-22 05:38:18,616] DEBUG: Checking status.
[2017-09-22 05:38:18,617] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 05:38:19,064] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 05:38:19,064] DEBUG: 
[2017-09-22 05:39:21,972] DEBUG: Checking status.
[2017-09-22 05:39:21,998] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 05:39:22,165] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 05:39:22,165] DEBUG: 
[2017-09-22 12:07:58,538] DEBUG: Checking status.
[2017-09-22 12:07:58,540] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 12:07:58,616] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 12:07:58,616] DEBUG: 
[2017-09-22 12:09:17,453] DEBUG: Checking status.
[2017-09-22 12:09:17,455] DEBUG: Invalid ip Http://192.168.1.188 trying hostname.
[2017-09-22 12:09:17,469] DEBUG: Invalid hostname Http://192.168.1.188.
[2017-09-22 12:09:17,471] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}}
[2017-09-22 12:09:20,242] DEBUG: Checking status.
[2017-09-22 12:09:20,243] DEBUG: Invalid ip Http://192.168.1.188 trying hostname.
[2017-09-22 12:09:20,261] DEBUG: Invalid hostname Http://192.168.1.188.
[2017-09-22 12:09:20,262] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}}
[2017-09-22 12:09:39,733] DEBUG: Checking status.
[2017-09-22 12:09:39,735] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 12:09:40,027] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 12:09:40,028] DEBUG: 
[2017-09-22 12:09:40,259] DEBUG: Checking status.
[2017-09-22 12:09:40,262] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 12:09:40,632] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 12:09:40,633] DEBUG: 
[2017-09-22 12:10:46,042] DEBUG: Checking status.
[2017-09-22 12:10:46,043] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 12:10:46,184] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 12:10:46,185] DEBUG: 
[2017-09-22 13:19:07,277] DEBUG: Checking status.
[2017-09-22 13:19:07,279] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 13:19:07,450] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 13:19:07,462] DEBUG: 
[2017-09-22 13:19:20,186] DEBUG: Checking status.
[2017-09-22 13:19:20,188] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 13:19:20,217] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 13:19:20,217] DEBUG: 
[2017-09-22 13:19:28,765] DEBUG: Checking status.
[2017-09-22 13:19:28,766] DEBUG: IP 192.168.1.188 is valid.
[2017-09-22 13:19:28,928] DEBUG: Sending command info to 192.168.1.188
[2017-09-22 13:19:28,928] DEBUG: 

Thanks!

jkliman, 2017-09-22:
sorry go ahead and close out I just realized this is a duplicate issue

closed 2017-09-22 by jkliman

jneilliii, 2017-09-22:
Just curious, is your TP-LINK smartplug also hardware version 2?