jneilliii/OctoPrint-TPLinkSmartplug #14: Multi Plug support

uwwint, 2017-09-22:
Hi, 

would it be possible to have multi plug support? I am intending to use two plugs. One controlling the printer and one controllling the lights.
Then hook into the event CAPTURE_START and CAPTURE_END, and switch on the light only when a shot is taken.
Inject a JS into the main page, which checks if someone opens the control view (#control_link has class 'active'). Use the visibility api ( https://developer.mozilla.org/en-US/docs/Web/API/Page_Visibility_API?redirectlocale=en-US&redirectslug=DOM%2FUsing_the_Page_Visibility_API ) to determine if someone is actually looking. If yes -> light on. If no -> light off. 

This is the general idea. I know there is 
a) timing issues
b) multi user 
to solve.

jneilliii, 2017-09-22:
That was actually my next feature update I was considering. It will take me a little bit to hack my way through it, but I would suspect an array of plugs with each of the current settings now for each one.  Will have to figure out a way to trigger them independently for gcode processing options, potentially extending the M80 and M81 commands with additional information. Once I've got that base concept working I would then look into hooking into the other events like you mention, capture_start/capture_end. Will have to look into the OctoPrint docs for what events are hookable on the camera side. 

I think the trickiest part is going to be dealing with the UI side. Do you think an independent button for each plug or make the current button a menu pull-down (if there are more than 1 plugs configured) would be better?

uwwint, 2017-09-23:
I believe normal users will not have more than 3 plugs (people also have said that about 64K). Therefore having three icons where the icon can be chosen sounds quicker access, but I am not a UX person :). 

Btw. you can query the name of the plug by sending the info command. The response name will be in data['system']['get_sysinfo']['alias']

Lights are already in GCODE:
M355

Try to implement a use case rather than generic support for multiple plugs. Will make your life easier :)

jneilliii, 2017-09-23:
I've got the basic interface working with a rudimentary editable grid in settings and multiple plug toggles in the navbar.  Going to work on getting the ui interactions tied back to the separate plugs.  As it stands now every button loaded is still just controlling the previously configured plug.  Once I get that working, I'll expand the grid for additional settings for each plug to match previous settings and pretty up the grid.

jneilliii, 2017-09-27:
I've got a lot of the core functionality (turning on and off by button and updating color).  I'll be working on the additional features like associated delay timing, system commands, etc. over the next week as I can.

uwwint, 2017-10-02:
Thanks! I am on vacation right now, but will test it as soon as i get back home.

jneilliii, 2017-10-07:
Everything but gcode processing has been incorporated into the dev branch.  Small caveat, you have to delete all the tplinsksmartplug plugin settings from config.yaml otherwise errors occur for missing settings on existing plugs (if on previous dev branch).  Will future proof that prior to release.  Feedback welcome. Main issue right now is newly added plugs don't get current status on save for some reason, so to get around that add the plug and then refresh the browser.

jneilliii, 2017-10-07:
gcode processing incorporated now.  Need to think about the logic behind a timer function for the gcode hook for #15.

jneilliii, 2017-10-16:
Just wanted to see if anyone has been able to test the mult-plug version in the dev branch yet?  It seems to be working for me ok, just wanted to get some feedback from others prior to release.  I've been able to do both hostname and ip to the same plug in order to verify functionality as well as a second plug altogether and everything seems to be working ok.  The interesting thing about the hostname and ip to the same plug allows for different settings for each.

jneilliii, 2017-10-22:
Version 0.7.0 released that incorporates multi-plug support along with a couple of other bug fixes related to using ip address and the warning prompt together.

closed 2017-10-22 by jneilliii