jneilliii/OctoPrint-TPLinkSmartplug #15: Timer Feature Request

dstinebaugh, 2017-09-23:
First off, **LOVE** this plugin! **Great work!**
My thought is that after a print is finished, the M81 is sent and your plugin captures it, but instead of powering off right then, to give it a few minutes to let the fan cool the hotend down to a safe level before powering off, so a simple sleep X then M81 is sent.
I've tried using M116 and M109 with a low temp value, which it does execute, but then goes to the next step without waiting and powers off anyways so I figured a sleep option on the power off from your script would be a good solution.

Again Thanks for your constant work on this!

jneilliii, 2017-09-23:
Will consider this for future updates.  Don't see a reason not to, similar to the connect delay for auto connection after power on.

jneilliii, 2017-10-07:
I think I've got the timers working now in my dev branch with gcode enabled.  It gives the options to set delays prior to sending the on/off command with the GCODE commands.  In order to work with the multi-plug options you'll need to change your M81 command to M81 <plug ip or hostname as put in settings>.  For example, if you have a plug with hostname "HS100" configured in tplinksmartplug settings you would use the gcode command "M81 HS100". This gives the ability to control different plugs independently of one another via gcode.

jneilliii, 2017-10-07:
Yes, that's true.  Only issue I've seen is when adding new plugs the initial state doesn't want to load.  I made a slight change so that when the icon is gray you can click it to check status and get the button functional.

jneilliii, 2017-10-22:
Just fixed an issue when using ip address and the warning prompt.  Incorporated into the master branch and released as 0.7.0 along with these changes for timing control.  Feedback welcome.

closed 2017-10-22 by jneilliii