jneilliii/OctoPrint-TPLinkSmartplug #16: Disconnect on power off

tcm0116, 2017-09-26:
Would it be possible to add the same "Disconnect on power off" setting as is available in the PSU Control plugin? The control board on my printer stays powered on via the USB connection to the RaspberryPi, and the current behavior of this plugin causes the control board to reset when a print is started because OctoPrint has to establish a new connection. The "Power On Delay" setting could also be useful for people that enable the "Disconnect on power off" setting to allow time for the printer to boot up.

jneilliii, 2017-09-26:
I'm not sure I understand the request,  This plugin does have disconnect on power off. It's listed in settings as disconnect before power off, as it makes more sense to me to disconnect prior to powering off the printer.  I am looking to integrate a delay in the future multi-plug support option that will put a pause between the disconnect command and turning off the plug.

tcm0116, 2017-09-27:
I completely missed the On and Off Options tabs, which have the settings in them. Sorry about that.

closed 2017-09-27 by tcm0116