jneilliii/OctoPrint-TPLinkSmartplug #17: HS110 energy monitoring

3axis, 2017-10-17:
For the extra £5 I got the energy monitoring version, it would be nice if the plug-in could be extended to use this feature.

spaelectronics, 2017-11-09:
I agree with this. This would be very helpful!

jneilliii, 2017-11-10:
I'm not familiar with that model and what those other features are, but what kind of information would you guys like to see and where?

3axis, 2017-11-10:
I will take a look tonight and put some info here and pictures if I can.

3axis, 2017-11-10:
The Kasa app shows a summary of todays usage in kWh and runtime in hours. All to 2 decimal places.

Clicking on the  > arrow for usage gives:
Today: Current power W, Total Consumption kWh
Past 7 days: Daily Average kWh, Total Consumption kWh
Past 30 days: Daily Average kWh, Total Consumption kWh

Clicking on the  > arrow for runtime gives:
Today: Current Runtime hours, Total Runtime hours
Past 7 days: Daily Average hours, Total Runtime hours
Past 30 days: Daily Average hours, Total Runtime hours

A stats page showing this could be useful. 

Options (tick box on stats page) to shown one or more of these pairs on the top bar would be good. 

Perhaps default (ticked) to Total Consumption kWh and Total Runtime hours pair showing in the top bar, for example "Total: 2.55 kWH, 6.5 hrs".

adammhaile, 2017-12-16:
+1 - Have that model too and this would be great.

jneilliii, 2018-08-29:
Sorry guys, been busy with work.  I'll probably have time this weekend to work on this, but because I don't have that model will lead on you heavily for testing.  Will post here with links when ready.

jneilliii, 2018-08-29:
I have an initial framework that I need to get one of you to install and send me back the plugin_tplinksmartplug_debug.log file so I can check on how the response comes back from the plug. Make sure that debug logging is enabled in the plugin settings first and then install using the url below in plugin manager and once OctoPrint reloads it should put the information I need in that file. I'm going to start with the realtime statistics, as that seems to be the most useful information initially and will potentially expand that later for the monthly and yearly stats.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.7.zip

KissmanCZ, 2018-08-29:
Ok ;) wait i second, going to install ;)

KissmanCZ, 2018-08-29:
[plugin_tplinksmartplug_debug(1).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2333346/plugin_tplinksmartplug_debug.1.log)

jneilliii, 2018-08-29:
Perfect, thanks.  This will help a lot.  I'll be able to work on this again maybe tomorrow night after work. Just have to figure out where I actually want to add this to the UI.  Navbar seems to make sense, but because it's possible to have more than one plug, it might make more sense as a pop-up or hover tooltip to the icon to show the data.  

Based on your log this is the pertinent information I think, just have to figure out how to translate it. Looks like your plug has been on for 1 hour with current of 23 mA, what is that voltage reading?

```
{'emeter': {'get_realtime': {'total_wh': 1, 'current_ma': 23, 'power_mw': 0, 'voltage_mv': 247579, 'err_code': 0}}
```

jneilliii, 2018-08-30:
@KissmanCZ, if you don't mind it would be helpful if you were to turn on the polling option and re-post that debug log file after performing a print, just to see how the numbers change over time.  This would verify that the realtime data is fluctuating as expected and possibly help me figure out the math involved and how to display this data.  

I don't think I want to get into full on logging of the realtime data, but eventually I will add in the other query options available through the api calls for power this month, power this year, etc.  as documented [here](https://github.com/softScheck/tplink-smartplug/blob/395c3521000f78bacd0d0cb32b6c6faf825f6c1a/tplink-smarthome-commands.txt#L119) and [here](https://github.com/softScheck/tplink-smartplug/blob/395c3521000f78bacd0d0cb32b6c6faf825f6c1a/tplink-smarthome-commands.txt#L122).

jneilliii, 2018-08-30:
I just noticed your note about not being able to toggle the power with this version.  Nothing I added should have effected that at all, but I could be wrong.  Looking through the log it looks like it's sending the off command and then checks the status and that is saying the plug turns off.  Is it just that the icon never changes to off and stays green? If you refresh the page, does it change the icon?  There was another issue posted that sounded like that with the last release prior to this testing version. They experienced the plug switch and then instantly switch back to the previous state.

KissmanCZ, 2018-08-30:
Voltage will be voltage_mv': 247579 in miliVolts, so 247,579 V. I can cross check values with original TP-Link app.

Regarding power off & on. If socket is already on, I can turn it off with plugin but green icon will rotate indefinitely. When I refresh page, green icon will stop rotating but still will be green (and socket will be off).

If socket is off, I am not able to power on. Only green icon rotating. When I refresh page, icon will not rotate anymore.

KissmanCZ, 2018-08-30:
Going to create log during print :) 20 min.

jneilliii, 2018-08-31:
Just made a slight change to try and thin down the logs to see if I was just recording more than necessary.  Please turn off debug logging and reinstall from the same url as before. With polling enabled it should record just the energy response from the server to the debug log file.

This change should also isolate the energy commands from the on/off commands. I wasn't having the same issue as you were with powering on/off my device so not sure it this change will help in your situation or not, but it should put it back to the way it was before.

KissmanCZ, 2018-08-31:
I have two HS110 in my setup ;)

jneilliii, 2018-09-01:
Ahh, that makes sense now.  I just need to look at the other info then and figure out what the differences mean. I'm assuming one was connected to printer while printing and the other was either off or just not drawing any power?

jneilliii, 2018-09-01:
I just pushed a new update that will request the daily statistics for the current month and add that to the debug log file.  That should be the last little bit I need to flush this out and figure out a path for development. Please reinstall using the 0.9.7 url above and send me back the log.

jneilliii, 2018-09-02:
Just pushed an update to 0.9.7 version that I'll need to get some feedback on.  for now I am just throwing the return data from the energy commands into the title attribute of the button.  This should show the information upon hovering your mouse over the icon.  Since I don't have a HS110 the return value is an error message.

[image]

If you guys could test and send me back the debug.log file I'll be able to push this new feature forward I think.

jneilliii, 2018-09-02:
Just pushed a small update to use bootstrap tooltip for the data display.

[image]

KissmanCZ, 2018-09-02:
This update deleted my second plug from plugin :( and reseted all setting for the first plug... and I am still not able to turn plug on. Status polling also not working. Only grey icon rotating now.
[plugin_tplinksmartplug_debug(3).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2343175/plugin_tplinksmartplug_debug.3.log)

jneilliii, 2018-09-02:
I forgot to mention, it will actually wipe out the settings for all plugs, you have to reenter them for both plugs.

KissmanCZ, 2018-09-02:
No errors in gui (only lost ability to power on/off plug - gray icon rotating all time). In octoprint log lot of errors:

`KeyError: 'err_code'
2018-09-02 11:47:59,410 - tornado.access - ERROR - 500 POST /api/plugin/tplinksmartplug (::ffff:192.168.99.254) 93.51ms
2018-09-02 11:48:59,550 - octoprint - ERROR - Exception on /api/plugin/tplinksmartplug [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1140, in decorated_view
    return flask_login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 99, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 166, in on_api_command
    self.check_status("{ip}".format(**data))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 137, in check_status
    if response["emeter"]["err_code"] != 0:`

jneilliii, 2018-09-02:
Perfect, that's what I needed to see...I just pushed an update that should now fix the on/off issue. Please reinstall and let me know.

KissmanCZ, 2018-09-09:
No, still not working :(

jneilliii, 2018-09-11:
Ok, hopefully I got it now.  Turn off the polling option in settings and then reinstall from the 0.9.7 url just like before.

physguyk, 2018-10-14:
Did you ever get your hands on the HS110 to play with for testing this?  If not, do you have a patreon or paypal or google wallet or whatever cool kids are using these days for money transfers, such that I could send you the money for one, and maybe some extra to pay for the time you're putting into this?  It's a very handy plugin, so I'd like to support its development.

horendus, 2018-10-30:
Hi, did you receive a HS110 yet? Im really looking forward to having energy monitoring  for these plugs showing in octoprint and ultimatly, have it logged/icluded in the Print History plugin. 

Are you still working on this?

jneilliii, 2018-10-30:
Nope haven't gotten one yet.

jneilliii, 2018-10-30:
But I did just get the funds to get one, so as soon as I order and it gets here I'll start looking into it.

horendus, 2018-10-30:
Sounds good. Ill put through a small donation to help spur you along. 
What are you thoughts on having it log energy usage of a print? It looks like what you have played around so far with is getting it to show the current power draw but the real end goal here would be showing energy usage of a total print (as well as adding the current power draw to the temp plotted chart. Do you know yet what other emeter values are returned when you poll the plug?  Does it have a running total? That would make it very easy to figure out the power usage of a print by recording the value before and then after the print.

jneilliii, 2018-10-30:
I haven't really put much thought into logging yet, just getting the values to display somewhere will be my first goal. What are your thoughts on where that should be? My initial testing was with the title attribute of the button.

jneilliii, 2018-10-30:
I've ordered the plug, should be here on Thursday and I'll be able to start work on the enhancements this weekend.  I'll see about some form of logging during print, never done that before, but I assume I should be able to accomplish that with some form of yaml storage provider.

horendus, 2018-10-31:
Agreed, getting the current/live emeter values displaying in the UI is the first logical step. I haven't done a deep dive on the ocotoprint plugin API (yet) but it would be good to inject it into the STATE display on the left hand side of the UI. Iv seen other plugin add extra items to this display so it must be achievable. 

In terms of displaying logging, having it display in a running list in the TPLINK Plugin setting menu would be a good start but injecting the final value into the "Print History" plugin (https://plugins.octoprint.org/plugins/printhistory/) would be the ultimate solution rather than trying to implement a whole new Print History menu.

horendus, 2018-10-31:
The energy usage stats of a print would be best pulled directly from the TPLINK plug as this is already a baked in function and trying to recalculate within octoprint based on reading the live stats could introduced issues with accuracy and potentially allot of work.

Looking at the TPLINK Smart Plug API there is an option called .eraseStats. 

I would be inclined to suggest doing an .eraseStats at the start of a print and then at the end of a print pulling the .getMonthStats. I wouldn't recommend pulling the .getDayStats as you would be limiting the reading to a maximum of 24 hours and you also run the risk of the day rolling over mid print which would throw the final calculation out.

jneilliii, 2018-10-31:
Oh yeah. Doing that in the state panel does make a ton of sense. Would probably configure updates using the print progress mixin which would poll the current status at each percent increase. 

Getting the integration incorporated into the print history plugin may be tricky as it appears to be utilizing a sqllite database backend and there potentially could be locking issues if trying to update at the same time, etc. but I'll take a deeper look into that code once I get the first part working right. It may eventually require a collaboration with that plugin's author.

horendus, 2018-10-31:
Yes it looks like the File History plugin would need to be modified in a few ways

Add an extra field into its database for energy usage
Add extra html column for displaying the energy usage value
Modifying the database write at the end of the print to include the energy usage 
Modifying the Table Display to include fetching the energy usage value 

Perhaps the prints energy usage can be written to a temp location (before the Print History database entry is made) which is then referenced by the Print History plugin when writing the database entry?

Im not sure how you would make sure this value is written BEFORE the Print History database write though. Thoughts?

jneilliii, 2018-11-02:
That's why I think integration is going to be tricky.  Anyway, I've got my 2 pack of plugs in and powered one up and connected and my dev branch is pulling the data exactly like I thought it would, just now have to update the code to put the information into the state sidebar panel.

[image]

jneilliii, 2018-11-03:
Got it working well in the 0.9.7 branch.  Will probably make this an official release after some more testing before moving on to the logging stuff.

I haven't integrated the print progress approach yet either, but the polling option is updating in the sidebar state panel. Let me know what you guys think about the layout?

[image]

jneilliii, 2018-11-03:
Just released version 0.9.7 that incorporates the print progress methods. Will keep this issue/request open to deal with the logging aspect of the request.

KissmanCZ, 2018-11-04:
I am not receiving values from my second plug :( and layout is not best... i am missing horizontal divider betwen plugin and % bar.

[image]

jneilliii, 2018-11-04:
I was seeing that when I first set-up the plugs, but after a couple of off/on toggles it finally got caught up and was in sync.  I did take the divider out between the last plug and progress bar.  I was just setting it up to be the same as default state panel.  I agree with you though, adding the divider back in at the end does look better.

KissmanCZ, 2018-11-04:
And rounded values should be great I think.

KissmanCZ, 2018-11-04:
And maybe you can put entire your data into separate box :) under files box.

jneilliii, 2018-11-04:
Putting the info in its own side panel would have been a lot easier than forcing it into the state panel. I'll probably make that change.

jneilliii, 2018-11-04:
Along with the rounding.

jneilliii, 2018-11-04:
How many decimals would you round to, two?

KissmanCZ, 2018-11-04:
2 or 1 should be ok ;) maybe together with own side panel, you can create tab with some kind of statistic?

viljamip, 2018-11-04:
[image]
I can't get this to work. I tried turning the printer on/off several times with the UI. I even uninstalled and reinstalled this plugin. I can see the readings from my plug by running the hs100.sh emeter script so it should be working.

jneilliii, 2018-11-04:
@KissmanCZ: I think once I get into logging I'll add additional features like graphing the data, etc., but that's going to be a while before I get to it.

@viljamip: Please enable debug logging and upload your plugin_tplinksmartplug_debug and octoprint log files.

viljamip, 2018-11-04:
Hopefully these will help you. I deleted the logs, enabled logging, restarted octoprint, turned the printer on, waited a minute, turned the printer off, waited a minute. The power stats are getting to the log just fine but I have never seen them working in the UI.
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2546153/plugin_tplinksmartplug_debug.log)
[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2546155/octoprint.log)

jneilliii, 2018-11-04:
@KissmanCZ: How is this? I need to either work out a way to show only the sidebar panel when there are energy monitoring devices only or move the navbar buttons into the side panel.  I'd personally prefer the former because I don't like the idea of a single button in a sidebar panel for those users that only have one plug.

[image]

@viljamip, the logs don't really show any errors, but I am working on some additional changes that should help in the logic.  In the meantime, if you install version 0.9.6 you should be able to get the plugin functional again. Use the following url in Plugin Manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.6.zip

KissmanCZ, 2018-11-04:
Look good... just implement logic which remove divider after last plug in the box. And center vertically divider between plugs. And yeah, showing panel only for plugs with power usage readings is OK.

jneilliii, 2018-11-04:
Tackled the border issue etc. using pure css approach.  

[image]

I've updated the logic for the on/off commands too, which might resolve the issue for you @viljamip.  If you guys don't mind being my guinea pigs you could install this version using the following URL in plugin manager. Let me know if it works out for you.  This version should also not wipe out your previous settings, but it will clear out the previous emeter readings, but that should re-update once the plugs' statuses are checked. So far it's working for me and my HS100 and (2) HS110 devices.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.8.zip

KissmanCZ, 2018-11-04:
Hmmm.... turn on/off action is now very fast :) perfect. Still no readings from my second HS110 :(

viljamip, 2018-11-04:
I tried 0.9.8 but I still can't see the readings in the UI. On/Off functionality is working nicely.

The new UI panel was so low in the UI that I had to scroll down to see it. Can I rearrange the UI somehow or could you put it maybe below the State panel and above the Files?

jneilliii, 2018-11-05:
There's an unpublished plugin called OctoPrint-SidebarOrder in github that was a fork of my tab order one that will let you rearrange the panels. Otherwise, you can just minimize the other panels to bring it up. 

As for the not updating part if you open the browser's console are there any errors?

KissmanCZ, 2018-11-05:
`Uncaught TypeError: Unable to process binding "if: function(){return emeter.get_realtime }"
Message: Unable to process binding "text: function(){return emeter.get_realtime.power().toFixed(2) }"
Message: emeter.get_realtime.power is not a function
    at text (eval at parseBindingsString (packed_libs.js?331a1bb5:11374), <anonymous>:3:84)
    at update (packed_libs.js?331a1bb5:11404)
    at function.a.B.i (packed_libs.js?331a1bb5:11378)
    at Function.Pc (packed_libs.js?331a1bb5:11357)
    at Function.Qc (packed_libs.js?331a1bb5:11357)
    at Function.aa (packed_libs.js?331a1bb5:11356)
    at Object.a.m.a.B (packed_libs.js?331a1bb5:11355)
    at packed_libs.js?331a1bb5:11378
    at Object.q (packed_libs.js?331a1bb5:11317)
    at m (packed_libs.js?331a1bb5:11377)`

viljamip, 2018-11-05:
This is what I see in my console:
`packed_libs.js?331a1bb5:11450 Uncaught TypeError: Unable to process binding "if: function(){return emeter.get_realtime }"
Message: Unable to process binding "text: function(){return emeter.get_realtime.power().toFixed(2) }"
Message: emeter.get_realtime.power is not a function
    at text (eval at parseBindingsString (packed_libs.js?331a1bb5:11374), <anonymous>:3:84)
    at update (packed_libs.js?331a1bb5:11404)
    at function.a.B.i (packed_libs.js?331a1bb5:11378)
    at Function.Pc (packed_libs.js?331a1bb5:11357)
    at Function.Qc (packed_libs.js?331a1bb5:11357)
    at Function.aa (packed_libs.js?331a1bb5:11356)
    at Object.a.m.a.B (packed_libs.js?331a1bb5:11355)
    at packed_libs.js?331a1bb5:11378
    at Object.q (packed_libs.js?331a1bb5:11317)
    at m (packed_libs.js?331a1bb5:11377)`

viljamip, 2018-11-05:
> `Uncaught TypeError: Unable to process binding "if: function(){return emeter.get_realtime }" Message: Unable to process binding "text: function(){return emeter.get_realtime.power().toFixed(2) }" [...]

I get this as well.

[29 more replies]

PR #101 (2019-05-01): 0.9.16
- Change socket receive to grab 1024 chunks until full response is received.
- Added strip support. Resolves #81.
- Added energy data logging and graph support via new tab. Resolves #17.
- Added temperature runaway protection. Resolves #65.
PR #107 (2019-05-11): 0.9.16
### Added
- Added HS107 and HS300 multiple plug support, resolves #81.
- Energy data logging and graphing via new tab, resolves #17.
- Temperature runaway protection, resolves #99.
### Fixed
- Change socket receive to grab 1024 chunks until full response is received for larger responses.
- Fixed settings migration issue, resolves #102.
- CSS changes to integrate better with TouchUI, potentially resolves #79.
- Status checks on settings save if plug list changes in anyway.
- Status checks no longer saves the settings in favor of real-time updates to current state, resolves #93.
closed 2019-05-01 by jneilliii