jneilliii/OctoPrint-TPLinkSmartplug #18: MQ2 Smoke Sensor Integration

nrepian, 2017-10-17:
Thank you for coding this excellent plugin - it makes managing the printer so much easier!

I wondered if you would consider adding a small extension to the script, to allow for integration with the extremely popular MQ2 gas/smoke sensor module (they're less then £1/$1 to buy and designed to work with RPi/Arduino).

It should be possible to use one of the GPIO pins to detect the smoke trigger level (via a pot on the MQ2 module), which could in turn disable power to the printer via the TP-Link SmartPlug. Essentially, using it as a backup mechanism for cutting power to the printer if smoke is detected, a very useful feature.

According to this page (https://www.sunfounder.com/learn/Sensor-Kit-v2-0-for-Arduino/lesson-10-flammable-gas-detection-sensor-kit-v2-0-for-arduino.html), d0 on the MQ2 sensor drops goes low once the smoke threshold is reached, so I hope this would be a simple modification to integrate :).

naggie, 2017-10-22:
I would suggest just using the REST API that this plugin exposes, the script does not have to be part of this. I've ordered the module, and will write such a script when I get time (project overload atm, but it should be a quick one)

nrepian, 2017-10-22:
Thanks, I'll check that out! Looking forward to seeing how you find the MQ2 for this :)

qbone, 2019-02-28:
I would also love to see this integrated! :)

jneilliii, 2021-01-27:
So as @naggie has mentioned this could be accomplished with an external script that can send the necessary API commands to the plugin to do this or possibly through other external means like Home Assistant. I don't really want to get into GPIO programming for the plugin directly as that will just add too much complexity and I'm trying to minimize feature creep. Hopefully @naggie was able to create the script and could potentially share it here for others to utilize. I hope you all understand.

Additional information on the API exposed by the plugin can be found [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/wiki/API).

closed 2021-01-27 by jneilliii

naggie, 2021-01-27:
Sorry I haven't had time.