jneilliii/OctoPrint-TPLinkSmartplug #19: Unable to load octoprint with 0.7.0

naggie, 2017-10-22:
Hi, 

I upgraded via the auto-update notification within octoprint. After the restart, the interface failed to load. I've attached a screenshot showing the chrome debugger detailing the exception.

[image]

Thanks!
Cal

naggie, 2017-10-22:
I recovered from this by manually removing the code in the js plugin file and restarting the octoprint server before removing the plugin. I happened to notice that the IP address of the tplink switch was no longer present in the plugin settings before I removed the plugin.

jneilliii, 2017-10-22:
Thanks.  The issue is related to adding the additional warn while printing option to the array I think.  I thought that the on_settings_migrate would resolve that using versioning, but I guess not completely.  It apparently sees the code as you noticed it wiped out the settings, but I guess that happens after binding, which is where the error occurs.  I would suspect, after you got the error restarting the octoprint service would have resolved itself.

marcbritten, 2017-10-22:
restarting the service does not help

jneilliii, 2017-10-22:
I'm unable to re-create this issue.  Maybe it's because I manually deleted all the plugin settings initially before an upgrade.  If anyone is able to do that, here are the steps...ssh to your pi and run these commands.

```
cd ~/.octoprint
nano config.yaml
delete the arrSmartplugs setting completely from the tplinksmartplug section.
ctrl+x
y to save
sudo service octoprint restart
```

quidan, 2017-10-22:
@jneilliii Your steps to resolve the issue worked for me. Thank you!

naggie, 2017-10-22:
That worked, thanks. For reference, here's what it was:

[image]

mikeap, 2017-10-22:
No success - still can't login - when I remove lines 99-102 above

mikeap, 2017-10-22:
How can I back-out tplinksmartplug if I can't login - can ssh

jneilliii, 2017-10-22:
Try the same steps, but remove the entire tplinksmartplug section from plugins in config.yaml.

mikeap, 2017-10-22:
Tried a clean install of octoprint /w 1.3.5 and then added tplinksmartplug - login now fails. Suggest you back-out 0.7.0 until you have a fix

jneilliii, 2017-10-22:
Can you send me the octoprint.log file please so I can see if there are any errors.  I've uninstalled, reinstalled, downgraded, upgraded, and can't reproduce the issue anymore.  I even just released 0.7.1 to see if my logic was flawed with how it was set-up for resetting the settings.  They now reset on upgrade to defaults instead of setting the plugs to null.

mikeap, 2017-10-22:
This maybe messed up - grep'd tplink from the upgrade from 0.5.0 to 0.7.0 octoprint.log

But as I said. I also tried a clean install of octopi, installed 1.5.3 octoprint, logged-in successfully. Installed the tplink 0.7.0 plugin and was unable to login

Below is my initial attempt to upgrade from 0.5.0 to 0.7.0 and first saw the failure on login

|  TP-Link Smartplug (0.5.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
2017-10-22 11:37:43,315 - octoprint.plugins.softwareupdate - INFO - Starting update of tplinksmartplug to 0.7.0...
2017-10-22 11:38:53,101 - octoprint.plugins.softwareupdate - INFO - Update of tplinksmartplug to 0.7.0 successful!
|  TP-Link Smartplug (0.7.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
2017-10-22 11:39:38,557 - octoprint.plugins.tplinksmartplug - INFO - TPLinkSmartplug loaded!
|  TP-Link Smartplug (0.7.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
2017-10-22 11:48:30,716 - octoprint.plugins.tplinksmartplug - INFO - TPLinkSmartplug loaded!
|  TP-Link Smartplug (0.7.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
2017-10-22 11:53:30,001 - octoprint.plugins.tplinksmartplug - INFO - TPLinkSmartplug loaded!
|  TP-Link Smartplug (0.7.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
2017-10-22 12:02:44,309 - octoprint.plugins.tplinksmartplug - INFO - TPLinkSmartplug loaded!
|  TP-Link Smartplug (0.7.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
2017-10-22 12:19:37,841 - octoprint.plugins.tplinksmartplug - INFO - TPLinkSmartplug loaded!
|  TP-Link Smartplug (0.7.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
2017-10-22 12:31:37,393 - octoprint.plugins.tplinksmartplug - INFO - TPLinkSmartplug loaded!
|  TP-Link Smartplug (0.7.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
2017-10-22 12:49:09,536 - octoprint.plugins.tplinksmartplug - INFO - TPLinkSmartplug loaded!
|  TP-Link Smartplug (0.7.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
2017-10-22 12:55:30,574 - octoprint.plugins.tplinksmartplug - INFO - TPLinkSmartplug loaded!
|  TP-Link Smartplug (0.7.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
2017-10-22 15:12:13,599 - octoprint.plugins.tplinksmartplug - INFO - TPLinkSmartplug loaded!
~

rob49152, 2017-10-23:
Yes .70 completely fubared login by not loading the config.yaml file. My only solution was to manually remove the plugin and it worked. I could log in again.

jneilliii, 2017-10-23:
I'm not able to re-create this issue at all, maybe my 0.7.1 release does actually fix the upgrade issue.  I did realize my case may be different because I disable accessControl and handle authentication outside of OctoPrint via haproxy configs.  So I removed the plugin, re-enabled accessControl, deleted the user.yaml and config.yaml and restarted OctoPrint to mimic a fresh install.  Went through start up wizard, creating account, etc.  Logged out, logged in to verify login worked without plugin.  Installed 0.5.0, which restarts OctoPrint again, reload page when it comes back up, logged in, set the tplinkmartplug setting for my plug, verified it worked.  Logged out, logged in, no issue.  Updated to 0.7.1, which restarts OctoPrint again, logged in, setup 2 plugs in tplinksmartplug settings, verified they worked. Logged out, logged in, no issue.  Reboot system, reload page when prompted, was auto logged in from previous chrome connection I assume.  logged out, logged in, no issue.  Opened Chrome in incognito mode and loaded page, logged in, no issue.  

Can you guys test 0.7.1 for me please and report back?

mikeap, 2017-10-23:
How can I remove this plugin - Can only ssh to octoprint?

jneilliii, 2017-10-23:
If you used an octopi install you can use these commands after ssh
connection is established.

cd oprint/bin
./pip uninstall OctoPrint-TPLinkSmartplug -y
sudo service octoprint restart

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

mikeap, 2017-10-23:
no success with 0.7.1 - won't login - if I uninstall 0.7.1 will regain login

naggie, 2017-10-23:
> I did realize my case may be different because I disable accessControl and handle authentication outside of OctoPrint via haproxy configs.

As do I, but with [nginx](https://github.com/naggie/nginx-https-base). This explains why my install is now working.

3axis, 2017-10-23:
I had the same issue, could not log in. 

Deleting the plug from the config file completely (lines 98-102 inclusive) as shown above between uninstalling the plugin and reinstalling worked. It may work if you just delete the array (line 100).

I have simplified what I did over ssh slightly, it should work, if not just do a service restart (4) after each step.

#-------------------------------------------------------------------------------------------------------------------------------
# 1) uninstall
~/oprint/bin/pip uninstall OctoPrint-TPLinkSmartplug -y

# 2) edit config file
nano ~/.octoprint/config.yaml
# delete the tplinksmartplug section (remember the IP address for setting up again)
# ctrl+x then 'y' to save

# 3) install latest (0.7.1)
~/oprint/bin/pip install "https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/master.zip"

# 4) restart octoprint service
sudo service octoprint restart
#-------------------------------------------------------------------------------------------------------------------------------

I can log in fine with 0.7.1 and it works as expected having re-setup the plug.

May be someone can post the simplest once they find out what it is to save others effort.

foosel, 2017-10-24:
@jneilliii I keep getting reports on this left and right. This is starting to become a serious problem. I'd like to put out a notification about this plugin to refrain from updating OctoPrint to 1.3.5 (since that seems to be the problematic combination) until this is definitely fixed on your end, but I need to know which versions are affected for that and currently that's completely unclear to me. Could you help me here?

I'd also like to understand what exactly you did in your plugin that managed to entirely nuke the javascript execution, so I can make OctoPrint a bit more error resilient there if possible to prevent something like this from happening again :)

foosel, 2017-10-24:
@jneilliii so it looks like 0.7.0 and 0.7.1 are both affected - fresh install, no config at all and I get bombarded with exceptions. Before you do anything else I *strongly* suggest to set both of those releases to "prerelease" on Github: 

[image]

That way OctoPrint will not prompt people to update. I've also disabled the plugin on the Plugin Manager for now so that people don't install the also broken version 0.7.1 and will also shortly push out notices for versions < 0.7.0 to tell people to not update to 0.7.0 or 0.7.1.

jneilliii, 2017-10-24:
Thanks so much Gina, pull request merged.  I'm curious why the 0.7.0 release had an issue, because with that one I wasn't using the get_defaults function and was just setting the value to null, which was having a similar effect?

foosel, 2017-10-25:
@jneilliii Without having taking a closer look at the sources, setting it to null was probably the issue. The problem arose when you attempted to read the length of an array that wasn't returned by your plugin's settings, making it null and therefore making the length request run into an exception. That would have been caught (and in fact was also caught) on binding, but you also did it in onAfterBinding and there OctoPrint was still lacking some try-catch-block to protect against errors - that's what I changed yesterday ;)

In general, you always want to ensure something you are trying to access that comes from an API is actually set - it might not be after all, even if you are fairly sure it should. I use this kind of defensiveness a lot to protect against my own mistakes of that kind (or weird version issues). Something like "if (object && object.property && object.property.length)" would have prevented this I think.

But again, I admittedly haven't looked a lot at your code, so I might be mistaken.

jneilliii, 2017-10-25:
Release 0.7.2 has been released that should in theory resolve these login issues.  Those of you that were having issues, please give this one a try and let me know how it goes.

mikeap, 2017-10-26:
Success with 0.7.2 - am able to login - plugin fully functional

jneilliii, 2017-10-27:
Thanks for the confirmation, I apologize again for the issue.

closed 2017-10-27 by jneilliii