jneilliii/OctoPrint-TPLinkSmartplug #21: M81 not triggering power off

hutchyx, 2017-10-23:
Using OctoPrint 1.3.5 and TP-Link Smartplug 0.7.1.  
I am able to turn the TP-Link plug on and off from the icon in the octoprint nav bar but i am unable to shut off the printer using the M81 command.  I do have GCODE checkbox checked in the settings.

jneilliii, 2017-10-24:
In the latest version you have to actually use M81 followed by the ip/hostname for the plug you add.  This was to be able to support multiple plugs.  So if your plug is ip 192.168.1.109 the command would be 

`M81 192.168.1.109`

Please let me know this works for you so I can close this issue.

hutchyx, 2017-10-24:
Thank you... This works perfectly. 
I apologize, but I hadn't seen it documented anywhere.
Thanks for adding the multiple plug support, this is great for me!

jneilliii, 2017-10-24:
Thanks for reporting back.

closed 2017-10-24 by jneilliii