jneilliii/OctoPrint-TPLinkSmartplug #23: Does not turn off after receiving M81 command

ATwig, 2017-10-25:
Does not turn off smart plug after receiving an M81 command.

I use Simplify3D for creating my GCode. I have edited the process post-print script to add an M81 command as the last command in the GCode.

When a print ends all of the other post-print commands are followed [Move the head out of the way and slide the surface out].

This was not an issue prior to 0.7

Here's the log from a 6 hour print. I use the icon in the OctoPrint nav-bar to turn on the printer. I set the print before I went to bed and woke up about two hours after the job finished. 

```
[2017-10-25 01:25:31,383] DEBUG: Turning on 192.168.1.14.
[2017-10-25 01:25:31,384] DEBUG: IP 192.168.1.14 is valid.
[2017-10-25 01:25:31,471] DEBUG: Sending command on to 192.168.1.14
[2017-10-25 01:25:31,472] DEBUG: «  -ý"system":{"set_relay_state":{"err_code":0}}}
[2017-10-25 01:25:31,473] DEBUG: Checking status of 192.168.1.14.
[2017-10-25 01:25:31,473] DEBUG: IP 192.168.1.14 is valid.
[2017-10-25 01:25:31,484] DEBUG: Sending command info to 192.168.1.14
[2017-10-25 01:25:31,487] DEBUG: « :è"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.2.2 Build 170907 Rel.164228","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:2D:D4:07","deviceId":"80068CF12EC51B7F0AFDC883B5B47EE4180EC128","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"00000000000000000000000000000000","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"3D Printer","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":1,"on_time":0,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-66,"led_off":0,"latitude":[snip],"longitude":[snip]}}}
[2017-10-25 01:25:31,515] DEBUG: Running system command .
[2017-10-25 01:25:41,378] DEBUG: Connecting printer.
[2017-10-25 08:11:17,237] DEBUG: Received M81 command, attempting power off.

~~End Of File~~
```

The Nav-Bar lightning bolt still functions correctly to turn the printer on and off. 

Here are my settings:
[image]

3axis, 2017-10-25:
Assuming that the intention of the "system command off" filed is to tell the plugin what to look for from the printer in order to automatically turn the plug off, don't you need to put M81 in the "system command off" box? 

I am not sure what the relationship is between the GCODE delays, connect/disconnect delays and system commands and delays. Please can someone document these in the Readme.md?

Is there a use case where one would want to send commands to the printer after it has powered on and stabilised? Does any of the settings cater for that?

Bazeone, 2017-10-25:
I believe the system command options is for running commands/scripts on the octoprint server when the printer is powered on or off. I'm having the same issues of the printer not powering off when sent the M81 Gcode.

ATwig, 2017-10-25:
In the previous 0.6.x versions it triggered off the M80/81 commands (ATX Power On/Off). If you look at the __init__.py file at line 206 you can trace where it hooks onto the M81 the command and attempts to process it. 

I just think its missing the bit to actually send the "off" command to the plug if you go down that path. I'm gonna play with it at home to see if I can get it to work how I'm expecting it to. I filed the bug before running off to work when I found my printer on this morning

Bazeone, 2017-10-25:
it's was working fine before the update.

3axis, 2017-10-25:
Sorry for my misunderstanding, clear documentation with examples would help understand how to setup. I have left them un-configured because I was not sure what they did.

ATwig, 2017-10-25:
Just realized this is a duplicate of this ticket

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/21

closed 2017-10-25 by ATwig

jneilliii, 2017-10-25:
Sorry guys, your assumptions are correct, system command are for local script files, etc. to run after power off/on.  Enabling gcode does use the M80 and M81 commands, but you have to append the hostname/ip of the plug you want to control in your gcode.  I'll try to document that better in the future.