jneilliii/OctoPrint-TPLinkSmartplug #24: Turnoff Plug After Print Complete?

wizbowes, 2017-10-25:
Hi,

Love this plugin.  No issues, but I was wondering if would be possible to configure this so that the plug turns off after printing is completed.  Either via a timer or when the head gets below a certain temp.

Thanks for the plugin

wizbowes

ttocsneb, 2017-10-26:
One thing that I have done with my setup was to add this to the end of the 'After print job completes' section of 'GCODE Scripts' in the system settings

    G4 P300000; Wait 5 minutes to let the Hot end cool
    M81; The TPLink plugin will interpret this, and turn off the smart plug

jneilliii, 2017-10-26:
That was going to be my suggestion as well. However, with the latest
version you need to issue the command as M81 ip. Example documented on
readme in the gcode section.

jneilliii, 2017-10-27:
This is possible via gcode commands, allowing it to be set in octoprint's settings to be applied for all prints or in am individual basis via end gcode in your slicer application.

closed 2017-10-27 by jneilliii