jneilliii/OctoPrint-TPLinkSmartplug #25: [FeatureRequest] Add Tasmota (Sonoff) support

Xeroxxx, 2017-10-27:
Hello,
do you see any chance to add support for Sonoff devices running Tasmota Firmware?
https://github.com/arendst/Sonoff-Tasmota

Its well documented and controlable via Web.

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

Sonoff devices are cheap devices to control power over Wifi.

Cheers

jneilliii, 2017-10-27:
Not as part of this plugin, but I'll consider writing a separate one.

Xeroxxx, 2017-10-28:
This would be awesome. The Sonoff will really have great use for 3D Printers.

jneilliii, 2017-10-29:
I've ordered one of the S20 plugs from Amazon to give this a go. Will receive it Monday and potentially be able to work on it next week.  Based on reading the Tasmota pages it looks like this will be extremely easy to achieve just using AJAX calls directly to the device itself and won't have to worry about integrating encryption like with the TP-Link devices.  Once I have an initial repository up I will close this "issue".

Xeroxxx, 2017-10-29:
@jneilliii Whaat? You're awesome! Great thing is, that behavior of all devices is the same with tasmota firmware.

If you need any help flashing the firmware, let me know. Hope you're soldering iron is ready if you don't wanna go the OTA Update method :)

Thank you!

jneilliii, 2017-10-29:
Yeah, the only thing I think will be tricky is handling the multiple relay devices as the command has to include a number to identify the relay being controlled.  This isn't a huge deal, just something that will have to be added to the options.

jneilliii, 2017-10-29:
I do plan on at least attempting the OTA update process.

jneilliii, 2017-11-04:
OTA update went great.  

While digging around on the tasmota and octoprint groups I stumbled across someone else doing [Tasmota control](https://groups.google.com/forum/#!searchin/octoprint/mqtt%7Csort:date/octoprint/uJWQCHu1i-Q/LBmmi4OMAwAJ) via the [OctoPrint-MQTT](https://github.com/OctoPrint/OctoPrint-MQTT) plugin, which is probably much more advanced than the method I'm using.  I noticed on that plugin, it actually generates helpers that could be utilized for publishing and subscribing your own plugin messages via MQTT.  

My assumption is most users aren't going to want to get into the nitty gritty of creating a MQTT server, so I hacked through my existing tp-link smartplug code and got it working for the Tasmota flashed sonoff devices.  

It in theory should allow for the control of multiple relays on the same device using the index value in the settings.  I have only tested the most minimal of operations with all settings disabled for my single sonoff S20, meaning powering it on and off from the button.  Seems to be working for me.  Please post any issues you may have on the [OctoPrint-Tasomota](https://github.com/jneilliii/OctoPrint-Tasmota) repository.

You'll have to install manually via the plugin manager with the below url for now.

[https://github.com/jneilliii/OctoPrint-Tasmota/archive/master.zip](https://github.com/jneilliii/OctoPrint-Tasmota/archive/master.zip)

Xeroxxx, 2017-11-05:
Thank you so much! I'll check it out as soon as I get my Sonoff from China :)

Yes I don't want any MQTT Server, no need for that.

Thanks!!!!

closed 2017-11-06 by jneilliii

Xeroxxx, 2017-11-06:
@jneilliii 

I added a new wiki page in Sonoff-Tasmoata Repo for your plugin (can be found on the right navigation too)

https://github.com/arendst/Sonoff-Tasmota/wiki/Octoprint

Just wanna let you know.