jneilliii/OctoPrint-TPLinkSmartplug #26: Can't connect or connection issue.

slimshizn, 2017-10-30:
I left my logs in a different issue, not sure if that was the right place for it. As title says I'm having connection issues. Was able to connect in a previous version but only after refreshing the tab. 
[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1427103/octoprint.log)
[plugin_tplinksmartplug_debug.log.2017-10-27.zip](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1427104/plugin_tplinksmartplug_debug.log.2017-10-27.zip)

jneilliii, 2017-10-30:
Just commented in the other issue.  If you just upgraded the settings will get wiped out due to changes.  Put your plug settings back in and it should work.

jneilliii, 2017-10-31:
Can you do me a favor and delete all your debug log files, make sure debug logging is enabled still in tplinksmartplug settings, restart octoprint, and send me the updated debug log file to see how it is talking back and forth?

slimshizn, 2017-11-01:
Yes sir, I’ll get  back to you asap.

From: jneilliii
Sent: Tuesday, October 31, 2017 7:36 PM
To: jneilliii/OctoPrint-TPLinkSmartplug
Cc: slimshizn; Author
Subject: Re: [jneilliii/OctoPrint-TPLinkSmartplug] Can't connect or connectionissue. (#26)

Can you do me a favor and delete all your debug log files, make sure debug logging is enabled still in tplinksmartplug settings, restart octoprint, and send me the updated debug log file to see how it is talking back and forth?
—

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

jneilliii, 2017-11-01:
Ok, it seems that the ip address is valid, but for some reason the plug is not responding to the web request, and therefor getting an "unknown" relay state in the plugin. The previous debug log file you sent was showing some responses from the plug on the same ip, but the last couple of requests it did not.  Do you remember what happened/changed between 2017-10-27 17:04 and 2017-10-28 00:32?

slimshizn, 2017-11-01:
Nothing to really take note of. The plug is working with the kasa app just
fine and has been updated to the latest firmware.. Any other steps that you
need me to take?

slimshizn, 2017-11-01:
I could port forward, which ports are good to use for this?

jneilliii, 2017-11-01:
I assume you're verified the ip address.  The only thing I can think of trying from here is testing the original python scripts that I appropriated for the plugin to see if it's returning any information. I just took a quick glance at the pull request on the repository and it looks like someone else may be having the same issue. Forgot I had replied over there already.  Looks like you guys may be using the same plug that isn't working.

https://github.com/softScheck/tplink-smartplug/pull/23#issuecomment-340311893

https://github.com/softScheck/tplink-smartplug/issues/22#issuecomment-340311235

jneilliii, 2017-11-01:
We could possibly test another fork of that project that appears to use a slightly different encrypt/decrypt function, but not sure if that would resolve it or not.  If you're comfortable with ssh to your pi (assuming you're using octopi) you could give that python script a try (linked below).  Appears you have to run a pip install future command prior to trying it though.

https://github.com/mjfwest/tplink-smartplug/blob/master/tplink-smartplug.py

slimshizn, 2017-11-01:
> tplink-smartplug.py

I'll give whatever it needs to work a try, I'll let you know what happens.

slimshizn, 2017-11-01:
Okay I went to check that out and it seems that is only for the other versions of the smart plug.

jneilliii, 2017-11-01:
So you downloaded the python file listed above and tried running the commands and it didn't work?  I don't think they updated the readme in anyway, so not sure if it will work. But the steps would be to ssh to your pi, then run the following commands.

```
wget https://github.com/mjfwest/tplink-smartplug/raw/master/tplink-smartplug.py
chmod +x ./tplink-smartplug.py
./tplink-smartplug.py -t 192.168.0.2 -c info
```
Let me know if it returns anything.  I just tested with my plug and got back the following:

> Sent:      {"system":{"get_sysinfo":{}}}
> Received:  {"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.163650","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:6C:3C:03","deviceId":"80061B3BFFE661ED6835AD8EE4E13F491873688C","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"59542398207847F07D67AB57530532B8","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"TP-LINK-LIVING","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":1,"on_time":680146,"active_mode":"none","feature":"TIM","updating":0,"rssi":-56,"led_off":0,"latitude":33.811649,"longitude":-84.474007}}}

slimshizn, 2017-11-01:
DEBUG: « -ÿ"system":{"get_sysinfo":{"sw_ver":"1.4.0 Build 170703 Rel.142006","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS105(US)","mac":"50:C7:BF:9C:F7:9F","dev_name":"Smart Wi-Fi Plug Mini","alias":"D6","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-39,"led_off":0

I see it's showing relay_state of 0 which your's shows 1, and on time of 0. Weird I'm using it right now.

jneilliii, 2017-11-01:
Cool, based on that response it appears that the decoding function may be screwed, because the return string you put is not a proper json formatted string, and therefore my plugin is unable to interpret it correctly.  It doesn't start with a { character and also looks like it's cut off short. Since I'm not really good at encryption I don't think I'll be able to fix this, but maybe someone like @uwwint would be able to figure out the necessary adjustments.  Just out of pure curousity, can you delete that downloaded python script and run the following and see if it returns the same? That is basically the encode/decode functions that I'm currently using in the plugin.

```
wget https://github.com/uwwint/tplink-smartplug/raw/master/tplink-smartplug.py
chmod +x tplink-smartplug.py
./tplink-smartplug.py -t 192.168.0.2 -c info
```

slimshizn, 2017-11-01:
Yes as soon as I can, that sucks by the way. Hopefully he might know what
to do

jneilliii, 2017-11-01:
I was just reviewing your comments in the other issue and remembered you said it worked in a prior version.  If that's the case, I wonder if the change I did to fix the AU plugs mucked up the HS105. Instead of the above commands let's try uninstalling the plugin, restart octoprint, and installing the older version using the following url in plugin manager.  This was prior to the encryption/decryption function change and multi-plug support.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.4.0.zip

If that works, then let's go up to the next version with this link which had the encrypt/decrypt function change.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.5.0.zip

If that works, then we know the most recent version has messed something up somewhere for sure.

slimshizn, 2017-11-02:
Okay yes I'll do that shortly and report back. Thanks for reading those.

slimshizn, 2017-11-02:
Okay some findings for you. I uninstalled the plugin, rebooted, installed 0.4.0. This did NOT work. I then tried using 0.5.0. It initially kept giving me the IP error. I have two of these plugs, so I tried the other IP for that one. It immediately worked, then I put the PREVIOUS IP in and it worked immediately. Can confirm that 0.5.0 WORKS, but with some odd behavior.

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

slimshizn, 2017-11-02:
Also wanted to include I did try switching IP’s and it did not work for 0.4.0.

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

jneilliii, 2017-11-02:
Yeah, I vaguely remember the issue you describe with the states not being in sync initially with 0.5.0, until you went into settings and saved. The odd part is the encrypt/decrypt function and logic between 0.5.0 and the current version is the same.  The only difference is multi-plug support.  So I guess the next step is to uninstall 0.5.0, completely delete the tplinksmartplug settings from config.yaml to avoid any upgrade issues, restart octoprint, install the latest version, and see where we are at.

slimshizn, 2017-11-02:
I have a couple different printers running OctopPrint, one that has never had TP link on it. I went ahead and installed that on that one and added the other plug. It is greyed out like the original issue began. What’s really odd is this log that came with it.

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

slimshizn, 2017-11-02:
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1438282/plugin_tplinksmartplug_debug.log)

jneilliii, 2017-11-02:
That is extremely odd, as if it's not changing the state of the plug itself.  It's definitely getting a response from that ip address and the relay state does show the plug as on.  Can you send me your tplinksmartpliug settings stored in config.yaml?

slimshizn, 2017-11-02:
Yeah no problem, the printers are running right now so I didn’t want to interrupt that. Might be a few hours.

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

jneilliii, 2017-11-08:
Thanks to @kchiem we may have a fix for the HS105 plugs.  Not sure if this will break the AU ones or not, but you can test by installing the plugin from my dev branch and let me know.  Here is the url to use in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/dev.zip

slimshizn, 2017-11-15:
I can confirm that the recent update to the mini plug's is what is not allowing 7.0+ to work. I have one that I did not upgrade and it works fine with 7.2.

slimshizn, 2017-11-15:
That being said, 7.2 works for the big plugs and 5.0 works for the NEWER update on the mini plugs.

jneilliii, 2017-11-17:
@slimshizn, did you try the dev branch version with the NEWER update mini plug? I just incremented the version on that branch to avoid confusion debugging. I had left it at the previous 0.7.2 it will now install as 0.8.0.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/dev.zip

gumpdy, 2017-11-21:
[2017-11-21 06:46:20,753] DEBUG: Checking status of 192.168.1.74.
[2017-11-21 06:46:20,754] DEBUG: IP 192.168.1.74 is valid.
[2017-11-21 06:46:20,833] DEBUG: Sending command info to 192.168.1.74
[2017-11-21 06:46:20,836] DEBUG: « 9ë"system":{"get_sysinfo":{"sw_ver":"1.2.9 Build 170808 Rel.145916","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS105(US)","mac":"50:C7:BF:BD:F1:CF","dev_name":"Smart Wi-Fi Plug Mini","alias":"3D Printer","relay_state":1,"on_time":5094,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-37,"led_off":0,"longitude_i":-1230097,"latitude_i":449225,"hwId":"E5D7E6089B060EF662783C23AE110522","fwId":"00000000000000000000000000000000","deviceId":"80066F057ACD0BA64E0162DCB326D60018AC73C4","oemId":"003E098AF0D44D4BAB796B3F6A7A830E","err_code":0}}}
[plugin_tplinksmartplug_debug (7).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1490295/plugin_tplinksmartplug_debug.7.log)

jneilliii, 2017-11-21:
@gumpdy, did you try the dev version? From what @slimshizn was saying the newer firmware might be breaking this, and the older version 0.5.0 plugin works for them.

jneilliii, 2017-11-22:
@gumpdy @slimshizn I've just made a quick change on a new branch for testing for you guys.  Install via the plugin manager using the following url.  Once installed and the page has been reloaded go into TPLinkSmartplug Settings and enable classic encryption functions. For the sake of testing, uncheck all boxes like this. If that works, then we can debug which options are breaking the functionality for you.

[image]

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/HS105.zip

jneilliii, 2017-11-25:
@slimshizn, @gumpdy was able to get the plugin working by installing another plugin.  Not sure how that made a difference, but it's working for him/her afterwards.

slimshizn, 2017-11-26:
@jneilliii I tried your 0.8.0, it did not work. It seems that sometimes the setup works and sometimes it does not. (with the old version 5.0)

slimshizn, 2017-11-26:
@jneilliii Which plugin are you talking about that gumpdy used?

jneilliii, 2017-11-26:
I think gumpdy just installed the action command plugin, and afterwards it was working. Not sure why, it shouldn't make a difference. I added a classic encryption setting that will use the same code that 0.5.0 used, which you said was working, but you have to install it from the HS105 branch to test.

jneilliii, 2017-11-26:
and enable that setting.

gumpdy, 2017-11-26:
I'm having issues now. Will install .8 and report back

gumpdy, 2017-11-26:
Ok, I went to print something a few minutes ago and the lightning bolt icon
was red while the plug was in the on position (I turned it on with "OK
GOOGLE") seconds before. When I clicked the icon, nothing happened. I
installed 0.8.0 and same thing. I installed "Action Commands" purely
because it's the first plugin in the plugin manager list and it now works
fine again. I uninstalled Action Commands and it still works fine.

gumpdy, 2017-11-26:
I'm back to where I was the other day... Uninstalled 0.8.0 and went back to
the one available in the octoprint repository and it works fine.

gumpdy, 2017-11-26:
When it breaks again do you want HTTP/SSH Access? I'd need your email if so.

gumpdy, 2017-11-27:
Turning it off with the green button resulted in it turning off but the
button stays green. After more tinkering, turning it on with a red icon
made it turn on but the icon stayed red.I installed another plugin (this
time terminal filter) and now it works just fine.

It appears adding any plugin after installing yours fixes the problem
entirely and restores functionality (at least for now)

gumpdy, 2017-11-27:
After uninstalling action commands, I installed another plugin that didn't
require a restart and the problem mentioned before persisted. Installed
action commands and it is fixed. I have winSCP installed if you want me to
snag any files for debugging.

slimshizn, 2017-11-27:
@jneilliii Got it working, I have it exactly how you have it in that picture.

gumpdy, 2017-11-28:
Had you installed a plugin after installing OctoPrint-TPLinkSmartplug?

jneilliii, 2017-12-05:
So I just recently was having an issue controlling my TPLink Smartplug and no matter what I did the commands were not getting to the plug.  I ended up unplugging the device from the wall and plugging it back in and full functionality was restored.  Not sure if you guys still having issues have tried that or not, but may be something to test.

closed 2018-08-06 by stale[bot]

atlraider, 2018-08-10:
Was there a verified fix for this issue? I have a HS105 (HW1.0,FW1.5.1), and it is giving me the "Status unknown. Double check ip address in TPLinkSmartplug Settings." error. 

Debug logs have this error over and over:

[2018-08-10 00:15:18,713] DEBUG: Could not connect to 192.168.1.245.
[2018-08-10 00:15:18,714] DEBUG: {'system': {'get_sysinfo': {'relay_state': 3}}}

Any help would be greatly appreciated!

jneilliii, 2018-08-10:
Not sure if everyone on this thread got theirs working. I vaguely remember the Australian model may still be an issue. Based on that error you are getting though I believe that your pi (assuming that's what you are using) is unable to communicate with that ip address that you gave it. Another user had that issue and it was because of a home network/routing issue. If you ssh to your pi are you able to ping that ip address?

atlraider, 2018-08-10:
That was it. Thanks!