jneilliii/OctoPrint-TPLinkSmartplug #27: HS100 not turning off at print end.

xkcd42, 2017-11-17:
I think I have the gcode correct with the IP address.  Log validates it ok, but it errors out at shutdown command.   
End of Gcode:
M84                         ;steppers off
G90                         ;absolute positioning
G4 P10000
M81 192.168.0.187
;{profile_string}

Debug log (repeats)
[2017-11-16 16:54:02,837] DEBUG: Checking status of 192.168.0.187.
[2017-11-16 16:54:02,840] DEBUG: IP 192.168.0.187 is valid.
[2017-11-16 16:54:02,949] DEBUG: Sending command info to 192.168.0.187
[2017-11-16 16:54:02,954] DEBUG: « 'õ"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.163650","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"xxxxxxxxxx","deviceId":"8006987341FA98689E31EFE2772EBABC17DC609E","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"59542398207847F07D67AB57530532B8","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"MPSM","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":1,"on_time":10270,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-65,"led_off":0,"latitude":0,"longitude":0}}}
[2017-11-16 17:15:34,145] DEBUG: Checking status of 192.168.0.187.
[2017-11-16 17:15:34,146] DEBUG: IP 192.168.0.187 is valid.
[2017-11-16 17:15:34,298] DEBUG: Sending command info to 192.168.0.187
[2017-11-16 17:15:34,304] DEBUG: « 'õ"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.163650","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"xxxxxxxxxxx","deviceId":"8006987341FA98689E31EFE2772EBABC17DC609E","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"59542398207847F07D67AB57530532B8","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"MPSM","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":1,"on_time":11562,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-68,"led_off":0,"latitude":0,"longitude":0}}}

jneilliii, 2017-11-17:
Which version of the plugin? Can you upload the entire debug log and verify if you have any errors in the console or octoprint.log as well? Also in config.yaml make sure the plugs setting is enabled for gcodeprocessing like below where gcodeEnabled is set to true. 

```
  tplinksmartplug:
    _config_version: 3
    arrSmartplugs:
    - autoConnect: false
      autoConnectDelay: '15'
      autoDisconnect: false
      autoDisconnectDelay: 0
      btnColor: '#00FF00'
      currentState: 'on'
      displayWarning: false
      gcodeEnabled: true
      gcodeOffDelay: 0
      gcodeOnDelay: 0
      ip: 192.168.0.187
      sysCmdOff: false
      sysCmdOffDelay: 0
      sysCmdOn: false
      sysCmdOnDelay: 0
      sysRunCmdOff: ''
      sysRunCmdOn: ''
      warnPrinting: false
```

From what it looks like it never gets to the point where it sends the off command.  There should be a line in the debug log like "Received M81 command, attempting power off."

I might actually in the near future change the logic here because I think there's too much communication back and forth between the javascript client side and the python processing side with how it's set-up now.  When the gcode string is detected it sends a message to the javascript side to lookup up the plug in the array by ip, initiate a timer to send the instruction back to the python side to actually do the work, and then another message back to the javascript side to update the icon display.  I think I figured out a better way to do the lookup on the python side and will do it that way with python timers instead and take a round trip of communication out of the equation.

jneilliii, 2017-11-17:
I've confirmed this is an issue with using ip address.  Using hostname works, ip address does not.  Will have to figure out why my regular expression isn't working as expected.

jneilliii, 2017-11-17:
I just retested and I am unable to duplicate your issue.  If the plug is configured and controllable via the icon, the gcode processing should be working fine.  I still will be looking into changing the processing logic in the future, but this is what the debug log should look like.

```
[2017-11-17 03:18:32,091] DEBUG: Received M81 command, attempting power off.
[2017-11-17 03:18:32,141] DEBUG: Turning off 192.168.1.130.
[2017-11-17 03:18:32,142] DEBUG: IP 192.168.1.130 is valid.
[2017-11-17 03:18:32,155] DEBUG: Sending command off to 192.168.1.130
[2017-11-17 03:18:32,157] DEBUG: «  -ý"system":{"set_relay_state":{"err_code":0}}}
[2017-11-17 03:18:32,158] DEBUG: Checking status of 192.168.1.130.
[2017-11-17 03:18:32,159] DEBUG: IP 192.168.1.130 is valid.
[2017-11-17 03:18:32,172] DEBUG: Sending command info to 192.168.1.130
[2017-11-17 03:18:32,176] DEBUG: « 1ã"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.2 Build 161014 Rel.164426","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:6A:6C:83","deviceId":"8006DB119EADD41BCE508828B19A949F1872F15C","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"6C4ACA62CB1B921060CF24148345CD43","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"CR-10","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":0,"on_time":0,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-67,"led_off":0,"latitude":0,"longitude":0}}}
```

xkcd42, 2017-11-17:
plug in version .7.2
No console log written today.
Only errors I spotted in the Octoprint log are serial comm connecting to the USB
Everything works fine with power on / off from the web interface. 
Full log.  Note last line does show the shut off attempt. 
I'll retry tomorrow - see if I can take out some variables and capture a single print job instead of multiple attempts. 

[plugin_tplinksmartplug_debug (3).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1480989/plugin_tplinksmartplug_debug.3.log)

jneilliii, 2017-11-17:
Thanks.  Also, just try it directly from the terminal tab, just enter the M81 line as it appears in you end gcode.  One thing I did think of, are you printing form SD card? I don't believe the gcode will get processed by octoprint in that case.

jneilliii, 2017-11-17:
Just saw in your log that it is seeing the M81 command, and shortly after it is doing a status check on the plug but it never looks like it sends the off command.  There are only a couple of reasons I can think of that would cause this.  It's not enabled in the settings for the plug, it can't find the plug on the javascript side from the ip address it is receiving from the python side, or the delays are messing up the math used in the javascript time delay function.  For me with these settings it works entering the M81 command in the terminal tab.

[image]

xkcd42, 2017-11-17:
Issue Resolved.  I was able to figure out the problem via the terminal commands (thanks for that guidance).  Somewhere in the set up I had turned the "Warning" parameter on (left check box of the pair).  So when the gcode issued the M81 shutdown command it stopped waiting for the confirm on the pop-up box.  Because it was an unattended shutdown - obviously I never confirmed.   Thanks for your help and thanks for all your efforts on this plugin.

closed 2017-11-17 by xkcd42