jneilliii/OctoPrint-TPLinkSmartplug #28: Smartplug Settings??

flowrider669, 2017-11-21:
First of all thanks for the amazing work. I'm trying to setup TP-Link Smartplug plugin 0.7.2 with Octoprint 1.3.5. 

I have input my IP into the top section but at am a loss as to what to put into the "System Command On" and System Command Off" boxes. As of right now they are empty and the lightning bolt is greyed out. If I hover over it I only get IP address. I've searched but am at a loss. Any help would be appreciated. 

Thanks

gumpdy, 2017-11-21:
I'm pretty sure it's broken.

flowrider669, 2017-11-21:
Uuuughhhhh. Well that would explain it.

gumpdy, 2017-11-21:
HS105?

flowrider669, 2017-11-21:
HS100

jneilliii, 2017-11-21:
Unless there is some other weird firmware issue like with the HS105, the 0.7.2 version should be working for you.  Can you please enable debug logging in TPLinkSmartplug settings and try again.  Then send me the debug log that it generates.

jneilliii, 2017-11-21:
As for system commands, those are just for running external commands when the button is triggered, leave those blank unless you want to do other stuff.

jneilliii, 2017-11-22:
@gumpdy @slimshizn I've just made a quick change on a new branch for testing for you guys.  Install via the plugin manager using the following url.  Once installed and the page has been reloaded go into TPLinkSmartplug Settings and enable classic encryption functions. For the sake of testing, uncheck all boxes like this. If that works, then we can debug which options are breaking the functionality for you.

[image]

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/HS105.zip

flowrider669, 2017-11-23:
Thanks @jneilliii. Here's the log. 
`[2017-11-23 03:00:47,558] DEBUG: Checking status of 192.168.1.141.
[2017-11-23 03:00:47,558] DEBUG: IP 192.168.1.141 is valid.
[2017-11-23 03:00:47,567] DEBUG: Sending command info to 192.168.1.141
[2017-11-23 03:00:47,569] DEBUG: ´ 9Î"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.163650","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:67:C3:C8","deviceId":"800602D2F7722E86CD5AE9AF35D74D21186A4199","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"59542398207847F07D67AB57530532B8","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"3D Printer","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":1,"on_time":197,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-54,"led_off":0,"latitude":49.132254,"longitude":-122.670379}}}
[2017-11-23 03:02:50,686] DEBUG: Checking status of 192.168.1.141.
[2017-11-23 03:02:50,687] DEBUG: IP 192.168.1.141 is valid.
[2017-11-23 03:02:50,742] DEBUG: Sending command info to 192.168.1.141
[2017-11-23 03:02:50,744] DEBUG: ´ 9Î"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.163650","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:67:C3:C8","deviceId":"800602D2F7722E86CD5AE9AF35D74D21186A4199","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"59542398207847F07D67AB57530532B8","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"3D Printer","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":1,"on_time":321,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-55,"led_off":0,"latitude":49.132254,"longitude":-122.670379}}}
[2017-11-23 03:03:51,197] DEBUG: Checking status of 192.168.1.141.
[2017-11-23 03:03:51,197] DEBUG: IP 192.168.1.141 is valid.
[2017-11-23 03:03:51,293] DEBUG: Sending command info to 192.168.1.141
[2017-11-23 03:03:51,296] DEBUG: ´ 9Î"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.163650","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:67:C3:C8","deviceId":"800602D2F7722E86CD5AE9AF35D74D21186A4199","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"59542398207847F07D67AB57530532B8","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"3D Printer","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":1,"on_time":381,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-56,"led_off":0,"latitude":49.132254,"longitude":-122.670379}}}
[2017-11-23 03:03:53,667] DEBUG: Checking status of 192.168.1.141.
[2017-11-23 03:03:53,667] DEBUG: IP 192.168.1.141 is valid.
[2017-11-23 03:03:53,682] DEBUG: Sending command info to 192.168.1.141
[2017-11-23 03:03:53,688] DEBUG: ´ 9Î"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.163650","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:67:C3:C8","deviceId":"800602D2F7722E86CD5AE9AF35D74D21186A4199","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"59542398207847F07D67AB57530532B8","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"3D Printer","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":1,"on_time":383,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-56,"led_off":0,"latitude":49.132254,"longitude":-122.670379}}}
[2017-11-23 03:03:54,481] DEBUG: Checking status of 192.168.1.141.
[2017-11-23 03:03:54,481] DEBUG: IP 192.168.1.141 is valid.
[2017-11-23 03:03:54,492] DEBUG: Sending command info to 192.168.1.141
[2017-11-23 03:03:54,493] DEBUG: ´ 9Î"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.163650","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:67:C3:C8","deviceId":"800602D2F7722E86CD5AE9AF35D74D21186A4199","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"59542398207847F07D67AB57530532B8","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"3D Printer","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":1,"on_time":384,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-57,"led_off":0,"latitude":49.132254,"longitude":-122.670379}}}
[2017-11-23 03:03:55,289] DEBUG: Checking status of 192.168.1.141.
[2017-11-23 03:03:55,290] DEBUG: IP 192.168.1.141 is valid.
[2017-11-23 03:03:55,299] DEBUG: Sending command info to 192.168.1.141
[2017-11-23 03:03:55,301] DEBUG: ´ 9Î"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.163650","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:67:C3:C8","deviceId":"800602D2F7722E86CD5AE9AF35D74D21186A4199","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"59542398207847F07D67AB57530532B8","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"3D Printer","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":1,"on_time":385,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-57,"led_off":0,"latitude":49.132254,"longitude":-122.670379}}}
[2017-11-23 03:03:55,321] DEBUG: Checking status of 192.168.1.141.
[2017-11-23 03:03:55,321] DEBUG: IP 192.168.1.141 is valid.
[2017-11-23 03:03:55,331] DEBUG: Sending command info to 192.168.1.141
[2017-11-23 03:03:55,333] DEBUG: ´ 9Î"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.163650","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:67:C3:C8","deviceId":"800602D2F7722E86CD5AE9AF35D74D21186A4199","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"59542398207847F07D67AB57530532B8","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"3D Printer","dev_name":"Wi [... cut ...]
[2017-11-23 03:04:00,619] DEBUG: ´ 9Î"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.163650","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:67:C3:C8","deviceId":"800602D2F7722E86CD5AE9AF35D74D21186A4199","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"59542398207847F07D67AB57530532B8","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"3D Printer","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":1,"on_time":390,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-62,"led_off":0,"latitude":49.132254,"longitude":-122.670379}}}
`

jneilliii, 2017-11-23:
Is that with or without the classic encryption function enabled? Is there any errors in your octoprint log?

flowrider669, 2017-11-24:
with classic encryption. I'll have to check the octoprint log tonight.

gumpdy, 2017-11-24:
Classic encryption enabled. Very fresh install of octopi.  Here's my logs.  Those logs don't seem verbose enough to tell me what the heck is going on. The status check says ""relay_state":0,"on_time":0,"" Does clicking the lightning bolt icon attempt to turn it on? Why isn't that posted to the log with the return response? How can I manually via SSH command line attempt to turn the plug on and off? 

[plugin_tplinksmartplug_debug (10).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1502663/plugin_tplinksmartplug_debug.10.log)

[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1502661/octoprint.log)

jneilliii, 2017-11-24:
The process currently is, on load the plugin checks the current state of the relay on the plug. If relay_state is 0 then the plug is off, and the lightning bolt should turn red.  When icon is red on click it will send the command to turn the plug on and get the response and change the icon green. If icon is gray when clicked it is supposed to get the current state of the relay and change the icon color accordingly.  From the logs it looks like it is getting the response.   I'm assuming this is still related to the HS105 bug with certain firmware.

In order to test from command line you would need to ssh to the pi and run the following commands. These are the original scripts I was using to incorporate into this plugin.

```
wget https://raw.githubusercontent.com/softScheck/tplink-smartplug/master/tplink-smartplug.py
chmod +x ./tplink-smartplug.py
./tplink-smartplug.py -t 192.168.1.74 -c info
./tplink-smartplug.py -t 192.168.1.74 -c on
./tplink-smartplug.py -t 192.168.1.74 -c off
```

jneilliii, 2017-11-25:
Just made a little change to the HS105 branch for additional information gathering.  It will now also log to the browser console the information being returned back from the python side.  Reinstall from the url below.  Let me know what you get in the console when you click.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/HS105.zip

gumpdy, 2017-11-25:
Will do. I did some digging since my last post and was able to turn it on
and off with

{
 "method":"passthrough",
 "params":{
 "deviceId":"80066F057ACD0BA64E0162DCB326D60018AC73C4",
 "requestData":"{\"system\":{\"set_relay_state\":{\"state\":1}}}"
 }
}

with a HTTP POST to

https://use1-wap.tplinkcloud.com/?token**MYTOKENTHATIDELETED**&appName=Kasa_Android&termID=**MYUUIDTHATIDELETED**&appVer=1.4.4.607&ospf=Android+6.0.1&netType=wifi&locale=en_US
HTTP/1.1

jneilliii, 2017-11-25:
That information helps me as well,  I don't use cloud control on my plugs, local only.  I don't know if the software does something to the local side is cloud control is enabled.  What happens if you turn that feature off on the plug?

gumpdy, 2017-11-25:
> Kasa_Android&termID=**MYUUIDTHATIDELETED**&appVer=1.
> 4.4.607&ospf=Android+6.0.1&netType=wifi&locale=en_US HTTP/1.1

Completely irrelevant, I know. But figured I'd share. It's a foolproof
method for turning it on and off as it uses the same POST method that the
app on the phone uses. Perhaps they made adjustments to how the app sends
commands on LAN? Maybe I can turn off data on the phone and somehow capture
those packets when it attempts to do a "local-only" switching request? Any
ideas on how to do that?

gumpdy, 2017-11-25:
> **&appName=Kasa_Android&termID=**MYUUIDTHATIDELETED**&

Setting it to local only didn't fix the problem.

jneilliii, 2017-11-25:
Unfortunately, the decrypting part with wifi is way over my head. You'd have to look into the original softchek repository, as their is much information there on how they did it.

https://github.com/softScheck/tplink-smartplug

gumpdy, 2017-11-25:
pi@octopi:~ $ ./tplink-smartplug.py -t 192.168.1.74 -c on
Sent:      {"system":{"set_relay_state":{"state":1}}}
Received:
pi@octopi:~ $ ^C
pi@octopi:~ $ ./tplink-smartplug.py -t 192.168.1.74 -c info
Sent:      {"system":{"get_sysinfo":{}}}
Received:

gumpdy, 2017-11-25:
I will try your updated plugin and report back shortly

gumpdy, 2017-11-25:
packed_core.js?6edaa7f5:13020 Error calling onDataUpdaterPluginMessage on
view model tplinksmartplugViewModel : ReferenceError: plug is not defined
    at tplinksmartplugViewModel.self.onDataUpdaterPluginMessage (
http://192.168.1.254/static/webassets/packed_plugins.js?b3e32fee:919:9)
    at
http://192.168.1.254/static/webassets/packed_core.js?6edaa7f5:13010:43
    at Pn (
http://192.168.1.254/static/webassets/packed_libs.js?d8971830:10655:530)
    at Function.<anonymous> (
http://192.168.1.254/static/webassets/packed_libs.js?d8971830:10676:66)
    at callViewModelsIf (
http://192.168.1.254/static/webassets/packed_core.js?6edaa7f5:13004:7)
    at callViewModels (
http://192.168.1.254/static/webassets/packed_core.js?6edaa7f5:12967:5)
    at
http://192.168.1.254/static/webassets/packed_core.js?6edaa7f5:11925:13
    at DataUpdater.self._ifInitialized (
http://192.168.1.254/static/webassets/packed_core.js?6edaa7f5:11943:13)
    at DataUpdater.self._onPluginMessage (
http://192.168.1.254/static/webassets/packed_core.js?6edaa7f5:11924:14)
    at
http://192.168.1.254/static/webassets/packed_client.js?4250cfb6:378:17

gumpdy, 2017-11-25:
> 6edaa7f5:13010:43
> d8971830:10655:530)
> webassets/packed_libs.js?d8971830:10676:66)
> webassets/packed_core.js?6edaa7f5:13004:7)
> webassets/packed_core.js?6edaa7f5:12967:5)
> 6edaa7f5:11925:13
> webassets/packed_core.js?6edaa7f5:11943:13)
> webassets/packed_core.js?6edaa7f5:11924:14)
> 4250cfb6:378:17

Starting dependency resolution...
packed_core.js?6edaa7f5:13480 ... dependency resolution done
packed_core.js?6edaa7f5:13738 Initial application setup done, connecting to
server...
packed_core.js?6edaa7f5:11834 Connected to the server
packed_core.js?6edaa7f5:13716 Finalizing application startup
packed_core.js?6edaa7f5:13625 Going to bind 28 view models...
packed_core.js?6edaa7f5:13666 Did not bind view model AnywhereViewModel to
target #wizard_plugin_anywhere since it does not exist
packed_plugins.js?b3e32fee:1084 checking 192.168.1.74
packed_core.js?6edaa7f5:13666 Did not bind view model CuraViewModel to
target #wizard_plugin_cura since it does not exist
packed_core.js?6edaa7f5:13666 Did not bind view model
SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it
does not exist
packed_core.js?6edaa7f5:2165 User admin logged in
packed_core.js?6edaa7f5:13700 ... binding done
packed_core.js?6edaa7f5:13712 Application startup complete
packed_core.js?6edaa7f5:13020 Error calling onDataUpdaterPluginMessage on
view model tplinksmartplugViewModel : ReferenceError: plug is not defined
    at tplinksmartplugViewModel.self.onDataUpdaterPluginMessage (
http://192.168.1.254/static/webassets/packed_plugins.js?b3e32fee:919:9)
    at
http://192.168.1.254/static/webassets/packed_core.js?6edaa7f5:13010:43
    at Pn (
http://192.168.1.254/static/webassets/packed_libs.js?d8971830:10655:530)
    at Function.<anonymous> (
http://192.168.1.254/static/webassets/packed_libs.js?d8971830:10676:66)
    at callViewModelsIf (
http://192.168.1.254/static/webassets/packed_core.js?6edaa7f5:13004:7)
    at callViewModels (
http://192.168.1.254/static/webassets/packed_core.js?6edaa7f5:12967:5)
    at
http://192.168.1.254/static/webassets/packed_core.js?6edaa7f5:11925:13
    at DataUpdater.self._ifInitialized (
http://192.168.1.254/static/webassets/packed_core.js?6edaa7f5:11943:13)
    at DataUpdater.self._onPluginMessage (
http://192.168.1.254/static/webassets/packed_core.js?6edaa7f5:11924:14)
    at
http://192.168.1.254/static/webassets/packed_client.js?4250cfb6:378:17
(anonymous) @ packed_core.js?6edaa7f5:13020

jneilliii, 2017-11-25:
ok, it looks like it's definitely an issue with the encryption functions or response not coming back from the plug.  Purely based on the comand lines...the plug isn't playing nice and responding.

```
pi@octopi:~ $ ./tplink-smartplug.py -t 192.168.1.74 -c on
Sent:      {"system":{"set_relay_state":{"state":1}}}
Received:
pi@octopi:~ $ ^C
pi@octopi:~ $ ./tplink-smartplug.py -t 192.168.1.74 -c info
Sent:      {"system":{"get_sysinfo":{}}}
Received:
```

Can you try the updated python script using these commands in ssh.

```
rm ./tplink-smartplug.py
wget https://github.com/kchiem/tplink-smartplug/raw/master/tplink-smartplug.py
chmod +x ./tplink-smartplug.py
./tplink-smartplug.py -t 192.168.1.74 -c info
```

We're hoping for output like this.

```
pi@octopi:~ $ ./tplink-smartplug.py -t 192.168.1.74 -c info
Sent:      {"system":{"get_sysinfo":{}}}
Received:  {"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.2 Build 161014 Rel.164426","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"50:C7:BF:6A:6C:83","deviceId":"8006DB119EADD41BCE508828B19A949F1872F15C","hwId":"5EACBE93FB9E32ECBE1F1C2ADE6DDE11","fwId":"6C4ACA62CB1B921060CF24148345CD43","oemId":"37589AA1F5CACDC53E2914B7760127E5","alias":"HS105","dev_name":"Wi-Fi Smart Plug","icon_hash":"","relay_state":1,"on_time":4895,"active_mode":"schedule","feature":"TIM","updating":0,"rssi":-72,"led_off":0,"latitude":0,"longitude":0}}}
pi@octopi:~ $
```

gumpdy, 2017-11-25:
Works. What'd you do?

jneilliii, 2017-11-25:
What works? The plugin or the python script?

gumpdy, 2017-11-25:
Python

jneilliii, 2017-11-25:
Ok, if that last python script works, try unchecking classic encryption, save settings, restart octoprint, etc.

gumpdy, 2017-11-25:
I came back to tell you that I installed another plug in and upon
restarting your plug-in had a red icon all of a sudden... Wtf? I didn't do
anything differently

jneilliii, 2017-11-25:
I have no idea what happened.

jneilliii, 2017-11-25:
Does the button work to control the plug now?

gumpdy, 2017-11-25:
Yes.

gumpdy, 2017-11-25:
I'm going to attempt to uninstall it and reinstall it and try to break it
again

jneilliii, 2017-11-25:
Let me know the outcome, and which version you install.

flowrider669, 2017-11-25:
Wow a lot has happened since this morning! Still broken for me. What should I do? LOL. Sorry I'm not Python literate at all.

jneilliii, 2017-11-25:
@flowrider669, I'd start by uninstalling the plugin you have now and install the one from the url below. If it doesn't work after installing try enabling classic encryption in settings and restart octorprint, ctrl+f5 to reload the web client, etc.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/HS105.zip

gumpdy, 2017-11-25:
It works even with the plug-in from the default Repository after doing an
uninstall and reinstall. All along I was doing control F5 to force a
refresh without cache.

Flowrider can you confirm that it's not working
?  Install the one from the Repository and not the link he gave you as I
can confirm it works just fine now.

I get that these commands are being sent locally directly to the plug so
they should not be affected by any external changes by TP Link but I am
just trying to have you recreate the steps I took.

If it still does not work after doing the requested restarts install a
plugin called Action commands. It is the first plug-in in the octoprint
Repository. Upon installing this plug-in and restarting all of a sudden the
icon turned red and it worked perfectly fine. Since then I have uninstalled
the dev branch version of the plug-in and am using the one from the
Repository in the plug-in manager.

Maybe installing that plugin and it working was just a coincidence but it's
worth a few clicks and a few minutes of your time to see. If it does work
is that just the result of not having the required dependencies?

flowrider669, 2017-11-25:
ok will give it a shot

jneilliii, 2017-11-25:
Interesting @gumpdy. I would expect to see errors in the octoprint.log if there were missing references, but I'll take a look at the Action commands plugin to see.

jneilliii, 2017-11-25:
Yeah, just looked and that plugin doesn't have anything in it's requirement.txt file besides OctoPrint, so don't think it's related to missing references.

flowrider669, 2017-11-25:
Weird it just worked!! Thanks @gumpdy and @jneilliii 
Not too sure what it was but it works now. Thanks

closed 2017-11-25 by jneilliii

gumpdy, 2017-11-25:
You just put tingles down my spine I'm glad we figured it out I'm sure
he'll have the answer for us shortly. I'm guessing it's either a dependency
missing or something to do with a change to the settings file. You would
think that deleting a plug-in would delete it's settings.

gumpdy, 2017-11-25:
If you are you interested in chasing down why this is happening I'm willing
to do whatever you need me to do to recreate the problem even if it
requires a fresh install of octopi.

jneilliii, 2017-11-25:
I have no idea why it would have just started working like that. Uninstalling a plugin definitely does not delete the plugin settings. If you can re-create the broken state I would like to try and figure it out.