jneilliii/OctoPrint-TPLinkSmartplug #30: Auto Off after print completion

adammhaile, 2017-12-16:
Feature Request: Automatically turn off the printer using the smart plug after a print completes, maybe with a configurable time delay (to let the fans run and cool things for a bit). I run a lot of prints that continue over night and it's be nice to not have the printer say on until I check it in the morning.

jneilliii, 2017-12-16:
duplicate of #16, #24, and #29. 

You can achieve this by enabling GCODE processing and in your slicer's end gcode set-up or in OctoPrint's settings > GCODE Scripts.  Follow the format detailed on the [readme](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug#settings-explained) page. Make sure to disable any of the warning options for this as it will interfere with the process.

closed 2017-12-16 by jneilliii

adammhaile, 2017-12-16:
Would be nice if that was a single checkbox. But thanks for pointing that
out.