jneilliii/OctoPrint-TPLinkSmartplug #31: M81 no longer working?

eightheads, 2017-12-18:
I have my Octoprint "After print job completes" GCode set to "M81 192.168.1.127 ; turn off printer" but it no longer seems to be working and my TP-Link Smartplug stays on. The print is completing and is shown as successful in Octoprint but it does not turn off the printer. If I run that Gcode directly in the terminal in the Octoprint web interface it will turn off.

jneilliii, 2017-12-19:
That's an interesting point.  I don't know if I ever tried to set it in the OctoPrint settings for After print job completes, only in my slic3r's end gcode.  I wonder if @foosel is not sending those through the octoprint.comm.protocol.gcode.queuing hook I'm using? Also, make sure you don't have the warning's enabled. Some have reported that having that enabled can interfere, although I don't know why.

eightheads, 2017-12-19:
I do not have either of the warnings on. This also worked fine previously. I also noticed tonight that I can send it through the Octoprint terminal and it will turn off, but if I try to send the same command through the terminal in the Printoid app if will not turn off.

eightheads, 2017-12-20:
Here is my debug file from last night.
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/1576639/plugin_tplinksmartplug_debug.log)

jneilliii, 2017-12-20:
That is strange @eightheads.  Looking at the log, it appears to have sent the command once, but later in the log it is recognizing the M81 command, but not sending the off command. for example, this is where it looks like it worked, as the check status right after returns a relay_state of 0.

```
[2017-12-19 14:36:03,930] DEBUG: Received M81 command, attempting power off.
[2017-12-19 14:36:04,008] DEBUG: Turning off 192.168.1.127.
[2017-12-19 14:36:04,009] DEBUG: IP 192.168.1.127 is valid.
[2017-12-19 14:36:04,016] DEBUG: Sending command off to 192.168.1.127
[2017-12-19 14:36:04,017] DEBUG: «  -ý"system":{"set_relay_state":{"err_code":0}}}
[2017-12-19 14:36:04,018] DEBUG: Checking status of 192.168.1.127.
[2017-12-19 14:36:04,019] DEBUG: IP 192.168.1.127 is valid.
[2017-12-19 14:36:04,029] DEBUG: Sending command info to 192.168.1.127
[2017-12-19 14:36:04,033] DEBUG: « Qƒ"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.164033","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS110(US)","mac":"50:C7:BF:74:82:44","deviceId":"8006EE697C0D122E4FCD57723C391D8A187C11C3","hwId":"60FF6B258734EA6880E186F8C96DDC61","fwId":"060BFEA28A8CD1E67146EB5B2B599CC8","oemId":"FFF22CFF774A0B89F7624BFC6F50D5DE","alias":"3D Printer","dev_name":"Wi-Fi Smart Plug With Energy Monitoring","icon_hash":"","relay_state":0,"on_time":0,"active_mode":"schedule","feature":"TIM:ENE","updating":0,"rssi":-58,"led_off":0,"latitude":30.442468,"longitude":-90.437664}}}
```

and this is where it is not sending the command at all for some reason later in the log.

```
[2017-12-19 21:47:32,220] DEBUG: Received M81 command, attempting power off.
[2017-12-19 21:52:00,053] DEBUG: Received M81 command, attempting power off.
[2017-12-19 21:52:06,364] DEBUG: Received M81 command, attempting power off.
[2017-12-19 21:52:23,899] DEBUG: Checking status of 192.168.1.127.
[2017-12-19 21:52:23,899] DEBUG: IP 192.168.1.127 is valid.
[2017-12-19 21:52:24,006] DEBUG: Sending command info to 192.168.1.127
[2017-12-19 21:52:24,010] DEBUG: « T†"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.1.1 Build 160725 Rel.164033","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS110(US)","mac":"50:C7:BF:74:82:44","deviceId":"8006EE697C0D122E4FCD57723C391D8A187C11C3","hwId":"60FF6B258734EA6880E186F8C96DDC61","fwId":"060BFEA28A8CD1E67146EB5B2B599CC8","oemId":"FFF22CFF774A0B89F7624BFC6F50D5DE","alias":"3D Printer","dev_name":"Wi-Fi Smart Plug With Energy Monitoring","icon_hash":"","relay_state":1,"on_time":6005,"active_mode":"schedule","feature":"TIM:ENE","updating":0,"rssi":-57,"led_off":0,"latitude":30.442468,"longitude":-90.437664}}}
```

I'm curious, have you tried enabling the classic encryption option? I don't think it would make a difference in this case at all, and not really sure why it would work at one point and not later in the day.

eightheads, 2017-12-20:
Here is some more context to the log above:
- "M81 192.168.1.127" command in the "After print job completes" GCode set in Octoprint appears to have attempted to run when the print completed but unit stayed on
- I then sent "M81 192.168.1.127" through the terminal in the Printoid 2 times but unit stayed on
- I then went to the Octoprint web browser and clicked the lightning bolt icon to power off and it successfully powered off.

eightheads, 2017-12-20:
Also, I am accessing Octoprint through a DNS routing address for Printoid... I don't know if that would matter at all being the command is sent via Octoprint locally to the printer itself... The M81 command worked fine previously, before the addition of the IP being added for sure.

jneilliii, 2017-12-20:
Sorry about that.  The classic encryption option is in a separate branch I did trying to debug the HS105 issues people were having.  It never made it into the master branch.  I still don't think that will help though because the command works when you press the button.  If it was an issue with the encryption functions that wouldn't work either. I'll do some debugging on my end this weekend to see if I can figure out what might be the cause.  One quick test though I'm curious about.  If you take that out of OctoPrint's after print completes settings and put the command directly in a blank gcode file and run it on your OctoPrint instance, does it power off the plug then? This in theory would be the equivalent of typing it into the terminal.

jneilliii, 2017-12-20:
Also, what do you have for the gcode delays?  Looking back at my code, the way it is processing the commands is receiving the gcode command -> send message back to web frontend to instruct it to power off the plug after the set gcode delay option -> web frontend is supposed to then send the power off command back to the server back-end.  I've been wanting to re-write the logic so all of this happens on the server-side without having to have the web front end, which is what I think you might be running into with it not working in the printoid app as well.

eightheads, 2017-12-20:
These are my current settings. I will try running a blank GCode tonight with the M81 command.
[image]

jneilliii, 2017-12-20:
If my theory is correct about the processing logic, it should work with the full web interface loaded, but fail if you have that interface closed and print the file from the printoid app.

jneilliii, 2017-12-20:
It will still put into the debug log that the M81 command was detected for both.

jneilliii, 2017-12-21:
I did a little work last night and pushed a new branch that should help with this issue.  It removes the dependency on the web front end for processing and handles everything from the server side now.  If you don't mind installing from the url below in plugin manager and report back that would be great and I can merge these changes into the master branch.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/python_timer.zip

eightheads, 2017-12-21:
@jneilliii will do, thanks.

eightheads, 2017-12-22:
Just ran a quick print and it shut off perfectly as desired. This was with M81 in the ending gcode in Octoprint still. Thanks!

jneilliii, 2017-12-22:
Thanks.  Merged with master and released with #32.

closed 2017-12-22 by jneilliii