jneilliii/OctoPrint-TPLinkSmartplug #36: 0.7.3 breaks the ability to issue M80, m81 in the terminal or from emergency stop switch plugin.

overling, 2017-12-24:
I use an output pin on my smoothieware MKS Sbase to turn on power supplies via the PSU plugin and M80, M81 commands.

After updating the TPlink plugin to 0.7.3 those commands are no longer able to be sent from another plugin or from the terminal manually.  It took a few hours to figure out why i couldnt turn on my printer.   After disabling the TPlink plugin I able able to use the PSU power plugin again.

The TPlink is configured to switch via IP address (not m80,m81) and that does still work.

jneilliii, 2017-12-24:
The update fixed an issue with not processing the gcode commands when the web ui was not loaded. This in turn probably interferes with the psu control plugin as we're probably using the same hooks. I'll have to review that plugins code to verify and see why there wills be a conflict.

jneilliii, 2017-12-24:
After going back and reading the documentation for [gcode queing hook](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-phase) the psu control and my plugin use to function, I can't see where mine would be preventing the other from working.  Reading the documentation, returning without any parameters (which is what I use) passes the gcode commands through without changing them. So unless the documentation is wrong, the way I have it coded is to do just that. I'll reach out to the other plugin author and see what there thoughts are.

overling, 2017-12-24:
Thank you for the quick reply.    Also, going to the terminal and manually sending the M80,m81 commands does not work when the TPlink plugin is installed.  Maybe you caught that above but when those commands are given they do not even display in the scrolling list.  That might be what you meant by the hook but I wasn't sure so just in case im adding this.

jneilliii, 2019-10-14:
I've re-opened your issue @overling and went over to the PSUControl plugin to submit an issue there as well so I could work with @kantlivelong but it seems that issue tracker is not enabled for his repo.  Can you do me a favor and verify this issue remains with installing the current development version using the link below in plugin manager?

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.17.zip

If it's still an issue, please share screenshots of your tplink plug settings and the PSUControl plugin settings so that I can set-up my OctoPrint exactly the same for additional troubleshooting.

overling, 2019-10-14:
Yah ill try this version.

jneilliii, 2019-10-31:
Sorry for missing that last post. Check your octoprint.log for any errors on initialization.

jneilliii, 2021-01-27:
Cleaning up my issue tracker. Please open a new issue if you are still experiencing issues as there have been many plugin updates since this issue was opened.

closed 2021-01-27 by jneilliii