jneilliii/OctoPrint-TPLinkSmartplug #37: Sending M81 to any Smartplug cause Octoprint to go off line

cpd1, 2017-12-27:
I'm running TPLinkSmartplug 0.7.3 and Octoprint 1.3.6 on a Raspberry Pi.
I'm using TPLinkSmartplug to control 3 plugs.
1) The printer's control box
2) The heat bed's external power supply
3) The heat/lights for the printer's enclosure.
On starting a print my startup gcode issues a M81 to turn off #2.
I need to turn off the AC power to the bed as it interferes with the capacitive auto bed level sensor.
The bed's power is successfully disconnected, however it seems that Octoprint is also instructed to go offline thus ending the print job.
From the Octoprint Terminal
Send : N10 M81 <IP#2>*35Changing monitoring state from 'Printing' to 'Offline'

I haven't been able to find any configuration option to stop the automatic disconnect of Octoprint, but it should be possible to turn off various non printer box plugs without Octoprint going off line
Thanks,
Paul

jneilliii, 2017-12-27:
You mean the setting for auto disconnect on power off, it's a checkbox in the plugin settings.

cpd1, 2017-12-27:
My plugin screen doesn't match the published one.  I don't have an auto disconnect check box.  
Based on your response I unchecked the preDisconnect check box and it now seems to work as expected. 
Thanks,
Paul.

closed 2017-12-27 by cpd1