jneilliii/OctoPrint-TPLinkSmartplug #38: Using the name of the switch rather than the IP address

Joemac05, 2018-01-21:
Is this possible or am I dreaming? If I have to remember the IP address to get the right switch most anything could happen. <Grin>

jneilliii, 2018-01-21:
It does support hostnames, but not sure if that's able to be set in the app. I use this method, but am setting the hostname on the network through the dhcp reservation options on my router.

Joemac05, 2018-01-21:
Thank you, looks like I have something to learn today.

closed 2018-01-21 by jneilliii

Joemac05, 2018-01-22:
This is now working fine here. It was a bit of a challenge in that I am running an older windows  server inside of a virtual machine. This was done to skip around having to do a complete reinstall of my old setup.  
Ugh, don't ask, my son the IT guy does that sort of stuff for me. I am now controlling two switches very reliable it seems.

jneilliii, 2018-01-29:
Others have requested the ability to label the buttons.  Expect a new version release sometime this weekend that adds the ability with a significant improvement to settings UI layout.

Joemac05, 2018-01-29:
Cool… It comes in very handy for more than just Octoprint. I’m sure a lot of folks are not into figuring out how to name an IP address.