jneilliii/OctoPrint-TPLinkSmartplug #39: Poll Switch for Current Status

egelmex, 2018-01-26:
The UI shows the last known value, on or off of the switch, not the current value. From time to time I switch the printer off using the button on the plug because it's more convenient. It would be useful to me if the UI periodically checked the status of the plug, assuming the plus provides and API to do this.

jneilliii, 2018-01-29:
The plugin would actually check the status of all plugs on refresh of the browser, but it makes sense to have this feature.  I just released an updated version 0.8.0 that allows the configuration of a polling interval to check the status every x minutes after the page is loaded.  Once enabled you must refresh the browser once to start the polling loop.

closed 2018-01-29 by jneilliii

egelmex, 2018-01-29:
Amazing, thank you very much. I am away this week but I will test it when I
get back.