jneilliii/OctoPrint-TPLinkSmartplug #41: Ability to add Name for Menu Bar Icon

jenovak, 2018-01-29:
I have two tp-link plugs on my CR-10 Mini setup. One plug controls the CR-10 and a second one controls a light for video recording. When I add the two plugs it shows the IP address when I hover over the icons on the menu bar. It would be great if I could supply a name to use instead of the IP address for this. I don't seem to have the ability to have my own DNS server on my home network ( Apple Airport Extreme Router ).

jneilliii, 2018-01-29:
I've done that for a couple of other plugins, don't see why I couldn't add this to this one.  I've been wanting to re-vamp the layout anyway because I can't stand how cluttered it is.  Will probably incorporate this into that change, expect to see a new version released sometime this weekend.

jneilliii, 2018-01-30:
Added with version 0.9.0.

closed 2018-01-30 by jneilliii

jenovak, 2018-01-30:
Thanks for such a rapid response. I've installed the new version and found a minor issue although I can work around it. If you try to add two plugs at the same time then the second one overrides the first and you get two plugs with the data of the second one. If you add one plug then do a save then come back in and add the second plug then save again it works fine. Once again thank you for the work you are doing on this plugin.

jneilliii, 2018-01-30:
Thanks.  I probably should have done some more testing on that before releasing.  I'll look into it tonight when I get home from work and see if I can resolve that.

jneilliii, 2018-01-31:
Got you squared away with latest 0.9.2 version.