jneilliii/OctoPrint-TPLinkSmartplug #46: [Request] Documentation for the API

anthonyst91, 2018-02-08:
Hello,

Thank you for your great work. I'm the developer of Printoid for OctoPrint, Android app to control OctoPrint.

A lot of users are requesting me to support your plugin in Printoid. No problem for me, I already support the PSU control plugin. Just read your code and can see that you already support API to control your plugin.

Can you please document it? How does it work? I've tried to send the command "turnOn" but it says that I need to provide the IP because that's mandatory. Why providing the IP, whereas this value is already set in your plugin config ?

Thank you in advance !

Anthony

jneilliii, 2018-02-08:
The ip is required because you can have multiple plugs configured within the settings and just sending a turnOn command wouldn't know which plug you were trying to turn on.

closed 2018-02-08 by jneilliii

rorosentrater, 2019-09-15:
Is the API actually even finished? I am also sending POSTS to 'my.print.server.ip/plugin/tplinksmartplug' with what I think is a valid body:
```
{
    'command': 'turnOff',
    'ip': 'my.smart.plugs.ip'
}
```
getting a 204 response....but nothing happens. The GET request (which needs a body?) is similarly returning a 204 with no data when was expecting the status of the specified plug to be returned.