jneilliii/OctoPrint-TPLinkSmartplug #47: Feature Request: Power off after x minutes of print completion

jbishop129, 2018-02-22:
It would be great to have the ability to say "Turn off plug [x] Minutes after print job completion".  after [x] minutes, it should check again to make sure another job hasn't been started. If the printer is still idle, turn off the plug. 

I do this currently via OpenHAB, where I'm able to monitor the printer via MQTT, and I do exactly the above logic. Would be great to have this built-in to the plugin!

jneilliii, 2018-02-22:
With the exception of checking if another job has started, you can accomplish this using the gcode options.

jbishop129, 2018-02-22:
Thanks for pointing this out. Is it possible to have the time delay in the
gcode options? I was initially thinking as well, "just insert the shutdown
M80 / M81" (whichever one is the shutdown) at the end of my gcode, but I
figured this would immediately shut the machine off, vs leaving it on for a
few minutes, in case I was intending to start another job right away.  Then
again, I suppose one just needs to power it back on and let it
auto-connect, which is also trivial. Now I'm torn!

BTW, I just found your gitHub yesterday, and I have to thank you for your
massive OctoPrint contributions to the community. You sir, are a gentleman
and a scholar :)

jneilliii, 2018-02-22:
Thanks, I'll have to review this plugin specifically, they all run together, but one of them I know has timing delays with gcode enabled.

jneilliii, 2018-02-22:
Just looked, yes it's in there. If you enable gcode you can set delays.

jbishop129, 2018-02-22:
Ha! I'm not sure how I didn't notice that! Thanks again for this plugin!

closed 2018-02-22 by jbishop129

jneilliii, 2018-02-22:
Sure, let me know if you run into any issues.