jneilliii/OctoPrint-TPLinkSmartplug #50: Cancel delayed shutdown by invoking M80

jaapp, 2018-06-09:
I have TPLink Smartplug setup so it will shutdown with a delay of about 10 minutes. Sometimes when I run batches this shutdown kicks in when the next print has already started.

It would be great if the M80 cancels the active shutdown timer so I could invoke this command on the start of every print.

jneilliii, 2018-06-09:
That would be an interesting approach, but in lieu of that method, what if I were to implement it the same way I'm doing in the TasmotaMQTT plugin where if you have Warn While Printing enabled for the plug it won't power off via M81?

jneilliii, 2018-06-09:
I haven't had a chance to test it yet, but 0.9.4 branch should enable this `Warn While Printing` option.  If you want you could test by installing from the below URL in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.4.zip

jaapp, 2018-06-09:
I've seen the warning option, and while it is useful, it still forces me to restart octoprint to prevent the shutdown (or am i mistaken?). Sometimes this is a hassle because that will also reset the printer and you'll lose whatever values you have in the memory. There is no way to cancel an active delayed shutdown.

One solution I thought of was using the native 'shutdown after x seconds'-function of the TPLink.  
 I believe this is the 'countdown'-command. This countdown can also be cancelled. M80 could be set up to send the 'cancel countdown'-command. And maybe canceling the countdown is a default behaviour of the TPlink when it is send the 'on'-command?

It also allows the user to cancel the shutdown himself via the KASA-app.

jneilliii, 2018-06-09:
You would only have to restart octoprint to install the update. Once installed octoprint won't process the off command if that option is set and the printer is printing. It should totally resolve your issue, but until you try it we won't know.

jaapp, 2018-06-14:
That sounded exactly like what i needed so I tested it out.

I set up a shot delay
[image]

And immediately started a print after the last was finished. 

The result was that the printer still shut down during the new print. And no warning was shown.

jneilliii, 2018-06-14:
Just to confirm, you reinstalled the plugin using this url?

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.4.zip

Restated/ Reloaded etc.?

jneilliii, 2018-06-14:
Sorry about that.  Just noticed that I didn't change the timer call to the new function.  I just pushed an update to that same 0.9.4 branch that should actually work.  About to test it now.

jneilliii, 2018-06-14:
I can confirm that this latest update is actually working.  Just did a quick test like you, set the delay to 60 seconds and after 3 minutes my plug is still powered on and an entry is added to the octoprint.log...

```
octoprint.plugins.tplinksmartplug - INFO - Not powering off TP-LINK-CR10 because printer is printing.
```

jneilliii, 2018-06-15:
I just pushed an official 0.9.4 version.  If you reinstall, use the following url in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/master.zip

closed 2018-06-15 by jneilliii

jaapp, 2018-06-16:
Thats pretty fantastic. Thanks!

One suggestion i have is to change the text from "Warn While Printing" to "Prevent shutdown while printing". It's more descriptive of the function, the warning is just a side effect.