jneilliii/OctoPrint-TPLinkSmartplug #52: Gcode command that would control lights. Enhancement

sirprintsalot8, 2018-08-03:
I have looked and have not found a solution. I would like for a way to use this set up to turn on and off a light with this setup. If you use the M80 with the IP of a plug with a light it will turn on the light but it also resets the printer.

jneilliii, 2018-08-03:
I'll see what I can do about utilizing a custom gcode command.

sirprintsalot8, 2018-08-04:
Thanks for the super fast work up on this. Im getting ready to install and try. Ill post after a test print.

sirprintsalot8, 2018-08-04:
Ran it and it turn the light on. But... after the light turned on it homed and started the bed warm up and restarted like it did with the M80 code.

@TPLINKON 192.168.1.35 ; Turns Printer Light On
; begin starting script
; home -> heat bed -> delay 2 minutes for PINDA warming -> mesh level -> heat extruder -> prime
M115 ; use the latest firmware version
G28 W ; home all axes without mesh bed leveling
M140 S[bed1_temperature] ; start heating the bed
M190 S[bed1_temperature] ; wait until bed heated
G4 S120 ; wait two minutes
M104 S[extruder0_temperature] ; start heating extruder
G80 ; run mesh bed leveling routine
M109 S[extruder0_temperature] ; wait for extruder temperature
G1 Y-3.0 F1000.0 ; prepare to prime
G92 E0 ; reset extrusion distance
G1 X60.0 E9.0  F1000.0 ; priming
G1 X100.0 E12.5  F1000.0 ; priming
M900 K30; PLA

This is my starting script that I am using in S3D. Dont know if it is something I am imputing or another error.

[image]

sirprintsalot8, 2018-08-04:
Scratch all of what I said! Sorry about my poor testing before the comment. I unchecked the auto connect and auto disconnect on the printer light and not the printer plug itself and it turned on the light and has done all its starting sets. Will confirm on the light off at the end of the print.(its a short test cube)

sirprintsalot8, 2018-08-04:
Can now confirm that it did shut it off at that end of the print. Thanks again for the fast perk add-on.

jneilliii, 2018-08-04:
Would be curious if that was the original problem the whole time and M80/M81 commands would have worked with those options unchecked as well? I know there could be some printers out there that utilize those commands for the purpose of powering off the printer which is the intended purpose of those commands in the firmware anyway. Doesn't really matter, better to have more options anyway for special use cases I suppose.

closed 2018-08-04 by jneilliii