jneilliii/OctoPrint-TPLinkSmartplug #53: Shutdown Raspberry before switching off plug

Tronnic, 2018-08-04:
Hello, I'm running a Raspberry Pi 3B with Touchscreen & TouchUI. When I turn off the printer with my TP Link Plug the Pi goes down as well. This is causing some major issues for me, since sometimes my installation gets corrupted. 

Is there a way to start a 30 second timer on the plug itself, so the raspberry can shut down successfully? When I control the Plug with my smartphone there is an option for a timer so it must be possible somehow. 

Thank you a lot for this plugin. If the above works it would be really great!

jneilliii, 2018-08-04:
Is your Pi powered from your printer's board? There are options for running system commands, so you could issue a sudo shutdown command to the pi, but once the pi is powered off it won't be able to send the command to the plug to power it off as octoprint will no longer be running. If you are powering your pi off the board then unfortunately none of the options I know of would work.

jneilliii, 2018-08-04:
Good point. I didn't think of that. Let me check into it and get back to you.

Tronnic, 2018-08-04:
That would be amazing! Thank you <3

jneilliii, 2018-08-05:
I think I have it worked out if you don't mind testing.  Use the below url in plugin manager to install.  

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.6.zip

Once installed set your plug's settings to something like this, using your plug's IP of course.

[image]

The GCODE off delay is what will be used to send the countdown timer to the plug.  Then in your end GCODE commands use `@TPLINKDELAYOFF 192.168.0.130` and it should achieve the goal.

In the above screenshot that will automatically disconnect octoprint from printer, send a countdown timer to the plug telling it to power off in 60 seconds and then immediately run the shutdown command.

Tronnic, 2018-08-05:
I tried a few times. First problem I have is since updating your plugin to 0.9.6 my whole webinterface runs horribly slow. Also needs about 5 minutes to load. 

I still tried my best to give it a shot but I couldn't make it do what I wanted it to do. 
When entering "@TPLINKDELAYOFF <IP>" into my end gcode (OctoPrint Settings > Printer > GCODE Scripts > After print job completes" the printer shuts down immediately after the print is done. 

When enterting into the plugin "Shutdown Printer" nothing happens.
[image]

jneilliii, 2018-08-05:
Disable the shutdown printer plugin. That one always forces a M81 command.

jneilliii, 2018-08-05:
I'm going to make some changes because I realized that when pressing the button it would still use the old methods but I want to make sure it's working.

Tronnic, 2018-08-05:
I will try right now, give me a moment. :)

Tronnic, 2018-08-05:
YouTube Video: https://www.youtube.com/watch?v=PtHjfr2-nD0
[image]

Works! Thanks for all the effort you put into it. I'm so happy now, hehe :)

jneilliii, 2018-08-05:
Great news. I'll release the updated version in a bit.

closed 2018-08-05 by jneilliii

frabbba, 2023-12-09:
Hello guys,
@Tronnic 
May I ask which type of TP-link plug you used for this purpose described above? I wanna do the same and looking for a known to be compatible plug.
@jneilliii 
As I understood you are the developer of this plugin. Could you tell me please with which TP-LINK plug would this method work? Any TP-link would do the job?
Thanks for the Feedback in advance! 🙂

jneilliii, 2023-12-09:
> Currently known compatible models are the HS100, HS103, HS105, HS107, HS110, HS300, KP105, KP303, KP115. Other Kasa app based devices may work. Tapo series devices will not work with this plugi

Tronnic, 2024-01-26:
> Hello guys, @Tronnic May I ask which type of TP-link plug you used for this purpose described above? I wanna do the same and looking for a known to be compatible plug. @jneilliii As I understood you [...]

Sorry for the late answer! I was using a TP-Link Kasa HS110 Plug. Still in use, still working since 2018! :-)