jneilliii/OctoPrint-TPLinkSmartplug #54: Bad icon color when power is on and can't power off TP-Link

Patrick60, 2018-08-06:
Hi,

With the latest release (0.9.6), the color icon of this plugin is no more green when power up but gray. And clicking on it doesn't power off the TP-Link plug, it does nothing. On the other hand, using the gcode M81 followed by the IP address is still working.

[image]

jneilliii, 2018-08-06:
Release notse mentioned that previously configured plugs would be removed. Did you add your ip info back in?

Patrick60, 2018-08-06:
Yes, I did, and I restarted my computer since and it's working now. Sorry for the trouble, everything's fine.

Patrick60, 2018-08-06:
Well, I left the computer for let's say half an hour, with the switch on and when I came back, I had the issue, the icon is gray and when I click on it, it turns green for a second and then goes gray again and I can't switch off the plug anymore. I'm gonna a try with Microsoft Edge (now I'm on Chrome).

jneilliii, 2018-08-06:
I'm not quite seeing the same thing here with Chrome.  Resfreshing the page does typically put it back in line with where it is supposed to be.  To better understand what is going on though, do you have polling enabled?  Can you enable debug logging and then upload the plugin_tplinksmartplug_debug.log file once you notice it switches.

Patrick60, 2018-08-06:
I'm going to try to reproduce it and once I understand how to, I'll activate the logs and send them to you. Polling is currently not activated.
It could be due to a system inactivity for some time though in the power settings only my screen enters in sleep mode after 15 min and my PC never but it's the only thing I see in common when it happened twice.

jneilliii, 2018-08-10:
Enable logging ahead of time so when you do notice we have the details.  Activating the logs later may not help.

Patrick60, 2018-08-11:
It will be a mystery, finally I couldn't reproduce it, sorry again for the trouble. I close this 'false' issue.

closed 2018-08-11 by Patrick60