jneilliii/OctoPrint-TPLinkSmartplug #57: Compatibility with TP-Link SmartBulb

m41miller, 2018-09-05:
Could this be configured to work with the TP-Link SmartBulbs?  Specifically the KB100?

jneilliii, 2018-09-08:
It might be possible, just depends on if those model bulbs use the same TP-Link Device Debug Protocol (TDDP) to accept commands.  Would just have to figure out what the commands necessary to send to the bulb are.  I didn't reverse engineer the protocol for the plug, just integrated it into an OctoPrint plugin, but looks like there are other model bulbs that would work based on this fork of the source I originally used for the plug communication.

https://github.com/hachisanju/Dumb_Bulb

I think because of the plugin name though, I would more than likely fork this plugin to a new one just for the bulb option and not "cross contaminate" code.

m41miller, 2018-09-10:
Cool.  Thanks @jneilliii .

jneilliii, 2018-09-22:
I've finally got around to getting a version of that may work.  I'll need your help in verifying and debugging any issues though since I don't have one of these bulbs.  Some of the settings may not make sense to keep like Auto Connect/Disconnect from printer, since power isn't getting interrupted and probably the system commands aren't really necessary either.  

Anyway, I've got it posted in a separate repository.  Install using the following url.  Enable debug logging in settings and then post the file plugin_tplinksmartbulb_debug.log to a new issue post over [**here**](https://github.com/jneilliii/OctoPrint-TPLinkSmartBulb/issues) .

https://github.com/jneilliii/OctoPrint-TPLinkSmartBulb/archive/master.zip

closed 2018-09-22 by jneilliii