jneilliii/OctoPrint-TPLinkSmartplug #59: Possible to Link Plug 'Off' Command to Temperature Failsafe Plugin? Shell Script Command?

disposablehero2, 2018-09-11:
Hi, 

Is it possible to link this to the temperature failsafe plugin such that an over-temp would turn off the plug?

I already have an Arduino with separate thermistors and MQ2 smoke sensors linked up to a 433mh socket but this seems like it would be cleaner.

The temperature failsafe plug allows a shell script to be executed upon overtemp. If I could just figure out what the shell script equivalent of pressing the thunderbolt icon is then it seems it would be a nice clean additional safety feature.

Not much of a coder I am afraid.

All the best.

jneilliii, 2018-09-12:
I think you'd be able to do this with a wget command maybe.  Let me do come testing on my end and I'll let you know.

jneilliii, 2018-09-12:
Yep, I think this will work for you.  Hopefully the instructions are easy enough to follow.

You will first need to go into you user settings and generate your api key.  Click your name in the top right corner and select "User Settings". In the dialog that opens press the refresh button to generate your key.  You will need to copy that into our shell script later.

[image]

Then once you have that information you'll SSH to your pi and run the following commands.

```
cd ~/
nano turn_off_printer.sh 
```

This will open a text editor to allow you to enter the commands for your shell script.  Type in the following into the blank document that was created from the above command. Making sure to replace `<put user api key here>` with the one you created earlier and `<put plug ip here>` with the ip of your tplink plug.

```
#!/bin/bash 
curl -H "X-Api-Key: <put user api key here>"  -H 'Content-Type: application/json' --data '{"command":"turnOff","ip":"<put plug ip here>"}' -X POST http://localhost:5000/api/plugin/tplinksmartplug
```

Hold down the ctrl key and press the x key on your keyboard and the answer `Y` to the prompt to save the file.  Now we have to make that file executable by running the following command.

```
chmod +x turn_off_printer.sh 
```

That should be it.  Now in TemperatureFailsafe plugin enter the following into the command to run.

```
/home/pi/turn_off_printer.sh
```

disposablehero2, 2018-09-12:
You are awesome! Thanks. I messed it up the first time but deleted it all and started again and it works!

Thanks again!

jneilliii, 2018-09-12:
Cool.

closed 2018-09-12 by jneilliii

Brainrock, 2019-01-19:
I have problems setting it up. The TP Link doesn't turn off. I sadly have no experience with raspberry pis so please help me. I will now describe what i have done:
- Generated API Key - site on my octoprint looks different (with QR- Code) but it generated one
- downloaded PuTTY and connected to the Pi.. followed the instructions (saved and executed) but entered the wrong ip of my TP Link.. connected again via PuTTY and simply put:
cd ~/
nano turn_off_printer.sh 
in again.. changed the ip to the correct one.. saved and again entered:
chmod +x turn_off_printer.sh 
(do i need to save something befor closing PuTTY?)
- finally entered:
 /home/pi/turn_off_printer.sh
in octoprint.. tested.. plugin works and shows errors put Tp Link is still active.
Any idea what i did wrong?
I got this Tp LINK : https://www.amazon.de/TP-Link-Steckdose-funktionieren-erforderlich-Steuern/dp/B06W586CDZ/ref=sr_1_1_sspa?ie=UTF8&qid=1547912955&sr=8-1-spons&keywords=tp+link+steckdose&psc=1

jneilliii, 2019-01-19:
You have to install this plugin and configure your plug in its settings for the script to work.

Brainrock, 2019-01-19:
Ok I don't know how to do this. (Sorry i honestly don't get it)
I will attach a pic of the settings screen. I can't find something to add my plug or configure the plugin at all.
I just installed the plugin via the Plugin Manager in the Octoprint Settings (Plugin Repository).
[image]

jneilliii, 2019-01-19:
Yeah @Brainrock, you don't have the OctoPrint-TPLinkSmartplug plugin installed so that's why it's not working.