jneilliii/OctoPrint-TPLinkSmartplug #60: Cannot shut off relay with Terminal commands

rullywowr, 2018-09-16:
Thanks for the great plugin! It is working well with my MQTT and Tasmota-ized Sonoff S31 and coexisting peacefully with my Home Assistant setup.

My only issue is that I cannot get it to auto power off, say after a print. I'm trying to follow the instructions regarding adding GCODE to the after print section but no dice. It is a single relay device so I'm wondering if this may have something to do with it?

I have sent all following in the Terminal and none of them seem to work to power off the relay:

(192.168.1.74 is the IP of my Sonoff/Tasmota Module)

M81
M81 192.168.1.74
M81 192.168.1.74 0
M81 192.168.1.74 1
M81 192.168.1.74 2
M81 192.168.1.74 3

Thanks!

jneilliii, 2018-09-16:
There was a change to the tasmota firmware at one point for single relay  devices. They added a SetOption26 to version 5.14.0 which when on means single relay devices can use 1 as the index. So in theory if you uncheck the single relay device iron in the plugin settings and set your index to 1 it should work. Your gcode would be `M81 192.168.1.74 1`.

jneilliii, 2018-10-01:
Can you show me your settings for the plug please?

rullywowr, 2018-10-01:
Sure, thank you for looking at this. Attached are settings of the plugin as well as the Octoprint GCODE terminal.
[image]
[image]

jneilliii, 2018-10-01:
Yep, you just need to check the box for GCODE Trigger, otherwise the plugin just skips right over the plug and doesn't do anything.

closed 2018-10-01 by jneilliii