jneilliii/OctoPrint-TPLinkSmartplug #62: Plugin removes custom system commands

viljamip, 2018-09-27:
I added custom System commands that use the hs100.sh script to turn the printer on and off. (I those can be used remotely with the Telegram plugin.)

If I use this plugin's on/off button (bolt-icon) from the Octoprint UI all of my custom System Commands are deleted for some reason.

Here is maybe a related issue where many users have mentioned that they use the HS100:
https://github.com/Salandora/OctoPrint-SystemCommandEditor/issues/5

jneilliii, 2018-09-27:
Do you have the system command editor plugin installed? Can you document how your system commands are configured so I can test on my end to see if I can reproduce. My plugin doesn't do anything related to system commands so not sure how it would be effecting it.

viljamip, 2018-09-27:
Yes I used the system command editor plugin. Here are my system commands copied from config.yaml:
system:
  actions:
  - action: 'on'
    command: /home/pi/Viljami/./hs100.sh -i 192.168.1.110 on
    name: 'On'
  - action: 'off'
    command: /home/pi/Viljami/./hs100.sh -i 192.168.1.110 off
    confirm: Power off?
    name: 'Off'
  - action: divider
  - action: focus
    command: v4l2-ctl -c focus_auto=0; v4l2-ctl -c focus_absolute=55
    name: Manual focus

I'm not completely sure that it is infact your plugin that is causing this but my system commands disappeared right after I had turned the printer on and off with your plugin's button. They were deleted multiple times so I feel that this issue is repeatable at least on my Octoprint. For now I disabled your plugin and I use only my system commands and it seems to be working.

jneilliii, 2018-09-27:
I'd be curious if the system command editor could be disabled after setting your commands up.  I think it writes directly back to config.yaml what it needs and there is no need to have it loaded unless you need to change your settings.

viljamip, 2018-09-27:
I disabled system command editor from the plugin manager and indeed I can't reproduce the issue anymore. I'll report back to you if my commands still get deleted but I think your guess was right. Thanks.

jneilliii, 2018-09-27:
Great, feel free to reply back here if it comes back. Closing for now.

closed 2018-09-27 by jneilliii