jneilliii/OctoPrint-TPLinkSmartplug #66: not working with u'module not support' error

ericpskl, 2018-11-03:
Hello!

The plug-in is failing with error 'module not support' on my install of Octopi.  The HS100 unit does not click on/off when the lightning bolt icon is clicked in the web UI.

Fresh installs of Octoprint and the OctoPrint-TPLinkSmartplug plugin; new HS100 unit just installed.  

- The HS100 works fine through the TPLink Kasa app;
- Able to ping ip address of the smartplug from CLI of Octopi.
- I am able to turn the plug on and off using the tplink-smartplug package (https://github.com/softScheck/tplink-smartplug), which I installed on the Octopi unit to test stuff.

> root@octopi:~/tplink-smartplug# ./tplink_smartplug.py  -t 192.168.1.32 -c on
> Sent:      {"system":{"set_relay_state":{"state":1}}}
> Received:  {"system":{"set_relay_state":{"err_code":0}}}
> root@octopi:~/tplink-smartplug# ./tplink_smartplug.py  -t 192.168.1.32 -c off
> Sent:      {"system":{"set_relay_state":{"state":0}}}
> Received:  {"system":{"set_relay_state":{"err_code":0}}}

Debug Log excerpt follows.  Thanks in advance!!!

> [2018-11-03 13:11:55,648] DEBUG: Turning off 192.168.1.32.
> [2018-11-03 13:11:55,652] DEBUG: {'autoConnect': False, 'ip': '192.168.1.32', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'gcodeEnabled': False, 'gcodeOnDelay': 0, 'emeter': {'get_daystat': {'err_msg': 'module not support', 'err_code': -1}, 'get_realtime': {'err_msg': 'module not support', 'err_code': -1}}, 'countdownOnDelay': 0, 'label': 'TAZ', 'warnPrinting': True, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'countdownOffDelay': 0, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'autoDisconnect': False, 'useCountdownRules': False, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': 10, 'currentState': 'on', 'sysCmdOn': False, 'sysCmdOffDelay': 0}
> [2018-11-03 13:11:55,653] DEBUG: IP 192.168.1.32 is valid.
> [2018-11-03 13:11:55,738] DEBUG: Sending command {"system":{"set_relay_state":{"state":0}}} to 192.168.1.32
> [2018-11-03 13:11:55,743] DEBUG:   -"system":{"set_relay_state":{"err_code":0}}}
> [2018-11-03 13:11:55,747] DEBUG: Checking status of 192.168.1.32.
> [2018-11-03 13:11:55,749] INFO: {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{},"get_daystat":{"month":11,"year":2018}}}
> [2018-11-03 13:11:55,751] DEBUG: IP 192.168.1.32 is valid.
> [2018-11-03 13:11:55,890] DEBUG: Sending command {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{},"get_daystat":{"month":11,"year":2018}}} to 192.168.1.32
> [2018-11-03 13:11:55,896] DEBUG:  a"system":{"get_sysinfo":{"sw_ver":"1.5.1 Build 171109 Rel.165709","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"B0:4E:xxxxxxx","dev_name":"Smart Wi-Fi Plug","alias":"Taz","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-40,"led_off":0,"longitude_i":-xxxxxxx,"latitude_i":xxxxxxxx,"hwId":"0DC28CDxxxxxxxxB8B68277E","fwId":"00000000000000000000000000000000","deviceId":"80065xxxxxxxx5195B779A","oemId":"80CB8C7ED4CAD7DA4570979BFF731BA1","err_code":0}},"emeter":{"get_realtime":{"err_code":-1,"err_msg":"module not support"},"get_daystat":{"err_code":-1,"err_msg":"module not support"}}}
> [2018-11-03 13:11:55,902] INFO: {u'emeter': {u'get_daystat': {u'err_msg': u'module not support', u'err_code': -1}, u'get_realtime': {u'err_msg': u'module not support', u'err_code': -1}}, u'system': {u'get_sysinfo': {u'oemId': u'80Cxxxxxxxxxxx1BA1', u'dev_name': u'Smart Wi-Fi Plug', u'longitude_i': -765649, u'on_time': 0, u'feature': u'TIM', u'fwId': u'00000000000000000000000000000000', u'icon_hash': u'', u'relay_state': 0, u'hw_ver': u'2.0', u'type': u'IOT.SMARTPLUGSWITCH', u'led_off': 0, u'hwId': u'0DC28xxxxxxxxxx68277E', u'latitude_i': xxxxxxxxx, u'mac': u'B0:4E:xxxxxxxxx', u'active_mode': u'none', u'deviceId': u'800654D14E738xxxxxxxxxx', u'sw_ver': u'1.5.1 Build 171109 Rel.165709', u'updating': 0, u'alias': u'Taz', u'rssi': -40, u'model': u'HS100(US)', u'err_code': 0}}}

jneilliii, 2018-11-03:
Are there any errors in your octoprint.log file?

ericpskl, 2018-11-03:
Hi jneilliii,

It doesn't appear so:  

> root@octopi:/home/pi/.octoprint/logs# grep -i tplink octoprint.log
> 2018-11-03 12:39:35,911 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/master.zip
> | !TP-Link Smartplug (0.9.7) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
> |  TP-Link Smartplug (0.9.7) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
> 2018-11-03 12:40:13,749 - octoprint.plugins.tplinksmartplug - INFO - TPLinkSmartplug loaded!

jneilliii, 2018-11-03:
That is odd. The error is a reference to the call for energy monitoring changes I just implemented, but the debug log is showing everything communicating properly. I have both hs100 (plugin was developed with this from the beginning) and hs110 (recently added for monitoring support in plugin) and am unable to reproduce the issue. Have you tried restarting octoprint and force reloading the web interface?

ericpskl, 2018-11-03:
Yes, have rebooted the pi several times.   

I just did a test using tshark and confirmed that there is a TCP session created between the pi and the TPLink unit every time the button in the web interface is clicked.   

Further testing shows that if the TPLink unit is in the Off state, it will turn on using the lightning bolt icon in the web interface, however it will not turn it back off via the plugin.   Apologies for not noticing this earlier. 

Thanks for your help!

jneilliii, 2018-11-04:
When it does this is it an endless spinning bolt or throws the popup error?

ericpskl, 2018-11-05:
The bolt spins once when you click it;  there is no error created  in the web UI (no popup error.)

jneilliii, 2018-11-05:
Could you try the upcoming release to see if the issue is the same for you?  I think I found the issue related to the HS100s and did some additional tweaking of the callbacks.  You can install using the following URL in the plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.8.zip

ericpskl, 2018-11-06:
Working perfectly using 0.9.8!  Thanks for your help! :)

jneilliii, 2018-11-19:
0.9.8 officially released.

closed 2018-11-19 by jneilliii