jneilliii/OctoPrint-TPLinkSmartplug #68: Not controlling plug after Update

Cliffhorner, 2018-11-04:
Hi, This morning I updated the plugin to 0.9.7. After the update I had to re-enter the ip address etc but now it will not control the plug. When I first log in and I click on the lightening icon it turns red and turns the printer on but that's all it will do. If I click it again it briefly flashes green, disconnects and reconnects Octoprint with the printer but does not turn the plug off. The lightening icon stays red.

I have uninstalled the plugin and reinstalled but it is still the same. I have rebooted both PC and pi but still the same. Is there a way I can go back to the previous version?

jneilliii, 2018-11-04:
If you go to the releases link at the top of the page you can get the link for any version by right-clicking the source zip url and selecting copy hyperlink. Then paste that into the plugin manager. Could you do me a favor though and enable debug logging and attempt to toggle your plug and then send me your plugin_tplinksmartplug_debug and octoprint log files before upgrading please?

Cliffhorner, 2018-11-04:
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2545927/plugin_tplinksmartplug_debug.log)

jneilliii, 2018-11-04:
Thanks for the log @Cliffhorner, in your octoprint.log are there any Traceback errors? The log looks like it should be ok.

devildant, 2018-11-18:
hello @jneilliii, 
a friend asked me for help on your plugin, because the button (on / off) tplink in the nav bar does not work anymore.
I downgrade to 0.9.6 and it works.
he gets this trace in octoprint.log : 
```
2018-11-18 14:03:28,722 - octoprint - ERROR - Exception on /api/plugin/tplinksmartplug [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1140, in decorated_view
    return flask_login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 99, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 170, in on_api_command
    self.check_status("{ip}".format(**data))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 143, in check_status
    response = self.sendCommand(check_status_cmnd, plugip)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 240, in sendCommand
    return json.loads(self.decrypt(data[4:]))
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 380, in raw_decode
    obj, end = self.scan_once(s, idx)
ValueError: Expecting object: line 1 column 1444 (char 1443)
2018-11-18 14:03:28,728 - tornado.access - ERROR - 500 POST /api/plugin/tplinksmartplug (::ffff:192.168.1.28) 109.32ms
```

and that in debug file : 
```
[2018-11-18 13:59:54,752] DEBUG: Checking status of 192.168.1.78.
[2018-11-18 13:59:54,753] INFO: {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{},"get_daystat":{"month":11,"year":2018}}}
[2018-11-18 13:59:54,754] DEBUG: IP 192.168.1.78 is valid.
[2018-11-18 13:59:54,804] DEBUG: Sending command {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{},"get_daystat":{"month":11,"year":2018}}} to 192.168.1.78
[2018-11-18 13:59:54,813] DEBUG: « N˜"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.2.5 Build 180410 Rel.182657","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS110(FR)","mac":"50:C7:BF:B6:B5:E2","deviceId":"8006C3B4ECBC9CEC1F0055956F180E3C18B7AF14","hwId":"797CB3D09D9A83B1C43A623C6DD103C2","fwId":"00000000000000000000000000000000","oemId":"0D13F703F04C040DC3D27DB248C01A1A","alias":"Norris","dev_name":"Wi-Fi Smart Plug With Energy Monitoring","icon_hash":"","relay_state":1,"on_time":5440,"active_mode":"none","feature":"TIM:ENE","updating":0,"rssi":-44,"led_off":0,"latitude":0,"longitude":0}},"emeter":{"get_realtime":{"current":0.069415,"voltage":243.070970,"power":6.174200,"total":0.030000,"err_code":0},"get_daystat":{"day_list":[{"year":2018,"month":11,"day":1,"energy":1.200000},{"year":2018,"month":11,"day":2,"energy":0.923000},{"year":2018,"month":11,"day":3,"energy":0},{"year":2018,"month":11,"day":4,"energy":0.752000},{"year":2018,"month":11,"day":5,"energy":0.296000},{"year":2018,"month":11,"day":6,"energy":1.075000},{"year":2018,"month":11,"day":7,"energy":1.965000},{"year":2018,"month":11,"day":8,"energy":1.866000},{"year":2018,"month":11,"day":9,"energy":0.208000},{"year":2018,"month":11,"day":10,"energy":0},{"year":2018,"month":11,"day":11,"energy":0.127000},{"year":2018,"month":11,"day":12,"energy":0},{"year":2018,"month":11,"day":13,"energy":0},{"year":2018,"month":11,"day":14,"energy":0},{"year":2018,"month":11,"day":15,"energy":0},{
```

facet02, 2018-11-18:
Hi there, Version 9.7 is also not working for me. Plugin cant find the plug. 
Old Version 9.6 works fine (except sometimes the releais does not work, then i have to try it some times oder with the mobile APP to switch on/off and the it works again with the plugin. You know this issue or is maybe my plug broken?)

jneilliii, 2018-11-18:
Yeah I've been working on a fix for this already, you guys should see an update in the next day or two.

devildant, 2018-11-18:
thx @jneilliii

facet02, 2018-11-18:
Thanks for the support! Sometimes the "status" of the plug is not right (off instead of on and the other way around) - how short should be the status polling?

jneilliii, 2018-11-18:
I've seen that occasionally where the state does not show correct.  Usually just a refresh of the page is enough to get them back in order.  Haven't yet been able to figure out why this is happening. The value for polling can be as short as you want, but I probably wouldn't go any lower than 5 minutes.

jneilliii, 2018-11-18:
If you guys don't mind trying the 0.9.8 branch and let me know how it works for you that would be great.  You can install using the url below in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.8.zip

devildant, 2018-11-18:
> If you guys don't mind trying the 0.9.8 branch and let me know how it works for you that would be great. You can install using the url below in plugin manager.
> 
> https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.8.zip

I'm waiting for my friend's return and I tell you

> I've seen that occasionally where the state does not show correct. Usually just a refresh of the page is enough to get them back in order. Haven't yet been able to figure out why this is happening. 

I think I know, you must make an ajax call to the status in order to trigger a return of WebSocket in the oncomplete callback.
This is a problem when starting the page, as indicated in the bug report  (you participated, I think: p):
https://github.com/foosel/OctoPrint/issues/2898

for fix, you will have to update via ajax only during the first call (you must essentially obtain a status during your call, to display the status and not to go through the websocket, at least for the first one).

PR #70 (2018-11-19): 0.9.8
### Fixed
- Non Energy Monitoring devices were causing errors on status checks resulting in status update failure. Resolves #68, Might Resolve #69.

### Issues
- When toggling energy monitoring devices on/off the energy usage reported in the sidebar panel is from prior to the status change, so off plugs aren't reporting 0 as expected. Upon the next polling interval the power usage shoould be back in sync.
closed 2018-11-19 by jneilliii