jneilliii/OctoPrint-TPLinkSmartplug #71: Current status only changes when reloading

enigma2k, 2018-11-19:
Hello!

First off, I have to say thank you for this wonderful plugin! I'm using the plugin with 2 smart plugs. One is to power the printer itself and another one is connected with a light that allows me to turn the printer light on and off from within Octoprint. The problem, however, is that when I press the bolt to turn the light on, the bolt will stay red instead of automatically changing to green. This way I can't click it again in order to turn the light off but instead have to reload the browser so it is showing green and then I can click it again for the light to turn off. 

Can you make the status to change automatically once it is clicked? I know there is the polling feature but this still won't allow me to quickly turn on and off a light. You always have to wait for the polling.

jneilliii, 2018-11-19:
A recent update was just pushed that should help with this issue.  If you're not using version 0.9.8, please upgrade. Let me know otherwise.

enigma2k, 2018-11-20:
thank you, it is working now perfectly with 0.9.8. I had to manually install it using the URL since it was not available in the Plugin Manager.

closed 2018-11-20 by enigma2k