jneilliii/OctoPrint-TPLinkSmartplug #77: M80 command stops print

chipdice, 2018-11-29:
I've got the plugin to turn on my tp link smartplug by putting in the M80 command with the correct IP, however this causes my printout to immediately get cancelled. I'm using the original Prusa i3 MK3 printer. How do I avoid it cancelling the printout?

jneilliii, 2018-11-29:
Can you please share a screenshot of the plug's settings?  I think the issue might be related to the Auto Connect option maybe?  If not then you may want to attempt using the custom action command `@TPLINKON` instead of M80.

chipdice, 2018-11-30:
[image]

chipdice, 2018-11-30:
Should I try with autoconnect off to see?

jneilliii, 2018-11-30:
Yeah, I'd try turning off the option to auto connect.

chipdice, 2018-11-30:
ok, will do and let you know. Thanks

chipdice, 2018-11-30:
Turning Autoconnect off worked. Thanks

closed 2018-12-21 by stale[bot]