jneilliii/OctoPrint-TPLinkSmartplug #78: M81 not working with HS110

ikoutone, 2018-12-05:
I have put both at the end of the g-code to test if it works, but nope. Neither of these work.
M81 192.168.0.32 ; turn printer off
@TPLINKOFF 192.168.0.32

Plug was turned on from octoprint nav bar. 
> [2018-12-05 22:00:05,721] DEBUG: Turning on 192.168.0.32.
[2018-12-05 22:00:05,725] DEBUG: {'autoConnect': True, 'ip': '192.168.0.32', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'gcodeEnabled': True, 'gcodeOnDelay': 0, 'emeter': None, 'countdownOnDelay': 0, 'label': 'Tulostin', 'warnPrinting': True, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'countdownOffDelay': 0, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'autoDisconnect': False, 'useCountdownRules': False, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': '5', 'currentState': 'off', 'sysCmdOn': False, 'sysCmdOffDelay': 0}
[2018-12-05 22:00:05,726] DEBUG: IP 192.168.0.32 is valid.
[2018-12-05 22:00:05,753] DEBUG: Sending command {"system":{"set_relay_state":{"state":1}}} to 192.168.0.32
[2018-12-05 22:00:05,754] DEBUG:   -"system":{"set_relay_state":{"err_code":0}}}
[2018-12-05 22:00:05,756] DEBUG: 0
[2018-12-05 22:00:05,758] DEBUG: Checking status of 192.168.0.32.
[2018-12-05 22:00:05,759] DEBUG: {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{}}}
[2018-12-05 22:00:05,760] DEBUG: IP 192.168.0.32 is valid.
[2018-12-05 22:00:05,946] DEBUG: Sending command {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{}}} to 192.168.0.32
[2018-12-05 22:00:05,952] DEBUG:  "system":{"get_sysinfo":{"sw_ver":"1.5.4 Build 180815 Rel.121440","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS110(EU)","mac":"AC:84:C6:92:3F:AE","dev_name":"Smart Wi-Fi Plug With Energy Monitoring","alias":"Tulostin","relay_state":1,"on_time":0,"active_mode":"none","feature":"TIM:ENE","updating":0,"icon_hash":"","rssi":-36,"led_off":0,"longitude_i":277094,"latitude_i":642156,"hwId":"044A516EE63C875F9458DA25C2CCC5A0","fwId":"00000000000000000000000000000000","deviceId":"8006ECDCE526845228FC85AA099401541A2406A8","oemId":"1998A14DAA86E4E001FD7CAF42868B5E","next_action":{"type":-1},"err_code":0}},"emeter":{"get_realtime":{"voltage_mv":230271,"current_ma":33,"power_mw":0,"total_wh":4129,"err_code":0}}}
[2018-12-05 22:00:06,009] DEBUG: Checking status of 192.168.0.32.
[2018-12-05 22:00:06,010] DEBUG: {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{}}}
[2018-12-05 22:00:06,012] DEBUG: IP 192.168.0.32 is valid.
[2018-12-05 22:00:06,090] DEBUG: Sending command {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{}}} to 192.168.0.32
[2018-12-05 22:00:06,097] DEBUG:  "system":{"get_sysinfo":{"sw_ver":"1.5.4 Build 180815 Rel.121440","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS110(EU)","mac":"AC:84:C6:92:3F:AE","dev_name":"Smart Wi-Fi Plug With Energy Monitoring","alias":"Tulostin","relay_state":1,"on_time":0,"active_mode":"none","feature":"TIM:ENE","updating":0,"icon_hash":"","rssi":-37,"led_off":0,"longitude_i":277094,"latitude_i":642156,"hwId":"044A516EE63C875F9458DA25C2CCC5A0","fwId":"00000000000000000000000000000000","deviceId":"8006ECDCE526845228FC85AA099401541A2406A8","oemId":"1998A14DAA86E4E001FD7CAF42868B5E","next_action":{"type":-1},"err_code":0}},"emeter":{"get_realtime":{"voltage_mv":230271,"current_ma":33,"power_mw":0,"total_wh":4129,"err_code":0}}}
[2018-12-05 22:00:12,217] DEBUG: Checking status of 192.168.0.32.
[2018-12-05 22:00:12,219] DEBUG: {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{}}}
[2018-12-05 22:00:12,221] DEBUG: IP 192.168.0.32 is valid.
[2018-12-05 22:00:12,300] DEBUG: Sending command {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{}}} to 192.168.0.32
[2018-12-05 22:00:12,305] DEBUG:  "system":{"get_sysinfo":{"sw_ver":"1.5.4 Build 180815 Rel.121440","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS110(EU)","mac":"AC:84:C6:92:3F:AE","dev_name":"Smart Wi-Fi Plug With Energy Monitoring","alias":"Tulostin","relay_state":1,"on_time":6,"active_mode":"none","feature":"TIM:ENE","updating":0,"icon_hash":"","rssi":-36,"led_off":0,"longitude_i":277094,"latitude_i":642156,"hwId":"044A516EE63C875F9458DA25C2CCC5A0","fwId":"00000000000000000000000000000000","deviceId":"8006ECDCE526845228FC85AA099401541A2406A8","oemId":"1998A14DAA86E4E001FD7CAF42868B5E","next_action":{"type":-1},"err_code":0}},"emeter":{"get_realtime":{"voltage_mv":230470,"current_ma":166,"power_mw":15907,"total_wh":4129,"err_code":0}}}
[2018-12-05 22:00:51,045] DEBUG: Checking status of 192.168.0.32.
[2018-12-05 22:00:51,047] DEBUG: {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{}}}
[2018-12-05 22:00:51,048] DEBUG: IP 192.168.0.32 is valid.
[2018-12-05 22:00:51,194] DEBUG: Sending command {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{}}} to 192.168.0.32
[2018-12-05 22:00:51,200] DEBUG:  "system":{"get_sysinfo":{"sw_ver":"1.5.4 Build 180815 Rel.121440","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS110(EU)","mac":"AC:84:C6:92:3F:AE","dev_name":"Smart Wi-Fi Plug With Energy Monitoring","alias":"Tulostin","relay_state":1,"on_time":45,"active_mode":"none","feature":"TIM:ENE","updating":0,"icon_hash":"","rssi":-36,"led_off":0,"longitude_i":277094,"latitude_i":642156,"hwId":"044A516EE63C875F9458DA25C2CCC5A0","fwId":"00000000000000000000000000000000","deviceId":"8006ECDCE526845228FC85AA099401541A2406A8","oemId":"1998A14DAA86E4E001FD7CAF42868B5E","next_action":{"type":-1},"err_code":0}},"emeter":{"get_realtime":{"voltage_mv":230764,"current_ma":159,"power_mw":14955,"total_wh":4129,"err_code":0}}}
[2018-12-05 22:00:55,868] DEBUG: Checking statuses during print progress (0).
[2018-12-05 22:03:51,426] DEBUG: Checking statuses during print progress (1).
[2018-12-05 22:03:54,490] DEBUG: Checking statuses during print progress (2).
[2018-12-05 22:03:57,554] DEBUG: Checking statuses during print progress (3).
[2018-12-05 22:04:00,630] DEBUG: Checking statuses during print progress (4).
[2018-12-05 22:04:02,676] DEBUG: Checking statuses during print progress (5).
[2018-12-05 22:04:05,242] DEBUG: Checking statuses during print progress (6).
[2018-12-05 22:04:08,820] DEBUG: Checking statuses during print progress (7).
[2018-12-05 22:04:11,369] DEBUG: Checking statuses during print progress (8).
[2018-12-05 22:04:15,447] DEBUG: Checking statuses during print progress (9).
[2018-12-05 22:04:18,521] DEBUG: Checking statuses during print progress (10).
[2018-12-05 22:04:22,100] DEBUG: Checking statuses during print progress (11).
[2018-12-05 22:04:26,694] DEBUG: Checking statuses during print progress (12).
[2018-12-05 22:04:29,246] DEBUG: Checking statuses during print progress (13).
[2018-12-05 22:04:33,832] DEBUG: Checking statuses during print progress (14).
[2018-12-05 22:04:36,391] DEBUG: Checking statuses during print progress (15).
[2018-12-05 22:04:41,508] DEBUG: Checking statuses during print progress (16).
[2018-12-05 22:04:44,562] DEBUG: Checking statuses during print progress (17).
[2018-12-05 22:04:48,137] DEBUG: Checking statuses during print progress (18).
[2018-12-05 22:04:52,717] DEBUG: Checking statuses during print progress (19).
[2018-12-05 22:04:55,273] DEBUG: Checking statuses during print progress (20).
[2018-12-05 22:05:00,370] DEBUG: Checking statuses during print progress (21).
[2018-12-05 22:05:02,925] DEBUG: Checking statuses during print progress (22).
[2018-12-05 22:05:07,002] DEBUG: Checking statuses during print progress (23).
[2018-12-05 22:05:11,584] DEBUG: Checking statuses during print progress (24).
[2018-12-05 22:05:14,151] DEBUG: Checking statuses during print progress (25).
[2018-12-05 22:05:19,250] DEBUG: Checking statuses during print progress (26).
[2018-12-05 22:05:21,802] DEBUG: Checking statuses during print progress (27).
[2018-12-05 22:05:26,901] DEBUG: Checking statuses during print progress (28).
[2018-12-05 22:05:30,474] DEBUG: Checking statuses during print progress (29).
[2018-12-05 22:05:33,547] DEBUG: Checking statuses during print progress (30).
[2018-12-05 22:05:37,630] DEBUG: Checking statuses during print progress (31).
[2018-12-05 22:05:40,702] DEBUG: Checking statuses during print progress (32).
[2018-12-05 22:05:45,802] DEBUG: Checking statuses during print progress (33).
[2018-12-05 22:05:48,372] DEBUG: Checking statuses during print progress (34).
[2018-12-05 22:05:51,241] DEBUG: Checking status of 192.168.0.32.
[2018-12-05 22:05:51,243] DEBUG: {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{}}}
[2018-12-05 22:05:51,250] DEBUG: IP 192.168.0.32 is valid.
[... cut ...]
[2018-12-05 22:10:09,072] DEBUG: {'autoConnect': True, 'ip': '192.168.0.32', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'gcodeEnabled': True, 'gcodeOnDelay': 0, 'emeter': None, 'countdownOnDelay': 0, 'label': 'Tulostin', 'warnPrinting': True, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'countdownOffDelay': 0, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'autoDisconnect': False, 'useCountdownRules': False, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': '5', 'currentState': 'off', 'sysCmdOn': False, 'sysCmdOffDelay': 0}
[2018-12-05 22:10:09,614] DEBUG: Checking statuses during print progress (100).

jneilliii, 2018-12-05:
Do you have the option enabled to warn while printing?

ikoutone, 2018-12-05:
Yes, it is enabled.

ikoutone, 2018-12-05:
Tested with warn while printing disabled, and plug turned off correctly.

jneilliii, 2018-12-05:
Thanks, one of my other plugins has the same issue report. Haven't had the time to work on it yet. The other option is setting a short delay on the gcode settings so the printer isn't detected as printing by the time the command is run.

nirurin, 2019-04-13:
I just had this happen to me on my HS110, and I do have 'warn while printing' turned on. Is the only current answer to not have any warnings?

jneilliii, 2019-04-13:
Not sure if anyone tried with the delay option, which might work.  Yes, the current status icon coloring does get out of sync sometimes and what you mention is accurate when I have seen it happen.  You can also typically refresh the page and it will re-sync with actual current status.

nirurin, 2019-04-13:
Which delay option? I'm using the gcode delay option of 300seconds in the
settings, is there another option?

jneilliii, 2019-04-13:
So I just looked at the gcode off logic and the delay should work.  The setting I'm referring to is the one highlighted below.

[image]

A threading timer is used to run the off command after that delay setting.  So the completion of the final printing operations need to take less time than the setting you used.  I haven't been able to test yet though as I've got a print running.

nirurin, 2019-04-13:
I had that command set to 300 to give me 5 minutes, and I had @tplinkoff
192.168etc as the last line in my custom gcode from Cura. I think there's
only two lines after that which do something like turn off stepper motors.
Later on when I do my next print I'll try it without any delay to see if
that works, but I'd rather not do that long term as it'll mean the hot end
losing its fan cooling while it's still very hot.

nirurin, 2019-04-22:
Tried it without the warning, still didn't work. TPLINK OFF as a command just seems to not work. I'm trying a print now using the M80 command instead. I am still using the 300 second delay, because turning off the printer without a cooldown is dangerous.

jneilliii, 2019-04-22:
Your @TPLINKOFF command, was it all caps?  If not it might not work.  Also, if you try the other, make sure to use M81 not M80.  `M80 ip` is the power on command whereas `M81 ip` is the power off command.

nirurin, 2019-04-22:
It was copy/paste from the github page, so it was all caps. Unfortunately I realise I did M80 for the current test, so when the print finishes I'll try a shorter one using the proper M81. 

I am putting this code into the Cura printer settings "End GCODE" box, where it automatically puts the gcode at/near the end of the file. Is this meant to work, or am I meant to manually put it as the actual final instruction in the document? Manually doing it would be a pain for every individual print doc.

jneilliii, 2019-04-23:
End gcode in Cura settings should work. One other thing I thought of, the print has to come from octoprint and not from SD card.  There is no way to monitor the commands sent via SD card.

nirurin, 2019-04-23:
Yeh I'm using octoprint directly, not transferring to the sd card.

epalzeolithe, 2019-08-24:
Hi all, same issue there with HS110, poweroff not working with M81 :(

jneilliii, 2019-08-24:
Please confirm your settings, and your gcode command.  Did you try the `@TPLINKOFF ip` option?

epalzeolithe, 2019-08-25:
HI

now it works, what i did :
add 300 to gcode off delay
add m81 command to gcode scripts of octoprint, even if cura was doing its job to add it to the end

jneilliii, 2019-08-25:
Cool, you could also probably disable the warm while printing option and that would allow the plugin to power off. What is happening isa the plugin checks again printer state.

overling, 2019-10-14:
This plugin still breaks the psu control plugin that I believe uses m81(not at the pc to check).  I say still because i reported it forever ago but was told its not possible.  I just tested it again since there might have been some updates but same issue.

greyhound3, 2020-03-08:
For me the more severe option is that I just want to see the energy consumtion and even after ticking the "GCODE trigger" checkbox OFF, M81 won't get send to the printer. The plugin just deletes the command.

jneilliii, 2020-03-12:
I just pushed an update to my current development branch 0.9.19 that changes how gcodes are processed and it should fix the conflict with psucontrol in theory. That being said, many of the features PSUControl has is not incorporated into this branch as well, including idle timer for automatic power off and an abort power off option.

jneilliii, 2020-05-10:
Is this still an issue for you guys with the latest versions?

greyhound3, 2020-05-10:
It works fine for me now, thanks!

closed 2020-05-10 by jneilliii