jneilliii/OctoPrint-TPLinkSmartplug #79: TouchUI menu altered

M4rabunta, 2018-12-20:
Hi
I'm running Octoprint on a raspy with a 3,5" LCD for some months without any particular problem.
But after the last update (1.3.10)  the menu of touchUI is full of ???.
After deactivating all the third party plugins and reactivating them one at a time it appears that TplinkSmartplug generates the problem.
I've noticed two things. The touchUI settings icon disappears and a brief reference to the url http://127.0.0.1:5000/#sidebar_plugin_tplinksmartplug_wrapper can be seen at the bottom of the screen when hovering over the question mark on the top right of the screen (picture 3). The on/off button is located under the menu.
The first picture is the screen seen when connecting with VNC.
The second one is scrolling the screen to the right. We can see the menu extended to the right
I had a similar problem with another plugin some months ago, don't remember which one, and solved it by disabling the plugin.
But i don't want to disable tplinksmartplug. So i've temporarily disabled touchUI.
Hope this helps you to find a solution.

[image]
[image]
[image]

jneilliii, 2018-12-21:
I haven't had a chance to take a look into this yet; but  this could be a combination of issues with both my plugin and TouchUI.  I know @BillyBlaze made updates to TouchUI to support sidebar panels, and my plugin does include that type of template.  I do a custom data-bind [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/28d941c09541be22b6d30e235ac5ecc7816e1e19/octoprint_tplinksmartplug/__init__.py#L97) on the template call however that could be causing problems.  It's intended to only show the sidebar for energy monitoring capable devices. I'm going to cross-post an issue on his repository just in case.

M4rabunta, 2018-12-21:
Hi
I have the explanation. It simply relies on the capabilities of epiphany-browser.
I've tried firefox and chromium (kweb too) and both work fine with little difference.
Firefox displays a vertical scroll bar while Chromium handles vertical scrolling without scroll bar what makes the screen a bit larger.

Both display your plugin in two parts now and both work as intended, i suppose, displaying power consumption and toggle power. Just a remark about the refreshing of the power on icon that takes a certain time to toggle. Maybe a general refresh question.

Finally i must apologize because i didn't think that my Octoprint installation could be outdated by using epiphany-browser. It relies in the fact that Octoprint was running first on a model A then a model B and now a model 3. And as everything worked fine i didn't spend any time changing this configuration.
Now i keep firefox running and see what happens.
Thank's for your great plugins.

leitenky, 2018-12-22:
I have a similar issue where the right drop down has a line that says "tp link smart plug"  and then an icon a few lines lower that shows the "lightening bolt".  I cannot get the smartplug actions to work unless I log in and use the browser with touchui disabled. I can use the Android app Printiod and use the on off commands no problem.

M4rabunta, 2018-12-22:
What browser do you use ?
Maybe try an apt-get update and apt-get upgrade
Then if you want to install firefox or chromium
sudo apt-get install firefox-esr
sudo apt-get install chromium-browser

The following picture shows the screen i have now with firefox on a 3.5" LCD

[image]

and with chromium

[image]

On both TPLink plugin works fine

jneilliii, 2019-01-10:
@leitenky, that is due to TouchUI moving the energy monitoring sidebar panel into the "tabs/navbar" menu of it's interface. At least that's why you see the two items listed as you describe.  So what you are seeing is the navbar icon and the sidepanel converted to a tab by touchui.  Could you please let me know if @M4rabunta's recommendations were of any use to you or not?

PR #107 (2019-05-11): 0.9.16
### Added
- Added HS107 and HS300 multiple plug support, resolves #81.
- Energy data logging and graphing via new tab, resolves #17.
- Temperature runaway protection, resolves #99.
### Fixed
- Change socket receive to grab 1024 chunks until full response is received for larger responses.
- Fixed settings migration issue, resolves #102.
- CSS changes to integrate better with TouchUI, potentially resolves #79.
- Status checks on settings save if plug list changes in anyway.
- Status checks no longer saves the settings in favor of real-time updates to current state, resolves #93.
closed 2019-05-11 by jneilliii