jneilliii/OctoPrint-TPLinkSmartplug #83: Disappearing sidebar?

Jozer99, 2019-01-14:
Firstly, thank you for the plugin, I love having the TP-Link control functionality in Octoprint.  

I notice that when I load the octoprint webpage in my browser, a side panel box with the current power consumption of the TP link plug is displayed.  This sidebar disappears when the page fully loads.  How do I get this sidebar box to stay around? I think having the power consumption of the printer displayed would be interesting.  Looking at the page source, the div "sidebar_plugin_tplinksmartplug_wrapper" is set to "display: none".

Jozer99, 2019-01-15:
My smartplug is an HS-110, and the TP-Link app provides information on power usage.  The sidebar panel still disappears.

jneilliii, 2019-01-15:
Can you enable debug logging in settings, restart octoprint, force a refresh of the browser (ctrl + F5) and then report back here with your plugin_tplinksmartplug_debug.log file please?

Dalewn, 2019-01-19:
Just saw the same behaviour. I only own a HS100, so sadly it is intended behaviour.

One a sidenote: Maybe add a note to your description as it is not clear that the HS110's power usage actually is supported.

jneilliii, 2019-01-19:
Good point @Dalewn, I haven't updated the readme with screenshots or details about that.  I'll add that to my todo list for the next update.

kalle-1989, 2019-01-19:
I have an HS110 and also the same issue with the disappearing sidebar panel for power consumption. I attache my log with removed IP und 
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2776085/plugin_tplinksmartplug_debug.log)
Ids.

jneilliii, 2019-01-20:
That's weird, I'll have to go back and review the code, but it should be working because it's returning the information for the plug for sure.  The last line toward the very end is the energy data.

```
"emeter":{"get_realtime":{"voltage_mv":233653,"current_ma":19,"power_mw":0,"total_wh":85,"err_code":0}}
```

Have you by chance tried restarting octoprint since setting up the plug?

Jozer99, 2019-01-20:
I enabled debugging.  How do I find the debug log file and get it off the pi?

Jozer99, 2019-01-20:
Here is my log, as requested.  
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2776238/plugin_tplinksmartplug_debug.log)

jneilliii, 2019-01-20:
@Jozer99 that looks like an exact copy of @kalle-1989 log file. The timestamps are even exactly the same.

Jozer99, 2019-01-20:
Opps, uploaded the wrong log file.  Here is the real one, I hope.  
[plugin_tplinksmartplug_debug.anon.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2776260/plugin_tplinksmartplug_debug.anon.log)

jneilliii, 2019-01-20:
yep, I see it returning the correct data back.  Have you tried restarting octoprint since adding the plug in settings?

Jozer99, 2019-01-20:
I've had the plugin with these settings for months.  I have restarted it several times today.  The energy monitor sidebar panel disapears during page load still.

jneilliii, 2019-01-20:
Thanks.  If you ssh to your pi and open `~/.octoprint/config.yaml` does the plugin settings show the emeter data saved with the plug?

jneilliii, 2019-01-20:
Thanks.  For some reason the emeter section isn't updating from what's returned by the plug when the plugin checks the status. Were you using a version prior to 0.9.8? I'd like trying deleting the entire tplinksmartplug section from config.yaml and then restart octoprint and reconfigure the plug in settings.

jneilliii, 2019-01-20:
I think I see the issue now.  It's definitely related to code issues on the plugin side.  I'm looking into possible fixes now.

jneilliii, 2019-01-27:
Could you guys test a potential fix for this issue by installing an updated version using the url below in plugin manager?  So far my testing has proven positive, would love to get some additional feedback prior to officially releasing it.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.11.zip

alexisju, 2019-02-03:
Worked for me with the 0.9.11 (didnt worked previously with my HS110(FR))

sysinfo":{"err_code":0,"sw_ver":"1.2.5 Build 180410 Rel.182657","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS110(FR)

jneilliii, 2019-02-09:
Version 0.9.11 officially released.  Thanks @alexisju for confirmation. I still want to split out the energy monitoring request to display accurately, to fix the issue where the sidebar data displays the plug as still drawing power when it's off and should show 0.

closed 2019-02-09 by jneilliii

Jozer99, 2019-02-10:
I can verify that it is now working correctly.  Thank you jneilliii for your plugin, and your excellent assistance with this issue.