jneilliii/OctoPrint-TPLinkSmartplug #85: TP-Link Smartplug (0.9.10) HS105 not working consistantly.

santanig5, 2019-01-31:
Greetings,
I've recently purchased TP-Link HS105 plugs and connected one to my printer, installed your plugin and what I'm finding is that when I click on the lightning bolt it does not change the plug state consistently. I will click it when it is off, the icon rotates but does not change color and the plug remains off, I click a second time and the icon rotates, the plug turns on, but the icon color does not change.
I have the polling set to 1 minute and after 60 seconds the color will change to green.
When I attempt to turn it off, I have to click the icon several times before it will change the state of the plug. Again, it is not consistent.
I captured a log, but am not sure how to retrieve it for you.

Thanks in advance.

jneilliii, 2019-01-31:
If you have debug enabled go into logging section in octoprint's settings and find the tpkinsmartplug log file, download and upload here.

santanig5, 2019-01-31:
Found it! Thanks, I hope its ok that I attached it here and not in GitHub.

jneilliii, 2019-01-31:
Doesn't look like it came through via email.

santanig5, 2019-01-31:
Take two, file is attached and here are the contents

2019-01-28 23:55:34,727 - octoprint.plugins.tplinksmartplug - INFO - ImmutableMultiDict([('checkStatus', u'192.168.198.181')])
2019-01-28 23:55:34,731 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2019-01-28 23:55:34,732 - octoprint.server - INFO - OctoPrint 1.3.10
2019-01-28 23:55:34,734 - octoprint.plugin.core - INFO - 28 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Autoscroll (0.0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_autoscroll
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Bed Visualizer (0.1.7) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
| !CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/cura
|  Custom Control Editor (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_customControl
|  Detailed Progress Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_detailedprogress
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
| !EEPROM Marlin Editor Plugin (1.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_marlin
|  Force Login (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/forcelogin
|  Fullscreen Plugin (0.0.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fullscreen
|  Gcodebar Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GCodeBar
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Octolapse (0.3.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Print History Plugin (1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
| !ProgressBasedOntime Plugin (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_ProgressBasedOnTime
|  SMS Notifier (with Twilio) (0.3.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_smsnotifier
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  TouchUI (0.3.12) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui
|  TP-Link Smartplug (0.9.10) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug
| !Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
2019-01-28 23:55:34,748 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1400.0
|    ram: 918188032
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model B Plus Rev 1.3
|      octopi_version: 0.15.1
|  python:
|    pip: 9.0.3
|    version: 2.7.13
|    virtualenv: /home/pi/oprint
2019-01-28 23:55:34,749 - octoprint.server - INFO - ------------------------------------------------------------------------------
2019-01-28 23:57:42,322 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-01-28 23:57:47,156 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {}
2019-01-29 00:10:34,738 - octoprint.plugins.tplinksmartplug - INFO - ImmutableMultiDict([('checkStatus', u'192.168.198.181')])
2019-01-29 00:12:42,324 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-01-29 00:12:47,138 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {}
2019-01-29 00:23:41,620 - octoprint.filemanager.analysis - INFO - Starting analysis of local:rudder_lh.gcode
2019-01-29 00:23:41,622 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=4800 --speed-y=4800 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/rudder_lh.gcode
2019-01-29 00:23:53,258 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Printing"
2019-01-29 00:23:53,269 - octoprint.plugins.detailedprogress - INFO - Printing started. Detailed progress started.
2019-01-29 00:23:53,282 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-01-29 00:23:54,008 - octoprint.plugins.tracking - INFO - Sent tracking event print_started, payload: {'origin': 'local', 'file': 'd42901adca082449eeef77356e7d0beb96a71ec7'}
[... cut ...]
2019-01-30 13:33:35,479 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-01-30 13:33:35,515 - octoprint.util.comm - INFO - Communication timeout while idle, trying to trigger response from printer.
2019-01-30 13:33:36,002 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {}
2019-01-30 13:33:46,210 - octoprint.server.preemptive_cache - INFO - ... done in 57703.28s
2019-01-30 13:48:35,483 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-01-30 13:48:36,082 - octoprint.plugins.tr

zlcoburn, 2019-01-31:
Hi,

I have a similar issue...The plugin is intermittent sometimes it will work and sometimes it wont.  I noticed in the attached log that sometimes it seems to be able to connect and others it cannot.  When it doesn't seem to work, I am still able to ping the device and control it with Kasa, so it itsn't a signal issue.  Thanks in advance for your help!

Zach

[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2819294/plugin_tplinksmartplug_debug.log)

jneilliii, 2019-01-31:
Thanks @zlcoburn.  From the log it seems at some point the device just stops responding to the messages being sent to it.  For example, taking the following snippet of your log.
```
[2019-01-31 17:16:01,738] DEBUG: Checking status of 192.168.2.28.
[2019-01-31 17:16:01,740] DEBUG: {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{}}}
[2019-01-31 17:16:01,742] DEBUG: IP 192.168.2.28 is valid.
[2019-01-31 17:16:02,402] DEBUG: Sending command {"system":{"get_sysinfo":{}},"emeter":{"get_realtime":{}}} to 192.168.2.28
[2019-01-31 17:16:02,407] DEBUG: « ™K"system":{"get_sysinfo":{"sw_ver":"1.5.3 Build 180528 Rel.120721","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS103(US)","mac":"B0:BE:76:DB:AB:9C","dev_name":"Smart Wi-Fi Plug Lite","alias":"3dPrinter","relay_state":1,"on_time":159,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-58,"led_off":0,"longitude_i":-748488,"latitude_i":399677,"hwId":"5D6724C33274433C7435032F3906DFAD","fwId":"00000000000000000000000000000000","deviceId":"8006A08C711C54DE7D8E55C0953324621B008F35","oemId":"0C4BB88C3A7DB96DD7A093FEAA9CD43C","next_action":{"type":-1},"err_code":0}},"emeter":{"get_realtime":{"err_code":-1,"err_msg":"module not support"}}}
[2019-01-31 17:19:13,181] DEBUG: Turning off 192.168.2.28.
[2019-01-31 17:19:13,184] DEBUG: {'autoConnect': True, 'ip': '192.168.2.28', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': False, 'gcodeEnabled': True, 'gcodeOnDelay': '10', 'emeter': None, 'countdownOnDelay': 0, 'label': 'TP-LINK', 'warnPrinting': True, 'gcodeOffDelay': '10', 'sysCmdOnDelay': 0, 'countdownOffDelay': 0, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'autoDisconnect': True, 'useCountdownRules': False, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': 10, 'currentState': 'unknown', 'sysCmdOn': False, 'sysCmdOffDelay': 0}
[2019-01-31 17:19:13,216] DEBUG: IP 192.168.2.28 is valid.
[2019-01-31 17:19:23,218] DEBUG: Could not connect to 192.168.2.28.
```
at 5:16 it's able to send the get status command, but then a couple of minutes later the device is not responding to the commands being sent to it, but the plugin does see the device as the ip is valid.  This means it's seeing it on the network, but the TP-Link protocol is not responding.  I honestly don't know how I could debug this situation as the communication protocol being used was code appropriated from other sources (linked in readme). I personally have not tested with a HS103 only HS100 and HS110 and wonder if there may be some difference there.  The only time I've seen issues like this the device would never respond and the only way to resolve was to unplug the unit from the wall and plug it back in and all was well.

Are there any errors in your octoprint.log at all?

I'd be curious if an older version of the plugin prior to the energy monitoring additions would make a difference in this situation.  Maybe the plug sees that error message and then blocks communication with the pi for some security reason maybe. If you don't mind try installing the plugin from the url below in plugin manager and see if it is more stable for you or not.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.6.zip

zlcoburn, 2019-02-01:
@jneilliii thanks for the quick reply!  I have attached my log below and will try to uninstall and reinstall from the link you gave me above.  I did find some TPLink errors in the log attached:

2019-01-31 15:54:49,307 - octoprint - ERROR - Exception on /api/plugin/tplinksmartplug [POST]
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1149, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1168, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 99, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 201, in on_api_command
    return flask.jsonify(response)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask/json.py", line 237, in jsonify
    return current_app.response_class(dumps(dict(*args, **kwargs),
TypeError: 'NoneType' object is not iterable
2019-01-31 15:54:49,314 - tornado.access - ERROR - 500 POST /api/plugin/tplinksmartplug (192.168.2.35) 10047.26ms
[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2819621/octoprint.log)

jneilliii, 2019-02-01:
I don't think the error we are seeing here is related to the issue, but I have identified the cause.  I've incorporated it into the next version 0.9.11 that has some additional energy monitoring fixes and was awaiting feedback from other users before releasing. If you want to try that version as well you can install using the following url.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.11.zip

santanig5, 2019-02-01:
[octoprint (1).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2821199/octoprint.1.log)
Unfortunately, it did not work any better for me.
I will try the 9.6 version and report back

santanig5, 2019-02-01:
YES! 9.6 works consistently for me!!
Log file attached.
[octoprint (2).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2821228/octoprint.2.log)

jneilliii, 2019-02-01:
Thanks.  The difference between 9.6 and the later versions was that the state logic was changed to be part of the api response rather than a saved value in the settings. I may go back to the previous logic in order for it to be more stable.

As for the intervals, if polling interval was set to 0 then every 0 minutes the plugin would be sending a request to the plug to get the data and then saving every time which is probably why you were having troubles getting it to change after making that setting.  In cases like that, it's usually easier to open up the config.yaml and edit out the faulting setting.

Not really sure about the countdown timer option as that is something I don't really use and was added as a feature request.  My assumption is that the plug's software doesn't like the 0 in it's own built in timer maybe.

santanig5, 2019-02-01:
Great! I will stay on 9.6 until told otherwise. Also worth noting your temperature failsafe plugin works with this configuration as well, although it seems to utilize the countdown timer delay, I would have thought the shutdown would be immediate.
I will have to play with the shutdown timer and determine if and what the delay should be, I added the code to my OctoPrint Gcode, not my slicer. Is this the recommended use?

zlcoburn, 2019-02-02:
@jneilliii I went straight to the 0.9.11, and it has work perfectly!  Thanks for the quick reply and all of the help.

12345Josparky, 2019-02-03:
Ok so can someone tell me what works here and if 0.9.11 works with HS105 then how do I install it. I only know how to install from the plugin manager. Mine is grayed out also and does nothing. I have the HS105 plug and version 0.9.10?? Thanks for any help.

santanig5, 2019-02-03:
I've reinstalled 0.9.11 since zlcoburn had success and it is working for me too with the intervals set to more than 0.
[image]

[image]

Copy the link from the post above and pate it as shown.

[image]

12345Josparky, 2019-02-03:
Thanks for the reply. Ok well get this I just used my phone to power it on and off and now it works even shuts the printer off at the 900 sec. 15 min I told it too after the print. so I will just stay with this one long as it is working now. Till he uploads the update to the plugin manager. So if anyone else is having the troubles I would recommend turning the plug on and off with your phone app to try it.

santanig5, 2019-02-03:
I probably did the same during my testing without even thinking about it. Good callout!

Sent from my Verizon, Samsung Galaxy smartphone

jneilliii, 2019-02-09:
Officially released 0.9.11.

closed 2019-02-09 by jneilliii