jneilliii/OctoPrint-TPLinkSmartplug #87: After installing this plugin undervoltage icon

12345Josparky, 2019-02-07:
After I installed this I keep getting the undervoltage icon on the right sometimes. Wasn't getting it before installing this plugin. Have other plugins installed also before this one. Any correlation you think??

jneilliii, 2019-02-07:
Are you sure that you aren't confusing the on/off button with the undervoltage icon?  By default the icon used for the button is a lightning bolt.

12345Josparky, 2019-02-07:
So sorry jneilliii I do believe my power supply is crapping out that i use for the raspberry. I pulled it apart and started checking the voltage and it was dropping more than I liked so I found another one laying around and hooked it up and it is all gone now. So sorry for even posting this. Note first thing I did was change the icon you used for a power button icon lol.

closed 2019-02-07 by jneilliii