jneilliii/OctoPrint-TPLinkSmartplug #88: '+' Icon Doesn't Work

DylanCDuke, 2019-02-10:
Hello, 

Just downloaded the plugin, and see the options in my settings. However the '+' icon doesn't do anything, which in turn doesn't allow me to add a link. Any suggestions?

jneilliii, 2019-02-10:
I'll need now more details, specifically are there any errors in your browser's developer console?

Samy83000, 2019-02-10:
Hello,
Same problem for me. Fresh Octoprint install. Tried in Firefox and Edge. The '+' icon doesn't do anything but no error message...?

jneilliii, 2019-02-10:
@DylanCDuke, are you also using one of these browsers? I need to test since I use chrome.

jneilliii, 2019-02-10:
I'm able to reproduce in chrome after removing all my previously configured plugins and starting from scratch.  The error happens on load and I'm looking into a resolution now.

NigelWGMajor, 2019-02-10:
I have the same issue.  Using Chrome, no console messages logging at all.  Not seeing any entries in the bound collection either.  is there a file we can edit the IP address into as a workaround?

This was working fine before I reloaded Octoprint and tried to reinstall the plugin.  I'm happy to debug if You need me to, I have the dev tools in Chrome/Chromium.

jneilliii, 2019-02-10:
I think I have it worked out.  If you guys could please confirm by installing using the url below in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.12.zip

NigelWGMajor, 2019-02-10:
Will do.
thanks.

NigelWGMajor, 2019-02-10:
I can edit now, 
T H A N K S ! 
The icon is spinning but I may have some settings messed up...

NigelWGMajor, 2019-02-10:
I'm rebooting the pi and will try again.

The settings are persisting it seems.  Would it help to do some logging?

NigelWGMajor, 2019-02-10:
I an confirm that the edits are visible and persisting, but the icon just spins.  Trued restarts, and tried adding a second instance but same effect.  If there's anything I can do to help please let me know, I'll check in an hour or two.

Thanks for a great plugin!

jneilliii, 2019-02-10:
I cannot reproduce the spinning issue.  I know why it's doing it but don't know why it would be if there are no errors. The front end sends a request to the backend at which time the spinning should start.  Once the reply comes back from the backend the spinning should stop. Enable debugging and check both octoprint.log and plugin_tplinksmartplug_debug.log for errors.

Samy83000, 2019-02-10:
It works fine for me now.
Thanks a lot!

DylanCDuke, 2019-02-10:
Sorry for the delay here...

@jneilliii I am using chrome.

There is no error in the console on click, but on page load I do see:

```
Uncaught TypeError: Cannot use 'in' operator to search for 'err_code' in undefined
    at packed_plugins.js?db0377f5:13344
    at Object.Ka (packed_libs.js?331a1bb5:11318)
    at packed_plugins.js?db0377f5:13343
    at Function.Pc (packed_libs.js?331a1bb5:11357)
    at Function.Qc (packed_libs.js?331a1bb5:11357)
    at Function.aa (packed_libs.js?331a1bb5:11356)
    at Function.ac (packed_libs.js?331a1bb5:11356)
    at Function.notifySubscribers (packed_libs.js?331a1bb5:11343)
    at Function.fa (packed_libs.js?331a1bb5:11346)
    at tplinksmartplugViewModel.c [as arrSmartplugs] (packed_libs.js?331a1bb5:11346)
```

Currently have a print running so I can not install plugin. But the print will end shortly and I will try. I'll update you ASAP.

jneilliii, 2019-02-10:
YEah @DylanCDuke that was the original error.  Version 0.9.12 linked above should fix that.

DylanCDuke, 2019-02-10:
@jneilliii -

Confirming this worked, thanks for the help!

NigelWGMajor, 2019-02-10:
I'm getting a 500 internal server error when the plugin tries to check status:
GET http://192.168.0.16/api/plugin/tplinksmartplug?checkStatus=192.168.0.100 500 (INTERNAL SERVER ERROR)
The 192.168.0.100 is the correct ip for the TPLink.  16 is the raspi that octoprint is running on.

Any ideas?

NigelWGMajor, 2019-02-11:
Could you please help me find the logs?  I enabled logging but am not sure
where to find them ....

NigelWGMajor, 2019-02-11:
I found the logs, forgot about the hidden folders :)
Also, I think i found the issue - my lan extender was frozen.  I am sorry
to waste you time,  your software Rocks!

Thanks once again!

PR #89 (2019-02-11): 0.9.12
### Fixed
- Issue introduced in previous version prevented fresh installs from working properly. Resolves #88.
closed 2019-02-11 by jneilliii