jneilliii/OctoPrint-TPLinkSmartplug #91: Lost of lightning icon + impossible to setup with "+" icon.

Getoval, 2019-02-17:
Hello,

Until 5 or 6 days my pluggin running well.  But now (I think after an Octoprint update) I lost the lightning icon into the taskbar.
And  when I go into the "TP-Link Smartplug Settings", when I click "+" icon, nothing appens.

I tried to uninstall and re - install but not better.
I saw some "warning" with old pip version during installation.
So I made this update like description here:
https://octoprint.org/blog/2017/11/01/octopi-0.12-pip-issues/?utm_medium=announcements&utm_source=octoprint&utm_content=1.3.10

And I retried to re-install TpLink pluggin.
But when I click on the advanced "option" : "Use --process-dependency-links with pip install" I got and error...

When I unclik this option, the installation is made, but the 2 problems still.

So finally I am lost ....
Thanks for your help.

Christophe.

jneilliii, 2019-02-17:
What version of the plugin are you running?

jneilliii, 2019-02-17:
Any errors in octoprint.log?

Getoval, 2019-02-17:
I installed the last one : 0.9.13

I will looking for log files ...

Getoval, 2019-02-17:
Find enlosed extraction of octoprint log file.
From my side I don't saw error ... (except when I tried to install with "Use --process-dependency-links with pip install" option)
[Extract_Log.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2873343/Extract_Log.txt)

jneilliii, 2019-02-17:
Look in your browser's developer console for errors. You could try deleting all the settings for the plugin in config.yaml and restart octoprint and add your plug back in.

Getoval, 2019-02-17:
I have this error into developer console:
[image]

[image]

jneilliii, 2019-02-17:
Yeah, I think that will get fixed if you clear out the plugin's settings in config.yaml.

Getoval, 2019-02-18:
Hi jneilliii,

The cleaning of the file config.yaml, solved the problem.
Thanks for your time with me.

Bye.
Christophe.

closed 2019-02-18 by Getoval