jneilliii/OctoPrint-TPLinkSmartplug #92: Clicking on the icon does not always turn the outlet off or on.

jimbrownsworkshop, 2019-02-19:
SOmetimes I need to click on the icon 3 or more times before it will turn on/off the power. Sometimes it will change the state but the icon is in the wrong state red when it is on or green when it is off. Clicking again will sync the two back up.

Also it does not turn off the outlet when print job is finished but probably related to the fact that it does not always turn the outlet off on the first try.

The phone app always changes the state from on to off or off to on.

I just un-installed it and re-installed it today so it has the newest version of the plug-in

Ideas on how to proceed I have added M81 192.168.100.167 to the gcode scripts to run when print job is done. 192.168.100.167 is the outlet ip address.

Thanks
Jim Brown

jneilliii, 2019-02-19:
Try adding a delay for the gcode off command so that octoprint has enough time to change from printing to operational. 

As for the sync/current state issues sometimes the pi gets bogged down or the connection between the client and server is stale and the commands are rejected.

jimbrownsworkshop, 2019-02-19:
Great thanks I'll add the delay in and see if that helps with the off at the end of the print job.

Thanks!
Great plug-in!

closed 2019-03-12 by stale[bot]