jneilliii/OctoPrint-TPLinkSmartplug #93: TPLinkSmartPlug Plugin is Changing the Config which Causes Interface Reloads

markschabacker, 2019-03-07:
The TPLinkSmartPlug plugin is changing the config:

```
pi@octopi-mk3:~/.octoprint $ ls -ltr --time-style=full-iso config.yaml*
-rw-r--r-- 1 pi pi 4370 2019-03-06 17:43:58.034429405 -0500 config.yaml-2019-03-06-17-43
-rw-r--r-- 1 pi pi 4372 2019-03-06 18:56:07.147713057 -0500 config.yaml
pi@octopi-mk3:~/.octoprint $ diff config.yaml config.yaml-2019-03-06-17-43
98c98
<           current: 1.500897
---
>           current: 1.09913
100,102c100,102
<           power: 127.945853
<           total: 36.119
<           voltage: 121.815145
---
>           power: 83.926276
>           total: 35.963
>           voltage: 122.055105
pi@octopi-mk3:~/.octoprint $
```

This change to config.yaml is propagating to the interface, which then is forcing a reload of the interface because of a new config hash. See foosel/OctoPrint#2950

jneilliii, 2019-03-08:
Yeah, this is something that I've been trying to figure out how to fix.  I need to detach the current energy monitoring data from being stored in settings to prevent this.  It was something that I've done for the current on/off state, just haven't had the chance to separate the energy monitoring data.

jneilliii, 2019-03-09:
Try installing the 0.9.14 branch by using the url in plugin manager and let me know if that is still giving you the same thing.  I just removed the settings save from the code and it appears that the energy monitoring display is still working for me on a fresh new install.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.14.zip

markschabacker, 2019-03-10:
The code in the `0.9.14` appears to have corrected the issue for me.  Thanks!

jneilliii, 2019-03-10:
With the caveat that the energy data never updates. I'll have to look deeper into that later in the week.

jneilliii, 2019-03-13:
I just pushed an update to the same 0.9.14 branch that now will update the energy data in the sidebar panel during prints or with polling enabled, but I'm still not completely happy with it yet to release. I'm considering a complete re-write of the data structure on the back-end.

PR #107 (2019-05-11): 0.9.16
### Added
- Added HS107 and HS300 multiple plug support, resolves #81.
- Energy data logging and graphing via new tab, resolves #17.
- Temperature runaway protection, resolves #99.
### Fixed
- Change socket receive to grab 1024 chunks until full response is received for larger responses.
- Fixed settings migration issue, resolves #102.
- CSS changes to integrate better with TouchUI, potentially resolves #79.
- Status checks on settings save if plug list changes in anyway.
- Status checks no longer saves the settings in favor of real-time updates to current state, resolves #93.
closed 2019-05-11 by jneilliii