jneilliii/OctoPrint-TPLinkSmartplug #94: Not turning Smart Plug on/off

hermy65, 2019-03-08:
I set this plugin up today and nothing i do will get this plugin to toggle the plug on or off. Attached is my debug log, hopefully that sheds some light on the issue.
[plugin_tplinksmartplug_debug (1).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2947198/plugin_tplinksmartplug_debug.1.log)

hermy65, 2019-03-18:
@jneilliii just noticed that my log file above is in a different language so here is a re-uploaded version
[plugin_tplinksmartplug_debug (3).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/2979884/plugin_tplinksmartplug_debug.3.log)

jneilliii, 2019-03-18:
First off start by removing all options.  It looks like it's trying to use the countdown timer to power on the device after 0 seconds.  Are there any errors in octoprint.log?

closed 2019-04-08 by stale[bot]