jneilliii/OctoPrint-TPLinkSmartplug #95: Unable to edit / add to configuration

pankomputerek, 2019-03-13:
I have been having a really hard time trying to add a new plug to my configuration; none of the configuration buttons appear to be working. I've tried, Edge, Chrome and Firefox with no luck. The following is console extract from Firefox.

`Error calling onBeforeBinding on view model tplinksmartplugViewModel : tplinksmartplugViewModel/self.filteredSmartplugs</<@http://192.168.1.248/static/webassets/packed_plugins.js?046a2b53:456:15
Ka@http://192.168.1.248/static/webassets/packed_libs.js?331a1bb5:11318:333
tplinksmartplugViewModel/self.filteredSmartplugs<@http://192.168.1.248/static/webassets/packed_plugins.js?046a2b53:455:19
Pc@http://192.168.1.248/static/webassets/packed_libs.js?331a1bb5:11357:355
Qc@http://192.168.1.248/static/webassets/packed_libs.js?331a1bb5:11357:128
aa@http://192.168.1.248/static/webassets/packed_libs.js?331a1bb5:11356:477
ac@http://192.168.1.248/static/webassets/packed_libs.js?331a1bb5:11356:333
notifySubscribers@http://192.168.1.248/static/webassets/packed_libs.js?331a1bb5:11343:106
fa@http://192.168.1.248/static/webassets/packed_libs.js?331a1bb5:11346:494
c@http://192.168.1.248/static/webassets/packed_libs.js?331a1bb5:11346:275
tplinksmartplugViewModel/self.onBeforeBinding@http://192.168.1.248/static/webassets/packed_plugins.js?046a2b53:482:12
callViewModelIf@http://192.168.1.248/static/webassets/packed_core.js?120a0354:14855:34
callViewModel@http://192.168.1.248/static/webassets/packed_core.js?120a0354:14809:5
bindViewModels/<@http://192.168.1.248/static/webassets/packed_core.js?120a0354:15531:25
Pn@http://192.168.1.248/static/webassets/packed_libs.js?331a1bb5:11208:530
ur/<@http://192.168.1.248/static/webassets/packed_libs.js?331a1bb5:11229:66
bindViewModels@http://192.168.1.248/static/webassets/packed_core.js?120a0354:15511:13
packed_core.js:15533:25
Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist packed_core.js:15563:33
Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist`

jneilliii, 2019-03-14:
Are there any errors in octoprint.log? Have you tried manually deleting the plugin's settings from config.yaml and stating from scratch?

asif5th, 2019-04-04:
[image]
I'm having the same issue. Ender 3. Configuration won't show up in any browser.

jneilliii, 2019-04-04:
Do you see the same error in the browser's developer console on launching octoprint?

asif5th, 2019-04-04:
No errors. Just unable to edit any sort of configuration. TP-Link Smartplug is turned on and working. Can control it with Google Home or through cellphone. But no configuration options. Uninstalled and reinstalled plugin.

jneilliii, 2019-04-04:
Was it a fresh install or upgraded install originally? May need to get you to try and delete the settings manually from config.yaml as a test.

asif5th, 2019-04-04:
> Was it a fresh install or upgraded install originally? May need to get you to try and delete the settings manually from config.yaml as a test.

Upgraded. I also deleted the TP-Link from my network and that didn't work either.

jneilliii, 2019-04-04:
Try this, uninstall plugin.  Manually delete all the tplinksmartplug settings from config.yaml under the plugin section and then install from scratch.

latchspring, 2019-04-24:
I had a the same issue with no ability to edit or add. Also the icon wasn't even showing up in the menu bar. Other plugins seemed to have issues too. I restarted the OctoPi a couple of times and it seemed to fix itself.

mhigginsmoy, 2019-05-10:
Same issue here as latchspring, but I haven't gotten it to work yet.

jneilliii, 2019-05-10:
Try my current development version by using the url below to install in plugin manager. 

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.16d.zip

jneilliii, 2019-05-11:
Just release 0.9.16 please let me know if this does not resolve the issue for you guys.

alokbhargava52, 2019-05-12:
Installed the release 0.9.16 and see the same issue as noticed above by others

jneilliii, 2019-05-12:
Was it a fresh install or did you have a previous version installed? Can you please check the developer console of your browser for errors on initial page load? Any errors in octoprint.log?

alokbhargava52, 2019-05-12:
> Was it a fresh install or did you have a previous version installed? Can you please check the developer console of your browser for errors on initial page load? Any errors in octoprint.log?

I had TPlink plugin installed earlier and that used to work ok. Today after nearly 6 months gap, when I started OctoPirnt, it showed updates and I installed the latest Octoprint update using Plugin manager. I noticed that my TPlink plug in stopped showing configuration page where I used to set IP address etc.

I uninstalled and reinstalled TP link plugin but it stiil remained in the same condition.

I get errors similar to shown above by pankomputerek

I m not sure how to locate config.yaml file and delete TPLink related entries as I dont know its location.

jneilliii, 2019-05-12:
If you connect to the pi via ssh tou can user the following commands to get to config.yaml.

```
cd ~/.octoprint/
nano config.yaml
```

Delete the tplinksmartplug section under plugins.

alokbhargava52, 2019-05-12:
I deleted tplinksmartplug section and all works now after reinstalling the plugin. Thanks

slimshizn, 2019-10-13:
> I deleted tplinksmartplug section and all works now after reinstalling the plugin. Thanks

I second this. Thank you.

closed 2019-12-26 by jneilliii